rwaldron/johnny-five #970: Arduino tank controlling two dc motors with L298n but servo motor is not working properly

EmilLipovsek, 2015-11-27:
I made a Bluetooth controlling Arduino Tank, with two DC motors, and one servo for the turret witch is holding a laser. But the issue is that when I putted "myservo.attach(4)", left motor stops working. I checked the code( if I putted the same pin somewhere) and I looked at the pins and all the wires( if I connected something wrong), but I can not see the issue. Please help me. 
Here is the code I used without servo(working): 

#define trigPin 13
#define echoPin 12
char junk;
String inputString="";
int enA = 10;
int in1 = 9;
int in2 = 8;
// motor two
int enB = 5;
int in3 = 7;
int in4 = 6;
int ledPin1 = 3;
int laserPin2 = 11;
void setup() {
  pinMode(11, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(trigPin, OUTPUT);
  pinMode(echoPin, INPUT);
  Serial.begin(9600);
  pinMode(enA, OUTPUT);
  pinMode(enB, OUTPUT);
  pinMode(in1, OUTPUT);
  pinMode(in2, OUTPUT);
  pinMode(in3, OUTPUT);
  pinMode(in4, OUTPUT);
}

void loop() 
{
      long duration, distance;
  digitalWrite(trigPin, LOW);  
  delayMicroseconds(2); 
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10); 
  digitalWrite(trigPin, LOW);
  duration = pulseIn(echoPin, HIGH);
  distance = (duration/2) / 29.1;
  if(distance < 20)
  {
  digitalWrite(in1, LOW);
  digitalWrite(in2, LOW);  
  digitalWrite(in3, LOW);
  digitalWrite(in4, LOW);
  delay(2000);
  digitalWrite(in1, LOW);
  digitalWrite(in2, HIGH);  
  digitalWrite(in3, LOW);
  digitalWrite(in4, HIGH);
  delay(500);
  digitalWrite(in1, HIGH);
  digitalWrite(in2, LOW);
  analogWrite(enA, 200);
  digitalWrite(in3, LOW);
  digitalWrite(in4, LOW);
  analogWrite(enB, 200);
  delay(500);
  digitalWrite(in1, HIGH);
  digitalWrite(in2, LOW);
  analogWrite(enA, 200);
  digitalWrite(in3, HIGH);
  digitalWrite(in4, LOW);
  analogWrite(enB, 200);
  }
if(Serial.available()){
    while(Serial.available())
    {
      char inChar = (char)Serial.read(); //read the input
      inputString += inChar;        //make a string of the characters coming on serial
    }
    Serial.println(inputString);
    while (Serial.available() > 0)  
    { junk = Serial.read() ; } 
    if(inputString == "a"){        
      digitalWrite(in1, HIGH); 
      digitalWrite(in2, LOW); 
      analogWrite(enA, 200);
      digitalWrite(in3, HIGH);
      digitalWrite(in4, LOW);
      analogWrite(enB, 200); 
    }
        if(inputString == "b"){        
      digitalWrite(in1, LOW); 
      digitalWrite(in2, HIGH); 
      analogWrite(enA, 200);
      digitalWrite(in3, LOW);
      digitalWrite(in4, HIGH);
      analogWrite(enB, 200); 
    }
        if(inputString == "c"){        
      digitalWrite(in1, HIGH); 
      digitalWrite(in2, LOW); 
      analogWrite(enA, 150);
      digitalWrite(in3, LOW);
      digitalWrite(in4, LOW);
      analogWrite(enB, 150); 
    }
        if(inputString == "d"){        
      digitalWrite(in1, LOW); 
      digitalWrite(in2, LOW); 
      analogWrite(enA, 150);
      digitalWrite(in3, HIGH);
      digitalWrite(in4, LOW);
      analogWrite(enB, 150); 
    }
    if(inputString == "e"){        
      digitalWrite(in1, HIGH); 
      digitalWrite(in2, LOW); 
      analogWrite(enA, 150);
      digitalWrite(in3, LOW);
      digitalWrite(in4, HIGH);
      analogWrite(enB, 150); 
    }
    if(inputString == "f"){        
      digitalWrite(in1, LOW); 
      digitalWrite(in2, HIGH); 
      analogWrite(enA, 150);
      digitalWrite(in3, HIGH);
      digitalWrite(in4, LOW);
      analogWrite(enB, 150); 
    }

      if(inputString == "h"){  
      digitalWrite(in1, LOW); 
      digitalWrite(in2, LOW); 
      digitalWrite(in3, LOW);
      digitalWrite(in4, LOW);
    }
      if(inputString == "i"){  
      digitalWrite(11, HIGH);
      }          
      if(inputString == "j"){  
      digitalWrite(11, LOW);
      }
      if(inputString == "k"){  
      digitalWrite(3, HIGH);
      }
      if(inputString == "l"){  
      digitalWrite(3, LOW);
      }    
   inputString = "";
  }
}

And the one with servo(nonworking):

#define trigPin 13
#define echoPin 12
#include <Servo.h>
Servo myservo;
char junk;
String inputString="";
int enA = 10;
int in1 = 9;
int in2 = 8;
// motor two
int enB = 5;
int in3 = 7;
int in4 = 6;
int ledPin1 = 3;
int laserPin2 = 11;
void setup() {
  myservo.attach(4);
  pinMode(11, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(trigPin, OUTPUT);
  pinMode(echoPin, INPUT);
  Serial.begin(9600);
  pinMode(enA, OUTPUT);
  pinMode(enB, OUTPUT);
  pinMode(in1, OUTPUT);
  pinMode(in2, OUTPUT);
  pinMode(in3, OUTPUT);
  pinMode(in4, OUTPUT);
  myservo.write(90);
}

void loop() 
{
      long duration, distance;
  digitalWrite(trigPin, LOW);  
  delayMicroseconds(2); 
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10); 
  digitalWrite(trigPin, LOW);
  duration = pulseIn(echoPin, HIGH);
  distance = (duration/2) / 29.1;
  if(distance < 20)
  {
  digitalWrite(in1, LOW);
  digitalWrite(in2, LOW);  
  digitalWrite(in3, LOW);
  digitalWrite(in4, LOW);
  delay(2000);
  digitalWrite(in1, LOW);
  digitalWrite(in2, HIGH);  
  digitalWrite(in3, LOW);
  digitalWrite(in4, HIGH);
  delay(500);
  digitalWrite(in1, HIGH);
  digitalWrite(in2, LOW);
  analogWrite(enA, 200);
  digitalWrite(in3, LOW);
  digitalWrite(in4, LOW);
  analogWrite(enB, 200);
  delay(500);
  digitalWrite(in1, HIGH);
  digitalWrite(in2, LOW);
  analogWrite(enA, 200);
  digitalWrite(in3, HIGH);
  digitalWrite(in4, LOW);
  analogWrite(enB, 200);
  }
if(Serial.available()){
    while(Serial.available())
    {
      char inChar = (char)Serial.read(); //read the input
      inputString += inChar;        //make a string of the characters coming on serial
    }
    Serial.println(inputString);
    while (Serial.available() > 0)  
    { junk = Serial.read() ; } 
    if(inputString == "a"){        
      digitalWrite(in1, HIGH); 
      digitalWrite(in2, LOW); 
      analogWrite(enA, 200);
      digitalWrite(in3, HIGH);
      digitalWrite(in4, LOW);
      analogWrite(enB, 200); 
    }
        if(inputString == "b"){        
      digitalWrite(in1, LOW); 
      digitalWrite(in2, HIGH); 
      analogWrite(enA, 200);
      digitalWrite(in3, LOW);
      digitalWrite(in4, HIGH);
      analogWrite(enB, 200); 
    }
        if(inputString == "c"){        
      digitalWrite(in1, HIGH); 
      digitalWrite(in2, LOW); 
      analogWrite(enA, 150);
      digitalWrite(in3, LOW);
      digitalWrite(in4, LOW);
      analogWrite(enB, 150); 
    }
        if(inputString == "d"){        
      digitalWrite(in1, LOW); 
      digitalWrite(in2, LOW); 
      analogWrite(enA, 150);
      digitalWrite(in3, HIGH);
      digitalWrite(in4, LOW);
      analogWrite(enB, 150); 
    }
    if(inputString == "e"){        
      digitalWrite(in1, HIGH); 
      digitalWrite(in2, LOW); 
      analogWrite(enA, 150);
      digitalWrite(in3, LOW);
      digitalWrite(in4, HIGH);
      analogWrite(enB, 150); 
    }
    if(inputString == "f"){        
      digitalWrite(in1, LOW); 
      digitalWrite(in2, HIGH); 
      analogWrite(enA, 150);
      digitalWrite(in3, HIGH);
      digitalWrite(in4, LOW);
      analogWrite(enB, 150); 
    }

      if(inputString == "h"){  
      digitalWrite(in1, LOW); 
      digitalWrite(in2, LOW); 
      digitalWrite(in3, LOW);
      digitalWrite(in4, LOW);
    }
      if(inputString == "i"){  
      digitalWrite(11, HIGH);
      }          
      if(inputString == "j"){  
      digitalWrite(11, LOW);
      }
      if(inputString == "k"){  
      digitalWrite(3, HIGH);
      }
      if(inputString == "l"){  
      digitalWrite(3, LOW);
      }    
   inputString = "";
  }
}
 

Everything is working normally without servo, but I need one for the future projects and for the school competition so please respond quickly. If you can help me email me on: cirilkoemil@gmail.com
And if needed I can show you via skype(witch I will give after emailing me).

dtex, 2015-11-27:
Hi @EmilLipovsek This doesn't look like a Johnny-Five project but I'll try and help. Use of the Servo library disables PWM on pins 9 & 10. Can you move enb to a free digital pin and use pin 5 for en1?

EmilLipovsek, 2015-11-27:
@dtex sure I can I will tell you if it worked in 10 min.
By the way thanks for the quick respond.

dtex, 2015-11-27:
Oops, that should be pin5 for in1, not en1

EmilLipovsek, 2015-11-27:
It is ok I understood. Thank you very much my tank is now working properly.
I found this website today. And I just wanted to know how to fix this problem, but I will probably continue using this forum. So I wanted to know what is Johnny-Five?

rwaldron, 2015-11-27:
If you're controlling this over serial, you might as well switch to Johnny-Five ;)

EmilLipovsek, 2015-11-27:
@rwaldron Ok, but what is Johnny-Five

dtex, 2015-11-27:
Johnny-Five is THE framework for Javascript robotics. You can learn all about it here:

http://johnny-five.io/

I suspect you could have written your robot's code in about 30 lines with Johnny-Five.

rwaldron, 2015-11-27:
@EmilLipovsek this github repository :D

closed 2015-12-07 by dtex