rwaldron/johnny-five #972: SI7020 Thermometer: Wrong Temperature Calculation?

mwittig, 2015-11-30:
https://github.com/rwaldron/johnny-five/blob/master/lib/thermometer.js#L438-L440

The code comment refers to a specification document describing the formula to compute the temperature from the raw data. However, the formula used in the code deviates from the specification.

[Specification](https://www.silabs.com/Support%20Documents/TechnicalDocs/Si7020-A20.pdf):  (**175.72** * Temp_Code/ 65536) - 46.85;
Code:  (**175.25** * Temp_Code/ 65536) - 46.85;

My question is, whether this is a **bug** or *feature**. This is why I put a questionmark on the subject line.

rwaldron, 2015-11-30:
That's a bug

mwittig, 2015-11-30:
Thank you for your swift feedback. 

I have filed a PR to fix the issue. I have changed it for HTU21, too. See page 15 at https://www.adafruit.com/datasheets/1899_HTU21D.pdf

rwaldron, 2015-12-01:
Thanks again!

closed 2015-12-01 by rwaldron