rwaldron/johnny-five #976: digitalRead constant loop

cannehag, 2015-12-04:
Hi, I've been playing around with a Particle Photon reading digital input from a magnet sensor i bought.
According to the documentation, reading a digital pin input using digitalRead should only execute the handler when the value changed. I do get a constant event with the same digital value of 1 or 0 even when the value is the same as previous.
Am I missing something?

board.digitalRead("D3", function(value) {
  console.log("magnet value: " + value);
});

rwaldron, 2015-12-04:
I think this is a documentation error. I don't have a Photon on hand at the moment (I will later today).

Can you share a link to the sensor?

rwaldron, 2015-12-04:
@cannehag update to particle-io@0.11.0

Related: I recommend using the `Sensor` (for analog sensors) or `Sensor.Digital` class for sensors.

closed 2015-12-04 by rwaldron

cannehag, 2015-12-06:
Hm, it did work using my old code, the event was only raised when the value changed.
But I did try the Sensor as suggested:
var sensor = new five.Sensor.Digital(3);
But I'm getting exceptions:

1449422353219 Device(s) particle-io  
1449422353869 Connected particle-io  
1449422354075 Repl Initialized  
>> Board connected...
/Users/joncan/projects/node_modules/particle-io/lib/particle.js:381
  pinInt = (pin.replace(/A|D/, "") | 0) + offset;
                ^
TypeError: undefined is not a function
    at Particle.pinMode (/Users/joncan/projects/node_modules/particle-io/lib/particle.js:381:17)
    at new Sensor (/Users/joncan/projects/node_modules/johnny-five/lib/sensor.js:78:11)
    at new module.exports.Digital (/Users/joncan/projects/node_modules/johnny-five/lib/johnny-five.js:91:10)
    at Board.<anonymous> (/Users/joncan/projects/lab02.js:24:15)
    at Board.emit (events.js:104:17)
    at process._tickDomainCallback (node.js:381:11)

These are the modules I use:
"johnny-five": "*",
"particle-io": "^0.11.0",

rwaldron, 2015-12-07:
Yeah, that's definitely a bug. I'll get this fixed as soon as possible

rwaldron, 2015-12-07:
Actually, that's not a bug—sorry for confusion. There is no pin `3` on the Core or Photon, the pins are: `"D3"` or `"A3"` (i.e. a string with the pin name as shown on the board). I will look at Particle-IO and see if I can safely change it to assume number arguments are always "D*".

cannehag, 2015-12-07:
Oh, I thought the Sensor resp Sensor.Digital did that by convention :) The documentation had "A0" for analog but 3 for digital so I thought it was expecting a number (https://github.com/rwaldron/johnny-five/wiki/Sensor)

rwaldron, 2015-12-07:
> Oh, I thought the Sensor resp Sensor.Digital did that by convention

Naw, pin naming normalization is done by each IO-Plugin, because it would be way too chaotic for Johnny-Five to be responsible for figuring out every single board/platform that it supports. I've always assumed it was easier to allow developers to type what they see printed on the board itself. 

> The documentation had "A0" for analog but 3 for digital so I thought it was expecting a number (https://github.com/rwaldron/johnny-five/wiki/Sensor)

That documentation is describing use with an Arduino. Eventually Johnny-Five's docs will have examples for all of the IO-Plugins and platforms that are supported, but that just takes time :)

cannehag, 2015-12-07:
Thanks for the update, it makes sense :)

A big thanks for quick response!

rwaldron, 2015-12-07:
I've been working on this a bit more today and found a few major issues with their firmware and I'm not sure what to do now—need to think it through a bit :\

1.The firmware attempts to be an I2C slave, but implements no register. 
2. The `cmd` buffer and `index` is not cleared until another command is received, which means that things you only want to do once will actually repeat every `loop` iteration. I'm going to try to send dummy data to avoid that. 
3. ultrasonic fails after a 3-4 reads

The good news: 

1. Things that are working: 
  - Analog read
  - digital read
  - digital write