rwaldron/johnny-five #977: Play with PWM on a Pin

AndreaLombardo, 2015-12-05:
Hi, I have a problem finding how to control PWM signal on a Pin.
This is my sample code
<pre>
var express = require('express'); 
var app = express();
var http = require('http').createServer(app);
var io = require('socket.io').listen(http);
var five = require("johnny-five");
var arduino = new five.Board();

var app.locals.port = 9000;
var arduino_ready = false;
var speed = 50; /*Default initial speed*/

var m_left = {in_1: new MyObject(1), in_2: new MyObject(2), pwm: undefined};
var m_right = {in_1: new MyObject(1), in_2: new MyObject(2), pwm: undefined};

app.use(express.static('./web/'));  /*Where i put my web interface*/

http.listen(app.locals.port, function(){
	console.log('Node ready');
});

arduino.on("ready", function() {
  arduino_ready = true;  
  /*Initialize pwm*/
  m_left.pwm = new five.Pin({pin:10, mode: five.Pin.PWM});
  m_right.pwm = new five.Pin({pin:11, mode: five.Pin.PWM});

  console.log('Arduino ready on '+arduino.port);
});

io.on('connection', function(socket){
	/*When recevie speed message, change the global var speed and set it to the pin*/
	socket.on('speed', function(speed_by_client){
		speed = speed_by_client;
		
		/*Set the value on the pins only if arudino is ready*/
		if(arduino_ready){
			five.Pin.write(m_left.pwm, speed);
			five.Pin.write(m_right.pwm, speed);
		}
		/*
		The above code not works, because I not notice any change of speed in the motor
		Also tried five.Pin.pwmWrite(m_left.pwm, speed); but not work
		*/
	});
});
</pre>
Thanks in advance, and sorry for my english (it'is not my language)

rwaldron, 2015-12-05:
You should be able to use the IO instance method: `board.io.pwmWrite(pin number, 0-255)`, but there might be a nicer way to do whatever you're trying—what are you making?

AndreaLombardo, 2015-12-05:
Thanks for the super fast answer! :+1: 
Im need to pilot two motors using an L298N-H bridge like the one present in this video (https://www.youtube.com/watch?v=dqMdIGI5Les)
The pin for direction are managed by Raspberry GPIO but i want change the speed using the arduino PWM.
I don't understand how to use my L298N-H bridge with the motor class. :-(

AndreaLombardo, 2015-12-05:
In your example board means my arduino instance?
So I must use
<pre>
arduino.io.pwmWrite(m_left.pwm, speed);
arduino.io.pwmWrite(m_right.pwm, speed);
</pre>

rwaldron, 2015-12-05:
Ok, then I recommend using the `Motor` class which has a very nice and very complete API for motor controls: http://johnny-five.io/api/motor/

AndreaLombardo, 2015-12-05:
When you are free, can you link me an example how to connect an L298N-H bridge (like one in my video) and the two motors?
Anyway I understand your firs answer. I used
<pre>
arduino.io.pwmWrite(m_left.pwm.pin, speed);
arduino.io.pwmWrite(m_right.pwm.pin, speed);
</pre>
Thanks again.

rwaldron, 2015-12-05:
Sorry, I didn't see that you posted two responses. Any of the examples marked "h-bridge" will work for your needs. Can I answer any specific questions about using the Motor class? Also if @dtex had a spare moment to help, that would be ❤️

AndreaLombardo, 2015-12-05:
Ook... Thanks :100: 0

rwaldron, 2015-12-05:
I've used almost the exact same breakout before, but can you link me to the one you're using? They might label things differently.

AndreaLombardo, 2015-12-05:
[image]
Actually I use Raspberry GPIO to interact with IN1, IN2 (motor left) and IN3, IN4 (motor right) and the Arduino PWM 10 and 11 to interact with EN1 and EN2.
After your instructions the project works fine. But if I understand how to optimize code and resource is better.
In a moment i try to build a fritzing schema to you.

rwaldron, 2015-12-05:
Here's the basic code I used with the version of that breakout that I have:

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var a = new five.Motor({
    // pwm = ENA, dir = IN1, cdir = IN2
    pins: { pwm: 6, dir: 4, cdir: 7 }
  });
  var b = new five.Motor({
    // pwm = ENB, dir = IN3, cdir = IN4
    pins: { pwm: 5, dir: 8, cdir: 2 }
  });
});
```

Those pins correspond to: 

[image]

dtex, 2015-12-05:
I'm here and happy to help in any way I can but sadly the Raspberry Pi is out of my milieu. @nebrius is the expert there.

I'm glad you got it working. 

Generally speaking Johnny-Five does a nice job of abstracting away the need to write directly to a pin. For this particular motor controller you would want to use the [3-pin motor controller](http://johnny-five.io/examples/motor-3-pin/). The H-Bridge controller assumes you have no additional logic (i.e. no breakout board, just an IC).

rwaldron, 2015-12-05:
> But if I understand how to optimize code and resource is better.

Instances of the motor class won't create any conditions that cause de-optimization. By using the IO methods, the program intention becomes less clear.

AndreaLombardo, 2015-12-05:
[image]
Arduino and Raspberry are connected via USB

AndreaLombardo, 2015-12-05:
O great, i see just now yours example and answers. 
Now is time to recode my project :-D
Thanks again.

rwaldron, 2015-12-05:
Where is your actual code running? I have no idea how this would work (with or without Johnny-Five). How does Arduino and RPi know when the other is doing anything?

AndreaLombardo, 2015-12-05:
The code is for Node.js running on Raspberry. Actually i use function like this:
<pre>
...
...
var arduino_ready = false;
var speed = 50;
var motore_sx = {in_1: new Gpio(12, 'out'), in_2: new Gpio(16, 'out'), pwm: undefined};
var motore_dx = {in_1: new Gpio(20, 'out'), in_2: new Gpio(21, 'out'), pwm: undefined};

arduino.on("ready", function() {
  arduino_ready = true;
  motore_sx.pwm = new five.Pin({pin:10, mode: five.Pin.PWM});
  motore_dx.pwm = new five.Pin({pin:11, mode: five.Pin.PWM});
});

io.on('connection', function(socket){
 //when receive message speed, set the speed for global use
  socket.on('speed', function(client_speed){
    speed = client_speed;
  });

...
...

  socket.on('movments', function(movments){
    //parse movments and call the appropriate functions below
    if(arduino_ready){
     //call function
    }
  });

});

function avanti(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, speed);
  arduino.io.pwmWrite(motore_dx.pwm.pin, speed);

  motore_sx.in_1.writeSync(1);  
  motore_sx.in_2.writeSync(0);
  motore_dx.in_1.writeSync(1);  
  motore_dx.in_2.writeSync(0);
}
function avantiDestra(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, speed);
  arduino.io.pwmWrite(motore_dx.pwm.pin, speed/2);

  motore_sx.in_1.writeSync(1);  
  motore_sx.in_2.writeSync(0);
  motore_dx.in_1.writeSync(0);  
  motore_dx.in_2.writeSync(1);
}
function avantiSinistra(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, speed/2);
  arduino.io.pwmWrite(motore_dx.pwm.pin, speed);

  motore_sx.in_1.writeSync(0);  
  motore_sx.in_2.writeSync(1);
  motore_dx.in_1.writeSync(1);  
  motore_dx.in_2.writeSync(0);
}
function dietro(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, speed);
  arduino.io.pwmWrite(motore_dx.pwm.pin, speed);

  motore_sx.in_1.writeSync(0);  
  motore_sx.in_2.writeSync(1);
  motore_dx.in_1.writeSync(0);  
  motore_dx.in_2.writeSync(1);
}
function dietroDestra(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, speed);
  arduino.io.pwmWrite(motore_dx.pwm.pin, speed/2);

  motore_sx.in_1.writeSync(0);  
  motore_sx.in_2.writeSync(1);
  motore_dx.in_1.writeSync(1);  
  motore_dx.in_2.writeSync(0);
}
function dietroSinistra(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, speed/2);
  arduino.io.pwmWrite(motore_dx.pwm.pin, speed);

  motore_sx.in_1.writeSync(1);  
  motore_sx.in_2.writeSync(0);
  motore_dx.in_1.writeSync(0);  
  motore_dx.in_2.writeSync(1);
}
function sinistra(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, speed/2);
  arduino.io.pwmWrite(motore_dx.pwm.pin, speed/2);

  motore_sx.in_1.writeSync(0);  
  motore_sx.in_2.writeSync(1);
  motore_dx.in_1.writeSync(1);  
  motore_dx.in_2.writeSync(0);
}
function destra(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, speed/2);
  arduino.io.pwmWrite(motore_dx.pwm.pin, speed/2);

  motore_sx.in_1.writeSync(1);  
  motore_sx.in_2.writeSync(0);
  motore_dx.in_1.writeSync(0);  
  motore_dx.in_2.writeSync(1);
}
function mstop(){
  arduino.io.pwmWrite(motore_sx.pwm.pin, 0);
  arduino.io.pwmWrite(motore_dx.pwm.pin, 0);  

  motore_sx.in_1.writeSync(1);  
  motore_sx.in_2.writeSync(1);
  motore_dx.in_1.writeSync(1);  
  motore_dx.in_2.writeSync(1);
}</pre>

writeSync is a method of onoff module to interact with GPIO.
https://www.npmjs.com/package/onoff

dtex, 2015-12-05:
So I guess there's just a USB cable missing from you diagram right? It's not clear to me why you are using both the Arduino and RaspberryPi GPIO's. Is there a reason?

rwaldron, 2015-12-05:
@AndreaLombardo The IO-plugin for RPi includes PWM support, here's the simple "getting started": https://github.com/nebrius/raspi-io#usage This way you can just use the RPi

AndreaLombardo, 2015-12-05:
Yes Arduino and Raspberry are connected with USB cable (as mentioned below the diagram :-) ).
The reason is that I use Raspberry as computer connected to wifi and internet and receive command from client connected to the same lan or via internet by Socket.io. That commands was interpreted by Raspberry and imparted to Arduino. Also, Arduino will be able to read analog data from sensor or other type of device and (with the future recode) i will have all the GPIO pin of Raspberry free to use for other scope.

monteslu, 2015-12-05:
The older pi devices only had a single PWM.  The A+/B+/pi2/pizero all have 2 PWMS:  https://github.com/nebrius/raspi-io/wiki/Pin-Information

rwaldron, 2015-12-05:
> The older pi devices only had a single PWM

Ah—I forgot about that.

dtex, 2015-12-05:
So moving all your IO pins to the Arduino is a reasonable alternative? If so, I say do it and go all in on Johnny-Five. It'll be a lot less code and a whole heck of a lot easier to reason about.

rwaldron, 2015-12-05:
@AndreaLombardo if you're just going to use the `arduino.io.pwmWrite()` methods, then don't do this: 

```js
  motore_sx.pwm = new five.Pin({pin:10, mode: five.Pin.PWM});
  motore_dx.pwm = new five.Pin({pin:11, mode: five.Pin.PWM});
```

Because that makes `motore_sx.pwm` is a `Pin` object and `arduino.io.pwmWrite(number, value)` expects `number` to be just a number.

AndreaLombardo, 2015-12-05:
Yes is my intent to port all on Arduino and use this awesome framework @dtex 
@rwaldron in the meaning time I'm using the form below and it's works fine, because i use motore_sx.pwm.pin when passing the number in the method pwmWrite:
<pre>
motore_sx.pwm = new five.Pin({pin:10, mode: five.Pin.PWM});
arduino.io.pwmWrite(motore_sx.pwm.pin, value);
</pre>
But repeat, my intent now is to recode everything. It's just to learn and have fun.
If I'll do something good I'll be happy to show you my creation (if i don't burn everything) and my code. So you can comment and optimize my idea.

rwaldron, 2015-12-05:
Like @dtex said: 

> moving all your IO pins to the Arduino is a reasonable alternative

That means your program becomes something like this: 

```js
var arduino_ready = false;
var speed = 50;

arduino.on("ready", function() {
  arduino_ready = true;

  motore_sx = new five.Motor({
    // pwm = ENA, dir = IN1, cdir = IN2
    pins: { pwm: 10, dir: 4, cdir: 7 }
  });
  motore_dx = new five.Motor({
    // pwm = ENB, dir = IN3, cdir = IN4
    pins: { pwm: 11, dir: 8, cdir: 2 }
  });  
});

io.on('connection', function(socket){
 //when receive message speed, set the speed for global use
  socket.on('speed', function(client_speed){
    speed = client_speed;
  });

...
...

  socket.on('movments', function(movments){
    //parse movments and call the appropriate functions below
    if(arduino_ready){
     //call function
    }
  });

});

function aventi(){
  motore_sx.forward(speed);
  motore_dx.forward(speed);
}
function avantiDestra(){
  motore_sx.forward(speed);
  motore_dx.forward(speed/2);
}
function avantiSinistra(){
  motore_sx.forward(speed/2);
  motore_dx.forward(speed);
}
function dietro(){
  motore_sx.reverse(speed);
  motore_dx.reverse(speed);
}
function dietroDestra(){
  motore_sx.reverse(speed);
  motore_dx.reverse(speed/2);
}
function dietroSinistra(){
  motore_sx.reverse(speed/2);
  motore_dx.reverse(speed);
}
function sinistra(){
  motore_sx.reverse(speed/2);
  motore_dx.forward(speed/2);
}
function destra(){
  motore_sx.forward(speed/2);
  motore_dx.reverse(speed/2);
}
function mstop(){
  motore_sx.stop();
  motore_dx.stop();  
}
```

rwaldron, 2015-12-05:
I might've gotten some of the directional stuff backward, but that's the general gist :)

dtex, 2015-12-05:
> It's just to learn and have fun.

:+1:  That's a noble goal

AndreaLombardo, 2015-12-05:
But now, it's time to cooking in Italy :spaghetti:, so after dinner there are few lines of code that are waiting for me...
Thanks again for your time and your suggestions guys.

dtex, 2015-12-11:
Two questions:

1. How was dinner?
2. Should we close this issue?

AndreaLombardo, 2015-12-11:
Yes, you can close it. Dinner means a lunch in the evening... Isn't it!?
Again Thanks!

dtex, 2015-12-11:
Dinner usually happens in the evening and more often than not is the last meal of the day. Occasionally on holidays we'll have "dinner" during the day because it's the big, official meal where everyone is present. 

Please let us know how your robot goes. [The Johnny-Five Gitter chat](https://gitter.im/rwaldron/johnny-five) is a great place to share pictures and stories about projects.

closed 2015-12-11 by dtex

rwaldron, 2015-12-13:
😃