rwaldron/johnny-five #979: Missing ubuntu section in the getting started document

elopio, 2015-12-08:
In the getting started page there is no Ubuntu section:
https://github.com/rwaldron/johnny-five/wiki/Getting-Started

I'm trying to run the led example in my arduino uno and I get:
Error: Cannot find module 'serialport'
For some reason it wasn't installed when I did npm install johnny-five. That might be a separate issue.

dtex, 2015-12-08:
@elopio yes it's a node-serialport issue but we should definitely add a note here. Can you tell me if this solution works for you?

https://github.com/voodootikigod/node-serialport/blob/d9a08ff62004aa86dc21a3097fb84848f47f90b4/README.md#desktop-debianubuntu-linux

If so, we can add a cross-reference to it from the Getting Started page.

elopio, 2015-12-08:
Yes, after installing nodejs-legacy I could npm serialport.

dtex, 2015-12-08:
This seems simple but would you mind reviewing? I'm not an Ubuntu user.

It's all subordinate to "pre-requisites"

------------------------
### Ubuntu

- Install Node.js >= 0.10.x
- Install the [nodejs-legacy](https://packages.debian.org/jessie/nodejs-legacy) package
- Install [build-essential](https://packages.debian.org/jessie/build-essential) or a suitable alternative

````
# Ubuntu node
sudo apt-get install build-essential
sudo apt-get install nodejs nodejs-legacy
````
------------------------
Is there a better, non-build dependent, link to use for packages (wheezy, sid, etc)?

elopio, 2015-12-09:
I would name the section "Ubuntu and Debian", should be the same for both.
And I just wouldn't like the packages to anything. I don't know about a good general link for them.

looks good to me. Thanks.

dtex, 2015-12-09:
Thanks @elopio I've incorporated your suggestions and updated the wiki

closed 2015-12-09 by dtex