rwaldron/johnny-five #981: Q: SI7020 Revisted - i2cRead Temperature & Humidity Register

mwittig, 2015-12-10:
Question 1:
I am wondering why the i2cRead command for the humidity register reads three bytes for a 16-bit register. Does the read cycle requires an extra byte to be read? I couldn't figure it from the data sheet.

https://github.com/rwaldron/johnny-five/blob/master/lib/imu.js#L617-L620

Question 2:
During a humidity measurement the temperature is also taken and kept in register 0xE0 which serves as cache. However, if the temperature is read from register  0xE3 a new measurement will be forced. So, my question is why is register 0xE3 used instead of using the "cache" register? As far as I understand this, using register 0xE0 will halve the measure cycles perormed.  

Question 3:
The driver reads the sensor registers at a subsecond frequency eventhough I have set the higher layer Temperature and Humidity objects to a frequency of 10000ms ("freq" property). Why is the i2c sensor read at such a high fequency? This causes a lot of traffic with the physical board (may results in higher CPI loads and network IO with remotes). It might even degrade sensor lifetime or result in degraded accuracy if the sensor heats up (this might be an issue with temperature readings in particular). Shouldn't the lowest frequency set at the higher layer constrain the read cycle of the driver?

rwaldron, 2015-12-13:
Sorry for the delay, I've been pretty swamped the also few days. I'm still mobile, but will have time to write up a proper response tomorrow.

mwittig, 2015-12-13:
No worries. I am looking forward to your response. Thanks your very much in advance for looking into this.

mwittig, 2015-12-19:
Should I place my questions elsewhere? I tried to reach the *NodeBots Community Forum* mentioned in the README, but apparently it is down.

rwaldron, 2015-12-19:
@mwittig this is the right place, sorry for the delay in responding :|

> Question 1:
> I am wondering why the i2cRead command for the humidity register reads three bytes for a 16-bit register value. Does the read cycle requires an extra byte to be read? I couldn't figure it from the data sheet.

Just looks like an incorrect data length, I've fixed it.

> Question 2:
> During a humidity measurement the temperature is also taken and kept in register 0xE0 which serves as cache. However, if the temperature is read from register 0xE3 a new measurement will be forced. So, my question is why is register 0xE3 used instead of using the "cache" register? As far as I understand this, using register 0xE0 will halve the measurement cycles perormed.

There was no particular reason why 0xE3 was used and this optimization is definitely ideal. Fixed. 

> Question 3:
> ...
> Why is the i2c sensor read at such a high fequency? 

To allow for highly responsive client code.  All sensors have always been read at least 50Hz—a precedent originally (it has since been changed to 1kHz) set by the StandardFirmata IO protocol (which is baseline for Johnny-Five)

> This causes a lot of traffic with the physical board 

For remote or co-processor scenarios, only one request should be sent for given pin or I2C address, indicating that pin or address must be a in "continuous reporting" mode. Once received, the remote or co-processor will store a continuous reporting flag in memory, which will indicate that the pin or address needs to be "sampled" on each cycle (within a given interval constraint, which I will explain at the very end).

> (may result in higher CPU loads and network IO with remotes).

The processor itself determines the frequency of sensor measurements and reports them when it can. Johnny-Five will process whatever it receives, generally by storing reported values in some internal state.

>  It might even degrade sensor lifetime or result in degraded accuracy if the sensor heats up (this might be an issue with temperature readings in particular). 

I want to research more about this, but my naive guess would be that if this were the case, then the sensor itself wouldn't have been designed for use with an I2C interface. According to http://www.i2c-bus.org/speed/: 

> The speed grades (standard mode: 100 kbit/s, full speed: 400 kbit/s, fast mode: 1 mbit/s, high speed: 3,2 Mbit/s) are maximum ratings. Compliant hardware guaranties that it can handle transmission speed up to the maximum clock rate specified by the mode.

Section "3.1.3 Data validity" of the [I2C specification](http://www.nxp.com/documents/user_manual/UM10204.pdf) states:

> One clock pulse is generated for each data bit transferred.

The Si7020 has a max serial clock frequency of 400kHz, so 400kbits/s. Here are two screenshots from a logic analyzer, each shows the master transmitting the address and register and the slave transmitting the address and bytes read.

[image]
[image]

Each "read" (a request for bytes from a specified register) of the sensor will be 40 bits transferred.

| Read Frequency | kbits/s |
|----------------|---------|
| 50Hz | 2 |
| 200Hz | 8 |
| 1kHz | 40 |

This doesn't come anywhere near the max 400kbits/s, so I surely hope that frequency won't degrade the sensor—if it does, I would demand a refund. 

> Shouldn't the lowest frequency set at the higher layer constrain the read cycle of the driver?

This isn't how other modern platforms (iOS, Android, etc) are designed, so I'm not sure about the basis of this suggestion. Sensor reading frequency is decoupled from user-code to allow the system to work on a multitude of hardware platforms with minimal behavioral disparity. The (poorly named) `freq` property determines the rate in which user code is allowed to act on (via a "data" event) the values delivered by the decoupled process. Not all hardware has the same reading _and_ reporting frequency, but Johnny-Five programs can be written such that this doesn't matter (it matters to the creator to select the appropriate hardware for their project), because the semantics are defined in a way that is decoupled from the actual interaction with the hardware itself.

All of that aside, there _is_ a way to completely override the reporting speed (for all analog and I2C sensors): 

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

  // Use the board's `samplingInterval(ms)` to
  // control the actual MCU sampling rate.
  //
  // This will limit sampling of all Analog Input
  // and I2C sensors to once per second (1000 milliseconds)
  this.samplingInterval(1000);

  // Keep in mind that calling this method
  // will ALWAYS OVERRIDE any per-sensor
  // interval/rate/frequency settings.
});
```

mwittig, 2016-01-12:
Happy new year! I am sorry for the delay in reply. Thank you very moch for your detailed answers which really helped to better understand some aspects of J5 and its underlying technology. I see, my (wild) assumptions about the device heating up (and degradation) were a bit far fetched and the inaccuracy of the measurement data in my case  is perhaps the result of the poor production quality of the device. As part of my tethering experiments I am planning to experiment with reducing the sampling interval. It might be sensible to limit it for remote Wifi enabled boards like Wemos D1.

rwaldron, 2016-01-12:
> Happy new year!

Same to you :)

> Thank you very much for your detailed answers 

You're very welcome!

> As part of my tethering experiments I am planning to experiment with reducing the sampling interval.

Excellent—please post back with any progress or pitfalls!

dtex, 2018-01-06:
Hi @mwittig I just wanted to touch base and see if it is okay to close this issue.

mwittig, 2018-01-06:
@rwaldron Thank you very much for the the detailed response to my various question and the  comprehesive analysis of the related issues. I

@dtex  Thanks for the reminder. Everyting has been covered. I am closing this now.

closed 2018-01-06 by mwittig