rwaldron/johnny-five #983: Support for WeMos D1

cannehag, 2015-12-11:
Hi, any chance johnny-five support WeMos D1 and/or D1 mini ?
http://www.wemos.cc/wiki/doku.php?id=en:start

dtex, 2015-12-11:
It's not supported yet but I don't see any reason it couldn't be supported in the future. For future reference, here is where we keep track of supported boards:  

http://johnny-five.io/platform-support/

rwaldron, 2015-12-15:
@monteslu @ajfisher where are you guys with ESP-8266 stuff?

ajfisher, 2015-12-15:
@rwaldron transparent bridge is definitely the current preferred support option. 

An IO port is still being worked through slowly. Currently there's a transport method which will allow arbitrary execution of messages across relevant transport methods (eg UDP-local, UDP-Serial, Serial-Local etc) but there's a bit to do yet on top of that. 

There's been some discussion of a direct port of firmata using the ESP8266 Arduino stuff which will *sort of work* but doesn't deal with things like timing resolution and transient network silliness, endpoint targeting, device discovery etc which is what we're trying to work through more comprehensively.

IMO the ESP8266 in bridge mode is adequate for most cases and the ESP32 will actually open up considerably more capability for a standalone device with more headroom (32 bit, 40 IO pins, 400KB RAM, BT, dedicated WiFI processor) and a comparable price point and on the market within 6 months. 

As such my sense is that this is a more viable future target for support as it's entirely possible we can be doing stuff like sending messages that can then be held locally and run rather than constantly orchestrated from J5.

rwaldron, 2015-12-15:
> entirely possible we can be doing stuff like sending messages that can then be held locally and run rather than constantly orchestrated from J5.

Similar to "reporting" and "continuous read"?

boneskull, 2016-01-13:
@rwaldron where do I send the hardware? :wink:

mateuszroth, 2016-05-07:
Check my solution: https://github.com/mateuszroth/esp8266-web-control-panel

mdingena, 2017-11-27:
@mateuszroth I've followed your instructions, and I've also cloned your repo. I'm unable to get it to work. I've enabled debugging and the serial console prints messages whenever a TCP connection is established or becomes disconnected. When I run `node ./devices/client.js`, a connection is established (as proven by the serial console, and the builtin LED turning on). But after 100 seconds (`timeout: 1e5`), the board `ready` event has still not fired, the connection times out, and the serial console immediately print `TCP connection disconnected`.

I'm running on Windows 7 x64, Node 8, npm 5, and these dependencies:
```
  "dependencies": {
    "etherport-client": "^0.1.3",
    "firmata": "^0.17.0",
    "johnny-five": "^0.11.7"
  }
```

dtex, 2018-11-23:
I've added a reference to this on the [Requested Features page](https://github.com/rwaldron/johnny-five/wiki/Requested-Features)

closed 2018-11-30 by stale[bot]

BluVueMatthew, 2019-03-23:
This still show on the requested features page, but it has been closed. Does that mean the board will not be supported ever? If so please update the requested features page. Thanks

dtex, 2019-03-23:
Hi @BluVueMatthew 

There is no reason to believe it will not be supported ever, that's why it still lives on the Requested Features page. We just close the requested feature issues because they would overwhelm the other issues (there's lots of hardware out there).

monteslu, 2019-04-17:
Just gave it a shot on a Wemos D1 r1
Works fine. A few notes:

* install the esp8266 stuff in arduino boards manager
* use StandardFirmataWifi sketch in arduino, and edit the wifiConfig.h for your wifi settings
* uncomment this line `#define SERIAL_DEBUG`
* use serial monitor at 9600 baud to see what address your device gets
* connect to it with https://github.com/mwittig/etherport-client/tree/master#using-standardfirmatawifi-with-esp8266

monteslu, 2019-04-17:
@BluVueMatthew 
one more thing, the pins do not match on the d1 r1 with what's printed at the board at all.   Not sure if that got any better on the r2.
I got an led to blink on what the board has labled as d11/d7 by specifying pin 13 in my johnny-five code.
[image]

akarabach, 2022-02-14:
any updates on it ? or it was released and wemos d1 supported since then ? @monteslu @dtex 

by the way, I can not find here https://github.com/rwaldron/johnny-five/wiki/Requested-Features link to this issue