rwaldron/johnny-five #984: Grove shield + Grove - LCD RGB JHD1313M1: no text

k33g, 2015-12-11:
Hello,

I can't display anything on the JHD1313M1, I can change the background color, but I can't display text.

I use this code:

    var five = require("johnny-five");
    var board = new five.Board();

    board.on("ready", function() {

      var lcd = new five.LCD({
        controller: "JHD1313M1"
      });
      lcd.bgColor(0,128,64).print("ping");
    });

I'm using an arduino uno + groveshield

I know that the JHD1313M1is ok because I've tested it on a GrovePi with Python Framework of Dexter Industries.

I can't find what I've done wrong

Thank you
Philippe

k33g, 2015-12-12:
an other sample:

    var five = require("johnny-five");
    var board = new five.Board();

    function linear(start, end, step, steps) {
      return (end - start) * step / steps + start;
    }

    board.on("ready", function() {

      var redLed = new five.Led(2);
      var blueLed = new five.Led(3);
      var greenLed = new five.Led(4);

      var lcd = new five.LCD({
        controller: "JHD1313M1"
      });

      var rotary = new five.Sensor("A0");

      // This will blink the LEDs
      redLed.blink(500);
      greenLed.blink(100);
      blueLed.blink(300);

      rotary.scale(0, 255).on("change", function() {
        var r = linear(0xFF, 0x4B, this.value, 0xFF);
        var g = linear(0x00, 0x00, this.value, 0xFF);
        var b = linear(0x00, 0x82, this.value, 0xFF);

        lcd.bgColor(r, g, b).cursor(0,0).print(this.value);
      });

    });

rwaldron, 2015-12-14:
I'm not sure what is causing your program to fail—I've set up the same (almost) hardware and it works as expected: https://www.youtube.com/watch?v=bCBC7-2Fjm0

k33g, 2015-12-15:
Argh! Tonight, I'm going to try this:
- the same thing with an older version of node (actually, I'm using 4.2.2)
- and with the same hardware, but with Python and firmata

Thank you very much for your time @rwaldron :)

keep in touch
Philippe

k33g, 2015-12-20:
Hi! It's probably because of my component. I've tried with older node versions, but I've got the same problem. Strangely, with a GrovePi and the python sdk, it's ok with my LCD component. Never mind! :) I will try to get an other LCD component later. Thx so much for your help

closed 2015-12-20 by k33g