rwaldron/johnny-five #985: error device or firmware

arirdhi, 2015-12-13:
if I run standard firmata, no error at all
but if you are running another program, I get this error
[image]

and here's the program

var plotly = require('plotly')('xxxxxx', 'xxxxxx');
var five = require("johnny-five");
var board = new five.Board();

var data = [{x:[], y:[], stream:{token:'25tm9197rz', maxpoints:200}}];
var layout = {fileopt : "overwrite", filename : "uv sensing nodey arduino!"};

board.on("ready", function() {

  // create a new Analog Sensor sensor object
  new five.Proximity({
    pin: "A0",
    freq: 1000 // get reading every 1000ms
  });

  // initialize the plotly graph
  plotly.plot(data,layout,function (err, res) {
    if (err) console.log(err);
    console.log(res);
    //once it's initialized, create a plotly stream
    //to pipe your data!
    var stream = plotly.stream('25tm9197rz', function (err, res) {
      if (err) console.log(err);
      console.log(res);
    });

    // this gets called each time there is a new sensor reading
    proximity.on("data", function() {
      var proxValue = this.value;
      console.log('prox : '+proxValue);

      var data = {
        x : getDateString(),
        y : proxValue
      };
      console.log(data);
      // write the data to the plotly stream
      stream.write(JSON.stringify(data)+'\n');
    });
  });
});

// little helper function to get a nicely formatted date string
function getDateString () {
  var time = new Date();
  // 14400000 is (GMT-4 Montreal)
  // for your timezone just multiply +/-GMT by 3600000
  var datestr = new Date(time - 14400000).toISOString().replace(/T/, ' ').replace(/Z/, '');
  return datestr;
}

can anybody help me ?
it's my first time using johnny-five
thank's

rwaldron, 2015-12-13:
Your Arduino must have StandardFirmata, or one of the capability-equivalent variants, uploaded.

mwittig, 2015-12-15:
Just to add to Rick's comment, you should add an "error" event handler to the board instance to handle board errors gracefully and avoiding your program exiting with an uncaught exception.

Resseguie, 2016-01-23:
@arirdhi can you confirm if you had StandardFirmata (or equivalent) on your arduino when getting this error?

dtex, 2018-01-16:
@arirdhi I'm going to close this since we never heard back. Feel free to re-open if necessary.

closed 2018-01-16 by dtex