rwaldron/johnny-five #987: Grove I2C Motor Driver support

rwaldron, 2015-12-15:
@dtex and I will work with @Legendaire to add support for these: http://www.seeedstudio.com/depot/Grove-I2C-Motor-Driver-p-907.html

From what I've learned reading through their C drivers, the motor speeds need to be set at the same time, which requires some level of coordination between the two `Motor` instances. Here's a rough sketch: 

- Implement an `Expander` controller that does the coordination and manages state persistence between two `Motor` instances. 
- Implement a `Motor` controller that exposes: 
  - `initialize` (similar to `PCA9685`)
  - `setPWM` (similar to `PCA9685`) 
  - `setPin` (similar to `EVS_EV3`)

dtex, 2015-12-15:
I'll wager that we can write one byte to 0x82 to control motor A or one byte to 0x83 to control motor B. We won't always have to write two bytes to 0x82.

rwaldron, 2015-12-15:
I'll have a look at their firmware and see what I can learn

rwaldron, 2015-12-15:
@dtex I was really hoping you were right, but then I found this: https://github.com/Seeed-Studio/Grove_I2C_Motor_Driver/blob/master/burn%20firmware%20for%20Atmega8%20using%20ISP/AVRStudio%20Prj/I2CmotorFV1/main.c#L239-L253

rwaldron, 2015-12-15:
But this seems promising... https://github.com/intel-iot-devkit/upm/blob/master/src/grovemd/grovemd.h#L89-L90

dtex, 2015-12-15:
Ya, just saw those. That does look good. 

So I guess instead of I2C'ing into an IC with a single register, we're I2C'ing into a micro controller with it's own logic, etc. More like a backpack.

rwaldron, 2015-12-15:
> So I guess instead of I2C'ing into an IC with a single register, we're I2C'ing into a micro controller with it's own logic, etc. More like a backpack.

Yep. I was just sketching out an `Expander` for this and immediately ran into an issue: direction. I was going to write this thing to treat the driver as an io expander that had 2 output pins: A & B. The `Expander` controller would implement `pwmWrite()` and track both A and B in its state object. The problem is that there is no such thing as "direction" (in the "cw" or "ccw" sense) in a `Expander`. And then there's this nonsense: the direction command is _always_ one command: https://github.com/Seeed-Studio/Grove_I2C_Motor_Driver/blob/master/burn%20firmware%20for%20Atmega8%20using%20ISP/AVRStudio%20Prj/I2CmotorFV1/main.c#L237 

As I wrote that I thought of this: implement `digitalWrite` for the `Expander` controller, all it would do is set the "direction" for A or B in the state object, then send the direction command if the new direction is different from the previous direction (initializes to `null`). I will put it together and you can tell me if sane

dtex, 2015-12-15:
One of the beautiful things about expander is that it can be used for many things. This is a one off. Don't clutter expander with this convoluted stuff that will only ever work on one friggin' (proprietary!) board with shitty firmware.

I seem to recall you recommended that Kas put one of her devices into a plug-in for J5. Is that the best path here? Do we have any example plug-ins yet?

rwaldron, 2015-12-15:
Generally the criteria I use for "should it be a plugin" is: if there is no existing component class and we can't find > 1 device variants, then it should be a plugins. This is just a motor driver, and we have a `Motor` class, so shouldn't it be supported in the framework? But maybe you're right, maybe an `Expander` controller is not the right way, maybe just a `Motor` controller? Here's what we know: 

- A & B can be separated for setting speed
- A & B cannot be separated for setting direction

As much as it's a hack, we could do something like this: 

```js
  GROVE_I2C_MOTOR_DRIVER: {
    REGISTER: {
      value: {
        ADDRESS: 0x0F,
      }
    },
    COMMANDS: {
      value: {
        SET_SPEED: 0x82,
        SET_PWM_FREQUENCY: 0x84,
        SET_DIRECTION: 0xAA,
      }
    },

    initialize: {
      value: function(opts) {
        var state = priv.get(this);
        var shared = priv.get("GROVE_I2C_MOTOR_DRIVER");

        if (!shared) {
          shared = {
            direction: {
              A: 0x01,
              B: 0x01,
            },
            speed: {
              A: 0,
              B: 0,
            }
          };

          priv.set("GROVE_I2C_MOTOR_DRIVER", shared);
        }

        state.shared = shared;

        this.address = opts.address || this.REGISTER.ADDRESS;

        opts.address = this.address;

        this.io.i2cConfig(opts);
      }
    },
    setPWM: {
      value: function(pin, value) {
        var state = priv.get(this);
        var speed = Board.constrain(value, 0, 255) | 0;

        state.shared.speed[pin.toUpperCase()] = speed;

        this.io.i2cWrite(this.address, [
          this.COMMANDS.SET_SPEED,
          state.shared.speed.A,
          state.shared.speed.B,
        ]);
      }
    },
    dir: {
      value: function(pin, value) {
        var state = priv.get(this);
        //...
      }
    }
  }
```

dtex, 2015-12-15:
I think that makes more sense than adding an expander.

dtex, 2015-12-15:
... Because we would be putting something that's specific to a single motor controller into expander and that would be messy.

rwaldron, 2015-12-15:
Thanks for making me re-think that :)

closed 2015-12-17 by rwaldron