rwaldron/johnny-five #988: Sparkfun dual H-bridge Edison block standby pin

Legendaire, 2015-12-15:
The Sparkfun H-bridge block needs to have a standby pin be put high or it stays in standby.  Hence, it will do nothing until this is pulled high.  Here is the snippet I used to make it work.

    var standbyPin = new five.Pin(47);
    standbyPin.high();
    
    var motorA = new five.Motor({
        pins: { pwm: 20, dir: 33, cdir: 46 }
    });

    motorA.fwd(255);

dtex, 2015-12-15:
I knew that was too easy :-P

@rwaldron For the API signature I propose we handle this pin similar to the way we handle current sensing pins. As a separate object or property in the opts.

Legendaire, 2015-12-15:
If it helps I found this example from sparkfun.  The reference is on line 70.
https://github.com/sparkfun/SparkFun_H-Bridge_Block_for_Edison_CPP_Library/blob/master/SparkFun_TB6612_Edison.cpp

dtex, 2015-12-15:
Thanks. It's pretty straightforward. Most h-bridge breakouts just wire that straight to + on the logic circuit. That's what we do with H-Bridges that we wire up ourselves. One catch is (as best I can tell) everyone else on earth calls this an enable pin, not standby. When I saw it was named standby I assumed that voltage needed to be low for the motor to work.

rwaldron, 2015-12-15:
> I knew that was too easy :-P

What was I even thinking?? There's no free lunch!

;)

rwaldron, 2015-12-15:
>  For the API signature I propose we handle this pin similar to the way we handle current sensing pins.

Seems like the right fix!

dtex, 2015-12-17:
I'm having second thoughts. I think I should put the enable pin into the pins object. Here are the two possible signatures:

````
var motor = new five.Motor({
  pins: {
    pwm: 9,
    dir: 8
  },
  enable: 7
});
````
OR
````
var motor = new five.Motor({
  pins: {
    pwm: 9,
    dir: 8,
    enable: 7
  }
});
````
The former is what I initially proposed, but enable is a pin so shouldn't it be in pins? The only drawback I can see if that if a user needs an enable pin then they must use a pins object instead of an array when creating their motor. We have no way of automatically differentiating between a 2-pin with enable and a 3-pin otherwise.

dtex, 2015-12-18:
Is there any reason I shouldn't just set the enable (aka standby) pin high when we instantiate the motor and just leave it that way for as long as our program is running?

CC @jonhylands

JonHylands, 2015-12-18:
In general, if your robot doesn't need to go into a power saving mode, then that is a reasonable thing. Not every h-bridge chip has an enable, and sometimes they are enable high and sometimes enable low.

If you're building a driver for that specific h-bridge then I would do it the way you're suggesting (enable it on instantiation).

dtex, 2015-12-18:
Thanks @JonHylands :+1:

dtex, 2015-12-18:
I've been on a high-horse re: people need to start building useful things with nodebots so I think a power saving mode makes sense. I'm going to make the enable pins switch on and off to meet the desired state. I do need to accommodate the possibility of an enable low pin so we will have ```pins.enable``` or ```pins.enableLow``` unless someone has a better suggestion.

Legendaire, 2015-12-18:
I believe they put the enable pin on this driver since it's really designed for a low power edison device where power consumption is important (most of the time).  So I'd say we would want to give the programmer the ability to turn it on and off.  How about three properties:

Legendaire, 2015-12-18:
enablePin - the pin which enables and disables the driver
enablePinState - High or low to define if the pin is a high enable or low enable
enabled - true or false.  This turns on and off the driver.
Is this simple enough?

dtex, 2015-12-18:
I kinda like leaving it up to the programmer to enable/disable. It sure would be easier to write. 

One thought, instead of an enabled property I'm inclined to expect the user to call ```motor.enable()``` or ```motor.dsiable()``` but I may just be saying that because it would be so much easier to code. 

If this gets any more complicated I might be tempted to rebuild motor with a state machine (not joking, it's getting pretty convoluted). I just discovered a completely superfluous param on an internal method.

Legendaire, 2015-12-18:
Would there be a way to know if the motor is already enabled?

dtex, 2015-12-18:
Hmmm, good question. A read only ```enabled``` property perhaps?

rwaldron, 2015-12-18:
w/r to signatures, I prefer the second one: 

```js
var motor = new five.Motor({
  pins: {
    pwm: 9,
    dir: 8,
    enable: 7
  }
});
```

>  instead of an enabled property I'm inclined to expect the user to call motor.enable() or motor.dsiable()

Yes (for behavior)

> A read only enabled property perhaps?

Yes (for state)

dtex, 2015-12-18:
Okay, I'm gonna leave out enableLow until we come across a board/shield that needs it.

dtex, 2015-12-18:
@Legendaire, take a look at #998 I don't have that sparkfun block so could you try it out?

````
var motorA = new five.Motor({
    pins: { pwm: 20, dir: 33, cdir: 46, enable: 47 }
});

motorA.fwd(255);
````

The motor is enabled by default. 

Available methods are:
```motor.disable()```, ```motor.enable()```

Available property is:
```motor.enabled```

rwaldron, 2015-12-18:
Mine arrived today, so i can test later when I get home

dtex, 2016-02-04:
#998 has merged, but I'd still like confirmation that it worked before I close this issue

rwaldron, 2016-02-04:
@dtex I actually made this and confirmed: https://github.com/rwaldron/johnny-five/blob/master/docs/motor-enable.md

closed 2016-03-17 by dtex