rwaldron/johnny-five #995: does this motor driver work with johnny?

nofear87, 2015-12-17:
https://www.pololu.com/product/2507

dtex, 2015-12-17:
Yes, use the 3-pin example:

http://johnny-five.io/examples/motor-3-pin/

The fritzing diagram shows a shield (which is not very helpful TBH) but the code will work.

If this answers your question, feel free to close the issue.

nofear87, 2015-12-18:
thanks, I will give it a try! Is it also possible to read out the encoders? 

I have used this motors: 

https://www.pololu.com/product/1447

to read the currents of the shield I only have to do an analog read?

dtex, 2016-01-23:
Hi @nofear87 sorry for the lack of response on this. We are considering adding support for encoders right into the motor class but that probably won't happen too soon. For now, you would have to write your own code to work with the two inputs. It's not something I've looked to deeply into... maybe someone else here knows more or has some examples?

dtex, 2016-03-17:
@nofear87 I'm assuming that the 3-pin example worked for you. Is it okay to close this ticket?

nofear87, 2016-04-27:
I need some time for testing. Then I will give some feedback! Thanks!

dtex, 2016-05-31:
@nofear87 I'm going to go ahead and close this. If for some reason the 3-pin example didn't work, feel free to re-open.

closed 2016-05-31 by dtex

cheerx, 2016-11-18:
@nofear87 Hello guys, did that 3-pins example worked for you? I got into trouble while working with "Pololu Dual VNH5019"

rwaldron, 2016-11-18:
@cheerx hope this helps: 

[image]

| Name      | Pin Name/Number | J5 Class & Pin | 
| --------- | --------------- | -------------- |
| M1INA     | 2               | Motor/Motors, `dir`   |
| M1INB     | 4               | Motor/Motors, `cdir`  |
| M1PWM     | 9               | Motor/Motors, `pwm`   |
| M2INA     | 7               | Motor/Motors, `dir`   |
| M2INB     | 8               | Motor/Motors, `cdir`  |
| M2PWM     | 10              | Motor/Motors, `pwm`   |
|           |                 |                      |
| M1EN/DIAG | 6               | Sensor/Sensors       |
| M2EN/DIAG | 12              | Sensor/Sensors       |
|           |                 |                      |
| M1CS      | A0              | Sensor/Sensors       |
| M2CS      | A1              | Sensor/Sensors       |
|           |                 |                      |

Using this shield to...

- Control motors:
  - With object params: 
    
    ```js
    const a = new five.Motor({
      pins: {
        pwm: 9,
        dir: 2,
        cdir: 4
      }
    });

    const b = new five.Motor({
      pins: {
        pwm: 10,
        dir: 7,
        cdir: 8
      }
    });

    // Or both in one object: 

    const motors = new five.Motors([
      { pins: { pwm: 9, dir: 2, cdir: 4 } },
      { pins: { pwm: 10, dir: 7, cdir: 8 } },
    ]);
    ```

  - With Array params: 

    ```js
    // PIN ORDER MATTERS: [pwm, dir, cdir]
    const a = new five.Motor([ 9, 2, 4 ]);
    const b = new five.Motor([ 10, 7, 8 ]);

    // Or both in one object: 

    const motors = new five.Motors([
      [ 9, 2, 4 ],
      [ 10, 7, 8 ],    
    ]);
    ```

- Check Motor Status:
  ```js
  var statuses = new five.Sensors([
    // M1EN/DIAG
    { pin: 6, type: "digital", id: "M1EN" },
    // M2EN/DIAG
    { pin: 12, type: "digital", id: "M2EN" },
  ]);

  statuses.on("change", target => {
    console.log(`${target.id}, on ${target.pin}, updated: ${target.value}`);
  });  
  ```

- Read Motor Current:
  (According to this: https://github.com/pololu/dual-vnh5019-motor-shield/blob/master/DualVNH5019MotorShield.cpp#L182-L194)
  ```js
  var currents = new five.Sensors([
    // M1CS
    { pin: "A0", id: "M1CS" },
    // M2CS
    { pin: "A1", id: "M2CS" },
  ]);

  currents.on("change", target => {
    // https://github.com/pololu/dual-vnh5019-motor-shield/blob/master/DualVNH5019MotorShield.cpp#L192
    const mA = target.value * 34;
    console.log(`${target.id}, on ${target.pin}: ${mA}mA`);
  });

  // or just one...

  var current = new five.Sensor("A0");

  current.on("change", () => {
    // https://github.com/pololu/dual-vnh5019-motor-shield/blob/master/DualVNH5019MotorShield.cpp#L192
    const mA = current.value * 34;
    console.log(`M1CS, on A0: ${mA}mA`);
  });  
  ```

nofear87, 2017-07-14:
@dtex are there any upddates regarding the implementation of the encoders?

dtex, 2017-07-14:
No updates. I have a project the requires encoders (using hall effect sensors), but it's still on my to-do list.

dtex, 2018-06-24:
No, I haven't worked on encoders. 

I don't recall the soft start issue, but we don't have a fix for that. It would be a nice addition to the motor class though. I'll add that to the [wish list](https://github.com/rwaldron/johnny-five/wiki/Requested-Features).