rwaldron/johnny-five #997: Could I use the board Serial to communicate to shield?

yuanhaoliang, 2015-12-18:
for example, I have to use the GSM Shield which communicates by the Serial port,such as pin 0/1(rx/tx),
however I didn't find the Serial lib in jonny-five.
did I miss something?

rwaldron, 2015-12-18:
Johnny-Five's IO protocol only recently gained serial/uart capabilities.

topkoa, 2015-12-19:
Are these capabilities documented yet?    I've been waiting for this as well..

dtex, 2015-12-19:
They are methods on the io class. Assuming you are using firmata to talk to an Arduino-like device over serial, the documentation is [here](https://github.com/jgautier/firmata#serial). Any other io classes that support serial should match this API.

rwaldron, 2015-12-19:
> Any other io classes that support serial should match this API.

None have implemented support yet, but that's ok. 

> the GSM Shield 

This is a very specific device and I'm not sure how it fits in Johnny-Five directly, if at all. Things like this, I've been building plugins for: 

- https://github.com/rwaldron/j5-ds-touch
- https://github.com/rwaldron/j5-sparkfun-weather-shield

These both use the same, **preferred**, pattern for implementing new component plugin support. To be clear, I'm not trying to discourage you from contributing to Johnny-Five—quite the opposite! This should get you started, as the basis for a GSM shield plugin: 

```js
var five = require("johnny-five");
var Emitter = require("events").EventEmitter;
var util = require("util");

var priv = new Map();

module.exports = function() {
  return (function() {

    function GSM(opts) {
      if (!(this instanceof GSM)) {
        return new GSM(opts);
      }

      five.Board.GSM.call(
        this, opts = five.Board.Options(opts)
      );

      var state = {
        portId: opts.portId || this.io.SERIAL_PORT_IDs.SW_SERIAL0,
        baud: opts.baud || 9600,
        // setup other private state here
      };

      priv.set(this, state);

      this.io.serialConfig({
        portId: state.portId, 
        baud: state.baud,
        rxPin: this.pins.rx,
        txPin: this.pins.tx,
      });

      this.io.serialRead(state.portId, function(bytes) {

        // read bytes and determine event types to emit
        // store received things in state object, use
        // for emitting events 
        // 
      }.bind(this));

      Object.defineProperties(this, {
        // Define any accessors here
      });
    }

    util.inherits(GSM, Emitter);

    GSM.prototype.write = function(bytes) {
      // implement
      var state = priv.get(this);

      this.io.serialWrite(state.portId, bytes);
    };

    GSM.prototype.sms = function(recipient, message) {

      // pre-process the data for an sms message
      // and send it on its way.

      // this.write(processed bytes);
    };
    return GSM;
  }());
};
```

This should be used like: 

```js
var five = require("johnny-five");
var GSM = require("j5-gsm-shield");

var board = new five.Board();

board.on("ready", function() {
  var gsm = new GSM({ 
    pins: { rx: 10, tx: 11}
  });  
  
  gsm.on("sms-receive", function(data) {
    console.log(data.sender, data.message);
  });

  gsm.sms("8005551234", "Hello!");
});
```

soundanalogous, 2015-12-19:
I wonder if we need a method to query which serial "ports" (`SW_SERIAL0`, `SW_SERIAL1`, etc) are in use in the case that multiple plugins attempt to default to the same port.

yuanhaoliang, 2016-01-19:
I have a try, thanks a lot.

closed 2018-08-28 by stale[bot]