rwaldron/johnny-five #1002: Multiple proximity sensors

Sparo, 2016-01-07:
Hi all,

I have 2 'HCSR04' and idea is to check two distances. 

I am using SeeedStudio Ultrasonic Range and grove base shield for Arduino Uno
http://www.seeedstudio.com/wiki/Base_shield_v2

My code is something like this 

```
...
var five = require('johnny-five');
var board = new five.Board();

var proximity1, proximity2;
var the_Array1 = [];
var the_Array2 = [];

board.on('ready', function() {
	...

	proximity1 = new five.Proximity({
		controller: 'HCSR04',
		pin: 'D3'
	});
	proximity2 = new five.Proximity({
		controller: 'HCSR04',
		pin: 'D4'
	});

	proximity1.on('data', function(data) {
		the_Array1.push(data);
		if (the_Array1.length === 50) {
			the_Array1.shift();
			// console.log(the_Array1);
		}
	});

	proximity2.on('data', function(data) {
		the_Array2.push(data);
		if (the_Array2.length === 50) {
			the_Array2.shift();
			// console.log(the_Array2);
		}
	});
});
```

so I have problem that looks like this:

```
/device/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1319
  if (this.pins[opts.pin].supportedModes.indexOf(PING_READ) === -1) {
                         ^

TypeError: Cannot read property 'supportedModes' of undefined
    at Board.pingRead (/device/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1319:26)
    at Proximity.<anonymous> (/device/node_modules/johnny-five/lib/proximity.js:176:19)
    at Proximity.Controllers.HCSR04.initialize.value (/device/node_modules/johnny-five/lib/proximity.js:182:9)
    at new Proximity (/device/node_modules/johnny-five/lib/proximity.js:440:10)
    at Board.<anonymous> (/device/hardware.js:19:15) // it is this line: --> proximity1 = new five.Proximity({
    at emitNone (events.js:67:13)
    at Board.emit (events.js:166:7)
    at doNTCallback0 (node.js:417:9)
    at process._tickDomainCallback (node.js:387:13)

```

so now I am not sure where to go next... I have not seen that somebody tried to create multiple proximity sensor on one board with johnny five lib ?

rwaldron, 2016-01-09:
Hi, I suspect there is no issue with have multiple hcsr04 devices. Two things:

- Currently, Johnny-Five does not accept digital pin names prefixed with "D" for Arduino board connections—this is because the convention is to use the pin name/number as it appears on the board, not as appears on arbitrary shields. I can try to make that work for shields we encounter, but there is no way to know how every shield labels its pins. I'd rather not special case pin normalization for every single shield, because I predict that could be very messy. I'm not against addressing cases that are probably pretty common (like this one). Let me think about that a little more. For now, just change those to the numbers `3` and `4`.
- Please make sure that you've followed the instructions for using hcsr04 with an Arduino board: http://johnny-five.io/api/proximity/#pingfirmata

It would be remiss of me if I didn't warn you that these sensors are process blocking on the board and may cause strange behavior if you're also attempting to do write operations to other components attached to the board.

Sparo, 2016-01-11:
I have tried just with 3 and 4 and it has worked without removing Grove base shield for Arduino Uno, so I guess it works like that also...

tnx for suggestion !

rwaldron, 2016-01-22:
No problem, glad to help!

closed 2016-01-22 by rwaldron