rwaldron/johnny-five #1012: Problem with Leds.pulse() using animation object and board.wait()

Dav-v, 2016-01-14:
Hi all,
It is a few days that I can't figure out why two animations that use more than one LED have a strange effect, as you can see in the video: if I try to run the code down here, both with 2 leds that with 6 leds, only the led on the last pin(11 or 5) continue with the second animation, while all other leds are simply stopped in the last state they were.
I'm running the code on a RaspberryPi server that controls an Arduino Uno board, and I tried to change everything: board, leds, server, cables, etc. but the problem is still here.
I also searched a lot the API documentation, but I didn't found anything.
Johnny five version:  0.9.17, installed from npm
Video: https://www.send18.com/?uid=ef81-3b1d7965 
password for the video: 2Nqn8OrB
Code:
```javascript
//with 6 leds
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

  var leds = new five.Leds([3,5,6,9,10,11]);
leds.pulse(
  {
    easing : "linear",
    cuePoints:  [0,0.5,1],
    keyframes: [0,255,0],
    duration: 1000
  }
)

board.wait(5000, function(){
leds.stop();
console.log("second animation started");
leds.pulse(    {
      easing : "linear",
      cuePoints:  [0,0.5,1],
      keyframes: [0,255,0],
      duration: 2000
    })
  });
});
```
```javascript
//with 2 leds, problem still here
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

  var leds = new five.Leds([3,5]);
leds.pulse(
  {
    easing : "linear",
    cuePoints:  [0,0.5,1],
    keyframes: [0,255,0],
    duration: 1000
  }
)

board.wait(5000, function(){
leds.stop();
console.log("second animation started");
leds.pulse(    {
      easing : "linear",
      cuePoints:  [0,0.5,1],
      keyframes: [0,255,0],
      duration: 2000
    })
  });
});
```
```javascript
//This code (inside the board.wait() function) has the same strange behavior

  leds[0].pulse(
    {
      easing : "linear",
      cuePoints:  [0,0.5,1],
      keyframes: [0,255,0],
      duration: 2000
    });
    leds[1].pulse(
      {
        easing : "linear",
        cuePoints:  [0,0.5,1],
        keyframes: [0,255,0],
        duration: 2000
      }
    );
    leds[2].pulse(
      {
        easing : "linear",
        cuePoints:  [0,0.5,1],
        keyframes: [0,255,0],
        duration: 2000
      }
    );
```

dtex, 2016-01-14:
HI @Dav-v I'll get this hardware set up later tonight and see if I can figure it out.

Dav-v, 2016-01-14:
Thank you very much @dtex

dtex, 2016-01-15:
Just wanted to let you know I was able to recreate this right away. Now I'm just peeling away the layers trying to find the problem... I'll keep you posted.

dtex, 2016-01-15:
So I've narrowed it down to Leds. Other collections (i.e. Servos) do not have the same problem. My brain is turning to mush on this and I'm not sure how long the fix will take to figure out. 

I do have a workaround if this is a "nodebots emergency"...

If you use [this branch](https://github.com/dtex/johnny-five/tree/Leds-pulse-fix) with the following code, it will work:

````js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var leds = new five.Leds([3,5]);

  var animation = new five.Animation(leds);

  animation.enqueue({
    keyFrames: [[{value: 0}, {value: 255}], [{value: 0}, {value: 255}]],
    metronomic: true,
    loop: true
  });

  board.wait(5000, function() {
    animation.stop();
    console.log("second animation started");

    animation.enqueue({
      keyFrames: [[{value: 0}, {value: 255}], [{value: 0}, {value: 255}]],
      metronomic: true,
      loop: true,
      duration: 2000
    });
  });
});
````

Your code should work (though it could be simplified) so I'm going to keep working on it. I just wanted to give you an option for moving forward.

Dav-v, 2016-01-15:
Thanks a lot, yes when I return home I'll try your code so I can proceed with my project (a node http server wich can control a lot of animations using leds).
Thanks again for the fast answer

dtex, 2016-01-15:
Okay, I found the cause of the error and also added some functionality for Led and Leds animation. I can't land this because I made a lot of changes and am sleepy so someone else should check it.

In fact CI just failed. I see I've introduced a new problem.

With the branch referenced above your code works, but it's leaky so it's not the final fix.

Oh, and earlier I mentioned that I think your code could be a simpler. The only difference between your two LED example and the code below is the easing function used:

````js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var leds = new five.Leds([3,5]);

  leds.pulse(500);

  board.wait(5000, function(){
    leds.pulse(1000);
  });

});
`````

By default, pulse will create a metronomic animation from 0 to 255 and will use ```inOutSine``` instead of ```linear```. I halved the durations because the duration used is one half of a metronomic cycle. If you really need linear easing, just do this:

````js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var leds = new five.Leds([3,5]);

  leds.pulse({duration: 500, easing: "linear"});

  board.wait(5000, function(){
    leds.pulse({duration: 1000 easing: "linear"});
  });

});
`````

I'll let you know when I have that leak solved.

dtex, 2016-01-15:
@rwaldron I have a question:

The tl;dr here is that when there are two or more animations they each spawn their own instance of temporal but those instances use a shared queue. If I don't clear the queue for an instance, temporal will just sit and spin forever. When I clear the queue it clears it for all animation/temporal instances. I only want to clear it for one animation/temporal instance but that's not an option.

It's nice to have a single queue and single call to processQueue each time through the event loop, but I think... and this is where I'm probably wrong... that we have a single queue and multiple calls to processQueue each pass through the event loop. If that's true, why not have multiple queues?

Also, is it possible to empty the queue of entries for a single task?

Dav-v, 2016-01-15:
I've tried your code and it works,thanks! The only thing is that it would be better if I could control each LED individually (also on speed and duration). I've tried with an array of Animation() objects, each one controlling one led, but again at the second time the animation is played there are some strange effects (even after calling animations[x].stop() for every led).

dtex, 2016-01-16:
Oh, you totally can, just use one Animation object:

````
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var leds = new five.Leds([3,5]);

  var animation = new five.Animation(leds);

  animation.enqueue({
    keyFrames: [
      [{value: 0}, {value: 255}], // These are the keyframes for the first led
      [{value: 255}, {value: 0}] // These are the keyframes for the second led
    ],
    metronomic: true,
    loop: true,
    duration: 2000
  });

});
````

Check out the [docs](http://johnny-five.io/api/animation/) for animation. You can do some crazy, complicated stuff.

Dav-v, 2016-01-29:
thanks. Sorry to bother you again, but I've recently tried to do a sort of "wave" effect with 6 leds, but since It seems that I can only use one cuePoints array for the whole animation object, I have to do something like this https://github.com/Dav-v/furry-octo-happiness/blob/master/animation_wave.js

dtex, 2016-02-01:
Sorry for the delay in response... honestly I think I would just write a custom algorithm for this, but I am also working to make multiple pulses not step on each other. I've got it figured out but we will need to make changes in one of Johnny-Five's dependencies. I'm going to go propose that change in an issue right now and I will link from here.

Dav-v, 2016-02-01:
Thanks @dtex  , so the problem is in the temporal dependence... in the meantime is it possible to change the method Animation.normalizedKeyframes to support more than one cuepoints array? 
Like this: https://github.com/Dav-v/furry-octo-happiness/blob/master/cuepoints.js
Maybe this would be helpful also for servos animations.

dtex, 2016-02-02:
Oh, that's interesting but I think the right thing to do is to fix our problem with multiple animations. That will allow you to have a different cuePoint array for each animation (and better yet, just use pulse like you tried in the first place). Once @rwaldron is able to come up for air and review my approach (or suggest something better), it shouldn't take long to add that feature to temporal.

I'll also ponder the multiple cuePoint arrays some more. The opportunity this affords for shorter keyFrame arrays is very, very appealing.

Dav-v, 2016-02-03:
Ok perfect. Approximately how long it would take?

dtex, 2016-02-05:
I got a thumbs up from @rwaldron so I will try and get that into temporal this weekend.

Dav-v, 2016-02-07:
Great! Keep me posted

Dav-v, 2016-02-17:
How is going @dtex  ? I also have a question: why in j5 there is a method like board.wait? Are there some differences from setTimeout?

dtex, 2016-02-17:
board.wait is a placeholder until something better comes along. Right now it is indeed just a wrapper for setTimeout, but I would use it. When that better thing comes along you won't have to edit any code.

I'm sorry I haven't got those changes into temporal yet. I haven't forgotten about you, just swamped at work. Hopefully that will be clearing up in the next few days.

Dav-v, 2016-02-17:
Thank you @dtex  , I've tried to see if I could do something, but temporal is a too complex lib for me :)  Don't worry, I asked you it because tomorrow I'll use led.animation() in a sort of light show, then I will use your fix.

dtex, 2016-03-17:
@Dav-v 

@rwaldron made a small change to Animation that may have fixed the problem. I can no longer recreate this. Are you still seeing it? (make sure you update to the latest Johnny-Five).

Dav-v, 2016-03-20:
Yeah, using the master branch it works! Thank you :)

closed 2016-03-20 by dtex