rwaldron/johnny-five #1017: Multiple boards with steppers: works one or two times then errors: "device or firmware error"

Snodvod, 2016-01-20:
I'm able to use one stepper motor, but when I try to connect another one and use it I get this timeout...

dtex, 2016-01-20:
Sounds like a brown out. Can we see a pic of the wiring?

Snodvod, 2016-01-20:
I can give a picture of it tomorrow, but is it possible that the steppers ask too much from the arduino pins? Maybe try to power it with an external 9V power supply...

dtex, 2016-01-20:
Oh yes, that's exactly what I was going to look for. Stepper motors will draw more current than your computer's USB port can supply. That will cause your Arduino to shut down. You will need to use external power for sure.

Snodvod, 2016-01-20:
I've been thinking about this problem the whole day and this solution suddenly popped into my head :D Thanks for the confirmation, I'll let you know if it worked. Thanks in advance

Snodvod, 2016-01-21:
So, with a proper power supply added I still ran into this problem... But it's not really the amount of steppers attached, cause when I connect 4 steppers to my arduino MEGA there's no problem, it's just when the 2 arduino's are connected...

dtex, 2016-01-22:
It still sounds like a brown out. Are both Arduinos on the same USB port?

Snodvod, 2016-01-22:
No on 2 different ports, already tried another arduino board to...

rwaldron, 2016-01-22:
@Snodvod can you provide a picture (or a few) of your project?

dtex, 2018-01-16:
@Snodvod I'm going to close this since we haven't heard back. Please feel free to re-open if necessary.

closed 2018-01-16 by dtex