rwaldron/johnny-five #1019: Add support for 5-wire servo

Resseguie, 2016-01-23:
In older electronics (particularly toys and RC vehicles), you often come across a "5-wire servo". It's actually a motor, gears, and potentiometer packaged together, much like a modern servo, but without the logic built in. When hacking on these devices and controlling them with Johnny-Five, it would be very handy to just treat it like any other servo (since functionally it is equivalent). But currently you have to re-write the logic each time to read the pot value in order to control precise position of the motor. Ideally I'd just like to use the existing Servo API.

I'm interested in this because I enjoy hacking vintage toys and I've come across this a few times. I've mentioned this in Gitter a few times (and some of you in person at hackathons), but I don't think we ever came to a consensus on the best way to handle it. Decided to create an issue to track things better.

Here is a sample of such a 5-wire servo from a Teddy Ruxpin toy that I'm working on:

[image]

discgolfer1138, 2016-05-06:
I too have been working on hacking a Teddy Ruxpin.  More specifically, I've been baffled by the servo wiring. I was trying my best to figure out what was in there without too much disassembly.

I've been able to get the eye servo and the upper and lower jaw servos to work by just applying power and then reversing it.  Nothing useful yet, just wanted to verify that they operate.

More importantly, have you figured out a reliable way to control that motor via Johnny-Five.

Since there is no consistency to the wire colors that Worlds of Wonder used.. I can only guess based on orientation.

Any additional info you have would be greatly appreciated :)

rwaldron, 2016-05-06:
Can you find me a link to buy one or two of these servos? Thanks

discgolfer1138, 2016-05-07:
I am going to replace the stock servos in my Teddy Ruxpin with 2 HiTec standard 3-wire servos. When I do, I will be careful in removing the original 5-wire servos and would be happy to send them to you.

rwaldron, 2016-05-09:
Whoa, that would be amazing. Keep me posted :)

discgolfer1138, 2016-05-09:
will do! making progress.. https://vimeo.com/16487186
The new servos should arrive Thursday, hoping to perform the surgery over the weekend.

dtex, 2017-08-17:
@yamanote1138 @Resseguie I'm on a mission to close all issues associated with servos. Do either of you have a Teddy Ruxpin servo you could part with for a couple of weeks?

yamanote1138, 2017-11-11:
Sorry, I ended up using an h-bridge to solve my problem, so the original motors are still in use.  On a related note, I now have an army of 3 Teddy Ruxpins and one Grubby (Teddy's creepy caterpillar friend).

here's a couple test videos:
[Teddy Test - Surely](https://vimeo.com/242348473)
[Grubby Test - Taunter](https://vimeo.com/242348635)

dtex, 2018-01-11:
Hi @Resseguie  ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for 5-wire motors there.

On a personal note, I really want to get this done. My friend @ryanjgill hacks a lot of toys so I'm pinging him to see if he has any motors (servos) that are wired like this.

closed 2018-01-11 by dtex

ryanjgill, 2018-04-03:
Sorry @dtex just saw this. I'm pretty sure I have 2 or 3 of these from old RC cars. I could get you a few if you still need them.

dtex, 2018-04-03:
I think it would be nice to add support or at least an example that mixes sensor and servo. 

I'll let you know if I'm able to make it out to a meetup anytime soon.

It doesn't look like we are doing NBD this year.

yamanote1138, 2020-01-05:
fwiw, I've recently put this project back on my workbench and am researching the servos once again. In case you haven't already seen it, I came across this [video](https://www.youtube.com/watch?v=sjR7502_tRI) which explains, in detail why Teddy's servos have 5 wires and where the on-board potentiometer fits into the equation. Just figured I'd share...

@Resseguie in your initial post, you mentioned working with these motors and, I think, figuring out how to handle/utilize the potentiometer. Any chance you have any snippets or a repo you could share?

in the meantime, my current project lives [here](https://github.com/yamanote1138/johnny-ruxpin) (warning: total WIP)