rwaldron/johnny-five #1020: PingFirmata not found

mulderp, 2016-01-24:
Hi,

I am playing with a small robot with HSR04 sensor.

Flashing the PingFirmata looked successfull, but I get an error when I start up the HSR04 example:

```
pm:vehicle pmu$ node proximity_hello.js
1453638857437 Device(s) /dev/cu.usbserial-AL01TQ76
1453638857447 Connected /dev/cu.usbserial-AL01TQ76
1453638860882 Repl Initialized
>> { supportedModes: [],
  mode: undefined,
  value: 0,
  report: 1,
  analogChannel: 127 }
/Users/pmu/projects/embedded/book/ch_9_making_robots_with_nodejs/vehicle/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1321
    throw new Error("Please upload PingFirmata to the board");
          ^
Error: Please upload PingFirmata to the board
    at Board.pingRead (/Users/pmu/projects/embedded/book/ch_9_making_robots_with_nodejs/vehicle/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1321:11)
    at Proximity.<anonymous> (/Users/pmu/projects/embedded/book/ch_9_making_robots_with_nodejs/vehicle/node_modules/johnny-five/lib/proximity.js:176:19)
```

mulderp, 2016-01-24:
Btw, I am trying to make this small robot work with Johnny-5: http://hackarobot.com/fabric-shield-monthly-giveaway/

Based on Arduino Nano, L293D to drive motors, and HC SR04 for proximity, some Arduino examples are here: https://github.com/ThomasCLee/funnyvale/tree/master/hackabot_nano ) The proximity sensor is wired to A0 and A1. Arduino Code for this is:

```
float getDistance() {
   digitalWrite(TRIGGER_PIN, LOW); 
   delayMicroseconds(2); 
   digitalWrite(TRIGGER_PIN, HIGH);
   delayMicroseconds(10); 
   digitalWrite(TRIGGER_PIN, LOW);
   return (  pulseIn(ECHO_PIN, HIGH) *0.017);
}
```
(/cc @ThomasCLee )

mulderp, 2016-01-24:
Can `pulseIn()` be written with JavaScript ?

ThomasCLee, 2016-01-24:
I am not familiar with JavaScript, but in theory, it should be doable.

You need to monitor the echo pin when it goes low, you store the current
time. It then goes into a while loop. When the echo pin goes high, it
exists. You compare the time when it exits verses the time you stored
earlier. The difference is the pulse width.

You may need to add a timeout check for the while loop just in case.

rwaldron, 2016-01-26:
HCSR04 won't work when attached to analog inputs, it needs to be attached to a digital input. 

> Can pulseIn() be written with JavaScript ?

No, but that's not really the issue here.

ThomasCLee, 2016-01-26:
A0 and A1 pins could be analog as well as digital inputs (not just analog).

rwaldron, 2016-01-26:
> A0 and A1 pins could be analog as well as digital inputs (not just analog).

Yes, I'm aware of this, but Johnny-Five has always used the user provided pin name value to infer user intent. Using the literal `"A0"` is assumed to mean: "I want this pin treated as an analog input pin, because that's a precedent that I learned from using Arduino". That being said, `Proximity` w/ HCSR04 can be updated to accept `"A0"` and do the right thing.

mulderp, 2016-01-26:
(Interesting discussion, I was not aware that A0/A1 are GPIOs as well, but indeed it makes sense. Arduino mentions it too "The analog input pins can be used as digital pins, referred to as A0, A1, etc." see https://www.arduino.cc/en/Reference/DigitalRead )

rwaldron, 2016-01-26:
@mulderp thanks for the report! v0.9.19 is on npm now. This was tested using: 

- HCSR04
- Parallax Ping

And an example + diagram here: https://github.com/rwaldron/johnny-five/blob/master/docs/proximity-hcsr04-analog.md

closed 2016-01-26 by rwaldron

mulderp, 2016-01-28:
Thanks!
Indeed, now it works great!
To make the example work with @ThomasCLee hackarobot, one need to add a jumper between A0 and A1. Next up, I'll check the combination with an MPU-6050 to read the bot's position. But was working earlier already.