rwaldron/johnny-five #1021: Proximity (Sharp GP2Y0A02YK0F) with Particle Photon reading is different after restart of node program

fabiantheblind, 2016-01-26:
__Board:__ Particle Photon
__Shield:__ none
__Hardware:__ Sharp GP2Y0A02YK0F distance sensor
__Johnny-Five Version:__ 0.9.17
__Node version:__ 0.12.9
__OS:__ OSX 10.11.3
__Expected result:__ same result after rerunning the program
__Current result:__ Values in first run are okay. sensor does not react to objects in second run
__Steps to reproduce:__ See the code and a fritzing sketch below
__video:__ https://vimeo.com/153079052

------

Hi. We already discussed this problem in the gitter chat. I'll add an issue here to have it persistent. When having a hardware setup like shown in the image, I get the following problem. 

I can run the attached code once and I get a reading that is slightly of but reacts to objects coming in range. Slightly of means the distance is a bit lower as the proximity module logs but it is consistent in its values. When stopping the program (by hitting CTRL+C twice) and rerunning it again. I get way to high values for the first seconds and then the value comes down and stays in a certain range without reacting to any thing coming in its range.  

When I restart the Photon board by pulling he RST pin low or unplugging the external power supply the reading is again as expected.  

### I already tried:  

1. Adding the capacitor for smoothing the result. Works with and without.
2. Running the sensor from the boards 3.3V. Does not work.
3. Exchange the board. Gives the same results.
4. Adding a diode on the analog input of the board. Works with and without.
5. Adding a voltage divider for lowering the voltage reaching the board. Works with and without.
6. Running the sensor from a different power source then the board. Does not work.
7. Setting the debug mode of the voodoo spark firmware to true. VS reports the analog values and they behave the same as the values logged by the node program.  
8. Running it without the Proximity module Just as analog pin. Same result.
9. particle-io directly not as a johnny-five plugin. same result.

### ToDos:  

1. Writing Arduino firmware and analyzing the result.  

### Workaround:

My current workaround is to add the pushbutton connected to the ground and just restart the board after killing the node program by pushing it once.  

### Links:  

- [Mastering Sharp Distance Sensors](https://github.com/FH-Potsdam/Knowledge-Exchange/blob/master/Sharp-Infrared-Sensor/2015-01-29_Sharp_Distance_Exchange_Final.pdf)  
- [GP2Y0A41SK0F Sharp Distance Measuring Sensor | Particle community](https://community.particle.io/t/gp2y0a41sk0f-sharp-distance-measuring-sensor/5303)

### Code:

__index.js__  

```js
var config = require('config');
var five = require("johnny-five");
var Particle = require("particle-io");
var keypress = require('keypress');

keypress(process.stdin);
// listen for the "keypress" event

var board = new five.Board({
  debug: true,
  io: new Particle({
    token: config.get('token'),
    deviceId: config.get('id')
  })
});
board.on("ready", function() {

  var proximity = new five.Proximity({
    controller: "GP2Y0A02YK0F",
    pin: "A1"
  });
  proximity.on("data", function() {
    console.log("Proximity: ");
    console.log("  cm  : ", this.cm);
    console.log("  in  : ", this.in);
    console.log("-----------------");
  });
  process.stdin.on('keypress', function(ch, key) {
    console.log('got "keypress"', key);
    if (key && key.ctrl && key.name == 'c') {
      proximity = null;
      process.exit();
    }
  });
});
```

__package.json__  

```json
{
  "name": "ir-sensor",
  "version": "1.0.0",
  "description": "",
  "engines": {
    "node": "<=v0.12.9"
  },
  "main": "index.js",
  "scripts": {
    "start": "node index.js",
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "keywords": [],
  "author": "Fabian 'fabiantheblind' Morón Zirfas <fabian.moron.zirfas@gmail.com> (http://fabianmoronzirfas.me)",
  "license": "ISC",
  "dependencies": {
    "config": "^1.17.1",
    "johnny-five": "^0.9.17",
    "keypress": "^0.2.1",
    "particle-io": "^0.12.0"
  }
}
```

__particle-io-bare-minimum.js__  
 

```js
var config   = require('config');
var Particle = require("particle-io");
var board = new Particle({
  token    : config.get('token'),
  deviceId : config.get('id')
});
board.on("ready", function() {
  console.log("CONNECTED");
  // Log all the readings for A1
  this.analogRead("A1", function(data) {
    console.log(data);
  });
});
```

__config/default.json__  

A file for storing the config data. Create it as config/defaul.json  

```json
{
  "id": "12345678900987654321",
  "token": "1234567899876543211234567iuztrewqertz",
}
```

### circuit:    

[image]

rwaldron, 2016-01-26:
Thanks for providing such a thorough bug report <3

> Writing Arduino firmware and analyzing the result.

I went ahead and did this, but not for an Arduino because there is no issue there. I wrote this: 

```cpp
#include "application.h"

SYSTEM_MODE(AUTOMATIC);

unsigned long lastms;
unsigned long nowms;
unsigned long sampleInterval = 100;

void setup() {
  pinMode(A0, INPUT);
}

void loop() {
  nowms = millis();
  if (nowms - lastms > sampleInterval) {
    lastms += sampleInterval;

    Serial.println(analogRead(A0));
  }
}
```

Note: particle photon firmware is at 0.4.9, which I had to download manually to the `particle-cli/updates` dir.

- Using USB power (https://www.sparkfun.com/products/12016 so same for both boards)...
- First I connected the sensor to an Arduino, with a multimeter connected to ground and the sensor's signal line. I placed a solid matte-white obstruction at 15cm, which produced a reading of 2.68V. I removed the obstruction and then returned it to 15cm, the reading changed but settled back to 2.66V
  - Connected the multimeter to the same ground and 5V that was previously used for the sensor, which produced a reading of 4.95V
- Next, I disconnected the usb from the Arduino and connected the Photon. 
  - Connected the multimeter to the photon's GND and VIN, which produced a reading of 4.68V
  - Connected the multimeter to GND and VIN, _after_ the 10µF cap, which produced a reading of 4.68V      
  - Connected the sensor GND and VCC to the 10µF node
  - Ensure the obstruction is set at 15cm
  - Connected the multimeter to the photon's GND and VIN (again), which produced a reading of 4.63V (note the 0.05V loss, this is consistent)
  - Connected the sensor Signal to the multimeter, which produced a reading of 2.68V
  - Connected the sensor Signal to appropriate node (sensor Vout signal source) 1K resistor
  - Ensure the obstruction is set at 15cm (again)
  - Connected the multimeter to the voltage divided Vout, which produced a reading of 1.33\*

\* I'm not sure how to resolve this discrepancy, there may be an obvious solution, eg adjusting the math used to determine a distance... but I don't want to rush into that. 

---------
w/r to this: 

> I can run the attached code once and I get a reading that is slightly of but reacts to objects coming in range. Slightly of means the distance is a bit lower as the proximity module logs but it is [...]

I could not reproduce the spike in readings (using the matte-white obstruction at 15cm—unmoved.)

> Adding the capacitor for smoothing the result. Works with and without.

Confirmed.

> Running the sensor from the boards 3.3V. Does not work.

Confirmed.

> Exchange the board. Gives the same results.

Confirmed (with an alternate Photon)

> Adding a diode on the analog input of the board. Works with and without.

Did not attempt

> Adding a voltage divider for lowering the voltage reaching the board. Works with and without.

Confirmed.

> Running the sensor from a different power source then the board. Does not work.

Did not attempt

> Setting the debug mode of the voodoo spark firmware to true. VS reports the analog values and they behave the same as the values logged by the node program.
> Running it without the Proximity module Just as analog pin. Same result.
> particle-io directly not as a johnny-five plugin. same result.

Confirmed all. 

-------------

I will continue working on this a bit later, I have to change gears for the rest of the day.

fabiantheblind, 2016-01-27:
Another workaround. When we register another Sharp sensor on another pin without adding anything to it, the readings are working after a restart of the program. We don't use it we just have it in our code. 🤔

    var proximity2 = new five.Proximity({
        controller: "GP2Y0A02YK0F",
        pin: "A2"
    });

rwaldron, 2016-01-27:
I'm not able to reproduce the issue where no readings are delivered, but the voltage divider changes the math and invalidates the formula being used. I'm still trying to figure out how to make so that's not an impediment, whether it's customizable or not, I'm not sure yet...

fabiantheblind, 2016-01-28:
> I'm not able to reproduce the issue where no readings are delivered
What do you mean by that?  
About the voltage divider. Yes I know it renders the calculated output useless. Maybe the Proximity module can have a option to provide a own formula.  

Did you try to just add another proximity object? It is pretty wired that just adding another object in the code makes the sensor reading work again.

rwaldron, 2016-01-28:
> Did you try to just add another proximity object?

I will try this.

> It is pretty weird that just adding another object in the code makes the sensor reading work again.

I've seen before in the Intel Galileo and Galileo 2 :\

fitzgr, 2017-03-29:
Can you supply the voltage of the 10uf capacitor and the specs on the diode please.

Kindest regards,
Grant Fitz

fabianmoronzirfas, 2017-03-29:
hi @fitzgr the project was finished a year ago and is already dismantled. Sorry

fitzgr, 2017-03-29:
Woops sorry for the repost.

dtex, 2018-01-05:
It sounds like we should close the issue since @fabianmoronzirfas has dismantled the project. Anyone feel free to re-open if there is a way we can make progress.

closed 2018-01-05 by dtex