rwaldron/johnny-five #1022: five.Led.Digits / edison-io / HT16K33 / 14-segment

rpsmithii, 2016-01-27:
Hello Rick - having a great time with your Johnny Five modules on Intel Edisons - I love the power.  Got it all running great with the seven segment backpacks -  also have a bunch of 14-segment ones but they do not work.  Below is the code I use to drive one 7-seg for time and 1 for temp.

As I said I have a bunch of these - if you had any ideas about updating the code for the 14-seg packs I would be happy to test for you.

Here is my code below - with the 7-segs it works - swapping out the the time 7-seg pack for a 14-seg - at 22:31 - I get the following

This might help...

LED 1: 2 - middle right segment not lit 
LED 2: 2 - middle right segment not lit 
LED 3: Blinking 8
LED 4: 3 - middle right segment not lit.

Thanks very much.

------snip snip -------
var moment = require("moment");
var five = require("johnny-five");
var Edison = require("edison-io");
var weather = require('openweathermap');
var board = new five.Board({io: new Edison()});
var txoffset = -5;
var minutes = 2, wxCheck = minutes * 60 * 1000;
var wx_now = {APPID: "3fa4460cd24f55062be92d87ff0be09f", units:'imperial',lang:'en',id:'4180439'}
var temp = 95;

board.on("ready", function() {
    var HHmm =    new five.Led.Digits({
        addresses: [0x75], 
        controller: "HT16K33",
    });
    var toggle = 0;
    var tempdsp = new five.Led.Digits({
        addresses: [0x77], 
        controller: "HT16K33",
    });
    weather.now(wx_now,function(err,data)
        {temp=Math.round(data.main.temp); 
        tempdsp.print(temp + "F");
        console.log("Temperature " + temp);
    });
    setInterval(function() {
        weather.now(wx_now,function(err,data)
            {temp=Math.round(data.main.temp); 
            tempdsp.print(temp + "F");
        });
    } , wxCheck);
    setInterval(function() {
        HHmm.print(time(toggle ^= 1));
    }, 1000);
});

function time(showColon) {
  var display = "    " + moment().utcOffset(txoffset).format(
    showColon ? "HH:mm" : "HH mm"
  );
  return display.slice(-5);
}

rwaldron, 2016-01-27:
The LED components: 

- `Led.Digits`
- `Led.Matrix`

Are currently being completely overhauled to eliminate constraints imposed by legacy porting efforts. I'll keep you posted here.

rwaldron, 2016-01-27:
I will also attempt to reproduce this in the meantime—maybe we can get you resolved before that work is complete.

rpsmithii, 2016-01-27:
Cool - happy to help anyway I can sir.

rwaldron, 2016-03-18:
Can you link me to the exact component that you're using? Thanks!!

rpsmithii, 2016-03-18:
Sure. Here you go sir.  

https://www.adafruit.com/products/1911

Appreciate you cycling back.  Have a great weather display working with the numerical version.  So ready to test your code at any time.

rwaldron, 2016-03-18:
Ah, ok. So I can confirm that these are indeed not yet supported. (Also, I'm really sorry for the delay in getting back to you—life is busy :\ ) Presently, only the 7-segment 4-digit HT16K33 units are supported

rpsmithii, 2016-03-18:
Sure I inferred that Rick.  I see you work pretty closely with Adafruit - do yo have a pipeline of these?  If I had one sent to you would that help?  You can’t expect to finance this all yourself, right?

rwaldron, 2016-03-21:
> Sure I inferred that Rick. 

Sorry, I'm not trying to be deliberately dense... balancing a lot of plates, sometimes I post because I don't want reporters to think I've forgotten about them. Sorry about that :\

>  I see you work pretty closely with Adafruit - do yo have a pipeline of these? 

I actually don't work with them at all, I don't think they've ever even acknowledged that this project exists. I have to buy these just like everyone else.

rpsmithii, 2016-04-09:
Sorry to take so long to get back to you Rick.  I have no affiliation with Adafruit at all except but a lot from them.

I have a few of these displays lying around  - very happy to send one to you - to keep

And  - since you seem to be much more embedded in the industry than I am - if you do have me send you the display - would you possibly have a pipeline into a charity or program that could use alot of surplus Arduino type gear?  If you do I could send it all to you along with the display to dispose of how you see fit.  I'm just a working stiff messes with this stuff to wind down.  Not involved locally at all.

Thanks 
Bob.

rwaldron, 2016-04-09:
Thanks for the offer—I've actually purchased one of these and it was delivered a few days ago, I just haven't had time to work on it (busy preparing for all the Tessel 2 boards to ship)

rpsmithii, 2016-04-09:
OK - will stand by when you get the time to test Rick.

Also any ideas about all of the surplus components I have?

dtex, 2018-01-11:
Hi @bethanysciences ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the HT16K33 Backpack there.

closed 2018-01-11 by dtex

monteslu, 2018-09-09:
Old issue, but the 14 segment display has been supported for a while now with this library:  https://github.com/louiemontes/node-led