rwaldron/johnny-five #1025: Having an error with johnny-five/lib/board.js

vumanhha, 2016-01-29:
Hi, I'm just new with johnny-five modules.
I'm follow to this instruction [https://github.com/rwaldron/johnny-five/blob/master/docs/led-blink.md](url) then i met this err when run js file.

> root@galileo:~/Workspace# node test.js
/home/root/node_modules/johnny-five/lib/board.js:91
        if (!rport.test(val.comName)) {
                           ^
TypeError: Cannot read property 'comName' of undefined
    at /home/root/node_modules/johnny-five/lib/board.js:91:28
    at Array.filter (native)
    at Object.Call (/home/root/node_modules/johnny-five/node_modules/es6-shim/es6-shim.js:278:14)
    at Array.filter (/home/root/node_modules/johnny-five/node_modules/es6-shim/es6-shim.js:1308:17)
    at Board.<anonymous> (/home/root/node_modules/johnny-five/lib/board.js:86:22)
    at /home/root/node_modules/johnny-five/node_modules/serialport/node_modules/async/lib/async.js:254:17
    at done (/home/root/node_modules/johnny-five/node_modules/serialport/node_modules/async/lib/async.js:129:15)
    at /home/root/node_modules/johnny-five/node_modules/serialport/node_modules/async/lib/async.js:32:16
    at /home/root/node_modules/johnny-five/node_modules/serialport/node_modules/async/lib/async.js:251:21
    at /home/root/node_modules/johnny-five/node_modules/serialport/serialport.js:589:15

I'm using: 

> Galileo Gen 1 
Linux galileo 3.8.7-yocto-standard #1 Tue Jun 9 22:07:14 GMT 2015 i586 GNU/Linux
version: 201506210418

I don't know how to fix this issue, hope you got an answer for this. 
Thank you very much.

Resseguie, 2016-01-29:
_I updated your post with a correct link to the led-blink example._

Resseguie, 2016-01-29:
@rwaldron why does the "getting started with Galileo" page not show the use of galileo-io? Am I missing something?
https://github.com/rwaldron/johnny-five/wiki/Getting-started-with-Johnny-Five-and-the-Intel-Galileo-gen2

vumanhha, 2016-01-29:
@Resseguie  Thanks for update.
I can't find page that show how to use galileo-io without johnny-five help. Example i want to read button state, i try this

`var board = new Galileo();`
`var buttonPin = 2;`
`var ledPin = 13;`
`var x = 0;`
`board.on("ready", fuction(){`
```
board.pinMode(ledPin, this.MODES.OUTPUT);
board.pinMode(buttonPin, this.MODES.INPUT);
var x = board.digitalRead(buttonPin);
if( x == 1){ board.digitalWrite(ledPin, 1);}
else { board.digitalWrite(ledPin, 0); }
```
`});`

The program still run but its auto turn on led although I try to hit, hold or release button the led won't turn off.

Resseguie, 2016-01-29:
@vumanhha I was just working on an updated sample. Just a second. You should be able to just specify the IO-Plugin you need. In older versions we attempted to auto-detect the platform, but that has been removed since some of those examples were updated. That's probably the root of the problem. 

We're working to update the examples as well.

Resseguie, 2016-01-29:
Try it like this, where the only difference is you pass in a reference to a galileo-io instance to the five.Board constructor.

``` js
var five = require("johnny-five");
var Galileo = require("galileo-io");
var board = new five.Board({
  io: new Galileo()
});

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink(500);
});
```

Resseguie, 2016-01-29:
I don't have that hardware to test myself, but perhaps someone else can jump in if that's not sufficient to get you going.

Resseguie, 2016-01-29:
FYI I update the sample code in that wiki link to include the use of `galileo-io`.
This blog post by @rwaldron goes into more details about the differences between a standard Arduino and the Galileo (and why the IO-Plugin is used).
https://bocoup.com/weblog/intel-galileo-javascript-nodejs

rwaldron, 2016-01-30:
> I can't find page that show how to use galileo-io without johnny-five help. Example i want to read button state

It's not recommended. Why do you want to do it without Johnny-Five? Galileo-IO is going to inundate your program with useless digitalRead information, whereas Johnny-Five will process that into actual `Button` information: 

```js
var five = require("johnny-five");
var Galileo = require("galileo-io");
var board = new five.Board({
  io: new Galileo()
});

board.on("ready", function() {
  var button = new five.Button(2);
  var led = new five.Led(13);

  button.on("press", function() {
    led.on();
  });

  button.on("release", function() {
    led.off();
  });
});

vumanhha, 2016-01-30:
@rwaldron @Resseguie 
Thanks for help me understand.
I guess that the reason why my program get an err when run follow code `led-blink` example is that code use on Arduino UNO. My galileo must add few more line
`var Galileo = require("galileo-io");`
`var board = new five.Board({`
 ` io: new Galileo()`
`});`

I've test and it run fine :)

Resseguie, 2016-01-30:
@vumanhha That's exactly it. Glad you got it working. We're always working on improving the documentation, so thanks for the input.

If that got everything taken care of for you, can we close this issue?

Note, if you have general questions while learning, feel free to join us on the Johnny-Five gitter chat channel. Lots of folks are there and happy to help.
https://gitter.im/rwaldron/johnny-five

vumanhha, 2016-01-30:
@Resseguie Thanks, you can close this issue.

closed 2016-01-30 by Resseguie