rwaldron/johnny-five #1027: Add own sensors/plug-ins

terkelg, 2016-02-02:
First of all thanks a lot for the amazing framework. I'm trying to add support for a new sensor, TH02 Temperature + Humidity Sensor. It's already ported to JS _(https://github.com/sensamo/th02js/blob/master/index.js)_ and I would like to add this to Johnny-five as a new component. I tried to look at the docs, but the only thing I can find is this: _https://github.com/rwaldron/johnny-five/blob/master/docs/plugin.md_. However, I'm not quit sure how this works, and I would love a little more explanation. Hope you can help me. Thanks in advance

rwaldron, 2016-02-02:
Exciting! New component controllers are always accepted :)

Since this is a multiple-sensor component, you'll need to create 4 things. I will use the HTU21D as an example since it's very similar to the TH02: 

1. A driver in `lib/imu.js` in the `Drivers` object: https://github.com/rwaldron/johnny-five/blob/master/lib/imu.js#L23-L74
2. A controller in `lib/imu.js` in the `Controllers` object: https://github.com/rwaldron/johnny-five/blob/master/lib/imu.js#L1054-L1095
3. A controller in `lib/thermometer.js`: https://github.com/rwaldron/johnny-five/blob/master/lib/thermometer.js#L297-L314
4. A controller in `lib/hygrometer.js`: https://github.com/rwaldron/johnny-five/blob/master/lib/hygrometer.js#L8-L25

A couple things I need to point out before you port your code: 

1. The IO is async, all the time, so this is disallowed: https://github.com/sensamo/th02js/blob/master/index.js#L51
  - There's many examples of status checks in the existing sensor sources
2. The register for the TH02 appears use the same two locations for both the temp and humidity, so you'll need to "cycle" the reads, back and forth. Here's an example of how the BMP180 is cycled: https://github.com/rwaldron/johnny-five/blob/master/lib/imu.js#L708-L786

I've just ordered one of these from Seeed, ping me when you're ready for review!

dtex, 2016-03-27:
It looks like support for the TH02 landed in https://github.com/rwaldron/johnny-five/commit/2401ca5fec1fbf885532cc25ae6b46123781c3d0 Can we close this issue?

terkelg, 2016-03-27:
Sorry I didn't reply. I tried to implement this myself, but without luck. I'm glad to see support for TH02 landed in 2401ca5. I'm also super exited to check out the source code – see what I did wrong and hopefully learn to how to do this myself with other sensors in the future. Once again, thanks for all your great work.

rwaldron, 2016-03-28:
@terkelg the previous implementation was not quite right. I've pushed some changes today that more appropriately respect the time-to-convert by awaiting a status message.

rwaldron, 2016-03-28:
But yeah, we can close this :)

closed 2016-03-28 by rwaldron