rwaldron/johnny-five #1029: DS18B20 - timeout

unseen1980, 2016-02-03:
Hi,

I am trying to read a DS18B20 temp sensor and I am getting the error below.
[image]

I used the example code from [here](https://github.com/rwaldron/johnny-five/blob/master/docs/temperature-ds18b20.md)

With the code from [here](http://bildr.org/2011/07/ds18b20-arduino/) it works fine so not a connection issue. Any ideas on what I am doing wrong?

Thanks

rwaldron, 2016-02-03:
Did you flash the board with ConfigurableFirmata?

soundanalogous, 2016-02-04:
ConfigurableFirmata does not support this sensor anymore. I plan to add support back at some point if I ever decided on the Firmata device API.

soundanalogous, 2016-02-04:
Nevermind... that's the OneWire device which is still supported. I was thinking of the DHT11 sensor which is not.

unseen1980, 2016-02-04:
Hi,

Yeap that was the problem. I thought the message had the opposite meaning.
Works fine now.

Thanks for your time.

closed 2016-02-04 by unseen1980