rwaldron/johnny-five #1033: Error device or firmware

twilchen, 2016-02-12:
Arduino Uno rev 3 with StandardFirmata (Firmata v2.5.1)

I get this error when i run the "Hallo world" blink led example.

1455232238928 Device(s) /dev/cu.usbmodem1411
1455232238953 Connected /dev/cu.usbmodem1411
1455232248957 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:146
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:144:17)
    at Board.log (/Users/wilchen/Sites/personal/arduino/node_modules/johnny-five/lib/board.js:636:8)
    at Board.(anonymous function) [as error] (/Users/wilchen/Sites/personal/arduino/node_modules/johnny-five/lib/board.js:647:14)
    at Board.<anonymous> (/Users/wilchen/Sites/personal/arduino/node_modules/johnny-five/lib/board.js:414:14)
    at Timer.listOnTimeout (timers.js:92:15)

Tried on mac and windows.

rwaldron, 2016-02-12:
I can't reproduce this with any Arduino UNOs that have :(

Can you give more information? Is your USB properly connected? Is it a non-arduino brand board? What else can you tell me?

Dav-v, 2016-02-12:
Hi, something similar also happens to me when I connect my Arduino to my powered usb hub(bought on amazon [here] ( http://www.amazon.com/AmazonBasics-Port-USB-Power-Adapter/dp/B00DQFGJR4/ref=sr_1_5?ie=UTF8&qid=1455314050&sr=8-5&keywords=powered+usb+hub ) ). If I plug in  the Arduino before starting the program, everything works, but if i plug-in the Arduino on a USB port while the program is already started (when on the screen is written "Board Looking for connected device") , it emit an error.
```
1455313893922 Board Looking for connected device  
1455313896777 Device(s) /dev/ttyUSB0  
1455313896786 Connected /dev/ttyUSB0  
undefined
undefined
undefined
1455313906792 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting  
events.js:146
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:144:17)
    at Board.log (/home/davide/node_modules/johnny-five/lib/board.js:633:8)
    at Board.(anonymous function) [as error] (/home/davide/node_modules/johnny-five/lib/board.js:644:14)
    at Board.<anonymous> (/home/davide/node_modules/johnny-five/lib/board.js:414:14)
    at Timer.listOnTimeout (timers.js:92:15)
```
I don't know if this problem is similar to the one encountered by @twilchen

twilchen, 2016-02-13:
It's a official board. I don't think there is a problem with the usb, I can upload Arduino sketches without any problem. I have also tried to select the port `var board = new five.Board({ port: "COM4" });` the arduino ide says it is connected to 'COM4' (this is on windows)

rwaldron, 2016-02-15:
> Hi, something similar also happens to me when I connect my Arduino to my powered usb hub(bought on amazon here )

It's possible that this is the problem, Johnny-Five implements an available port detection heuristic based on a few well-known port naming patterns. To do this, Johnny-Five will get a list of active serial ports and analyze the names/paths. It's possible that the listing operation (which is provided via node-serialport) isn't finding those active ports. 

Please post the result of running this: 

```js
var serialport = require("serialport");
var SerialPort = serialport.SerialPort; // localize object constructor

serialport.list(function(err, ports) {
    console.log(ports);

});
```

(be sure to `npm install serialport` first)

Dav-v, 2016-02-15:
Thanks @rwaldron , I've attached the output. I'm on Ubuntu.
[output.txt](https://github.com/rwaldron/johnny-five/files/131493/output.txt)

rwaldron, 2016-02-15:
@Dav-v yep, as I expected: none of those matches the common patterns for detecting an Arduino compatible device—in fact, there is nothing about them that can be correlated :\ What is displayed in the Arduino IDE's Tools > Port ?

Dav-v, 2016-02-15:
The port is "dev/tty/USB0", but there is also the port "/dev/tty/S0". Also if I run the script above with the Arduino plugged in the usb hub , I get this: 
[outputWithArduino.txt](https://github.com/rwaldron/johnny-five/files/131521/outputWithArduino.txt)

rwaldron, 2016-02-15:
Oh, I thought the Arduino was plugged in before :P

Can you try explicitly specifying the port as: 

```
var board = new five.Board({ port: "/dev/tty/USB0" });
```

Dav-v, 2016-02-15:
Yeah sorry ;) So I've tried, but the error is the same, also with an explicit port path: only the first time the program ran, then I changed the port and nothing works anymore.If I try to run the serial port script, it still tell me that the device is connected on USB0. Maybe tomorrow I'll do better tests to try to figure out what's happening.

rwaldron, 2016-02-15:
> then I changed the port and nothing works anymore.

What do you mean? If you unplugged the device from the port and then plugged it into another port, the port path will change. If it works with the path you explicitly set, then you need to change the explicit path every time you move it the board to another port—unless I'm misunderstanding something?

> If I try to run the serial port script, it still tell me that the device is connected on USB0

You're _absolutely_ positive of this?

ajfisher, 2016-02-16:
Just another point in here as an ubuntu user. It's not an arduino Uno R3 if it's presenting as `/dev/ttyUSB0` - on ubuntu, an UNO should present at `/dev/tty/ACM0` which again should be autodetected.

`/dev/tty/S0` through to `/dev/tty/S31` are the standard serial consoles and used for various parts of the OS.

If you're not seeing `/dev/tty/ACM0` or `/dev/tty/USB0` then do `lsusb` and you should see the devices connected with appropriate vendor IDs.

It's entirely possible you need to update some udev rules or groups to have access to the device. Ubuntu can be a little bit flakey with dropping connections to a serial device I've seen over the years. Best thing is to remove the device physically and ensure there is nothing presenting in /dev/tty* and then connect and do `ls /dev/tty*` again and you'll see it connect.

Failing that `dmesg` just after you plug it in and see if it's trying to connect and something is failing. Ubuntu tends to be the most "relaxed" when it comes to serial devices so even clones tend to work fine.

Dav-v, 2016-02-16:
> What do you mean? If you unplugged the device from the port and then plugged it into another port, the port path will change. If it works with the path you explicitly set, then you need to change [...]
Yeah, but even if I change the USB port, as you can see in the video, node serialport tells me that the Arduino  is still connected to USB0.
Thanks @ajfisher : It seems that everything is connected, because when I connect the board appears ttyUSB0 in /dev, also in the Arduino IDE everything works fine.
Video : https://www.dropbox.com/s/hz5d70fd705qjo8/VID_20160216_211515.mp4?dl=0

rwaldron, 2016-03-06:
Thanks everyone for working to get this resolved!

closed 2016-03-06 by rwaldron

wbashir, 2016-04-13:
Same issue.. Can someone offer any assistance still unable to follow. Output of ports 

````
[ { comName: '/dev/cu.Bluetooth-Incoming-Port',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: '/dev/cu.usbserial-DN0122ND',
    manufacturer: 'FTDI',
    serialNumber: 'DN0122ND',
    pnpId: undefined,
    locationId: '0xfa130000',
    vendorId: '0x0403',
    productId: '0x6015' } ]

````

ajfisher, 2016-04-13:
@wbashir as you're using an FTDI device, are you certain you have the FTDI driver installed on your system? Best thing to try is to program an arduino using arduino IDE if you're in doubt and that may help debug where the problem lies.

rwaldron, 2016-04-13:
@wbashir what kind of device is that?

wbashir, 2016-04-13:
@ajfisher I made sure to install the FTDI driver for my Mac OS X version before using this node module and I am able to successfully program with the `Arduino IDE` with no issues.

@rwaldron The device I am using is the 'Arduino UNO' from SparkFun redboard as part of the kit.

rwaldron, 2016-04-13:
Can you double check that the board has StandardFirmata or StandardFirmataPlus uploaded to it? Best way is to just upload either of them, even if you're certain you've done it previously. Let me know how that goes

wbashir, 2016-04-16:
@rwaldron !!!! What a guy. Thanks

rwaldron, 2016-04-16:
That worked?

wbashir, 2016-04-17:
@rwaldron on a fresh install, different mac os x. I installed `StandardFirmata` on my redboard and everything worked as expected.

rwaldron, 2016-04-18:
@wbashir ah—excellent :)

rwaldron, 2016-04-18:
@wbashir also, please join us in http://gitter.im/rwaldron/johnny-five (very welcoming, new-user friendly crowd)

markvdesign, 2017-03-15:
I know this is a very old thread but I had the same issue and I ended up resolving it by increasing the timeout property when initialising a new board.

**From this:**
`const board = new five.Board({
            timeout: 30
 });`

**To this:**
`const board = new five.Board({
            timeout: 3600
});`

The documentation states that timeout is in seconds but I was getting this error immediately when initialising the board. Increasing the timeout made all the difference.

studentIvan, 2017-07-17:
If somebody will stucked with this problem - try to do this:

1. go to arduino ide
2. select the board/port
3. open examples -> firmata -> standart firmata
4. upload to arduino

it solved the problem for me

rwaldron, 2017-07-20:
Yep, that always works! https://github.com/rwaldron/johnny-five#setup-and-assemble-arduino

rishirebel, 2017-09-02:
I was facing the same problem. Was getting this error 
`1504331572007 Available /dev/cu.usbmodem1411
1504331572025 Connected /dev/cu.usbmodem1411
1504331582027 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then ru
n your program again.
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])

    at Board.emit (events.js:163:17)
    at Board.log (/Users/rishirebel/Desktop/testter/node_modules/johnny-five/lib/board.js:635:8)
    at Board.(anonymous function) [as error] (/Users/rishirebel/Desktop/testter/node_modules/johnny-five/lib/board.js:646:14)
    at Board.<anonymous> (/Users/rishirebel/Desktop/testter/node_modules/johnny-five/lib/board.js:390:14)
    at ontimeout (timers.js:386:14)
    at tryOnTimeout (timers.js:250:5)
    at Timer.listOnTimeout (timers.js:214:5)`

Later on, I just uploaded [https://github.com/firmata/arduino/blob/master/examples/StandardFirmataPlus/StandardFirmataPlus.ino](url) firmware and everything worked fine.
`1504345526832 Connected /dev/cu.usbmodem1411
1504345528532 Repl Initialized
>>
(To exit, press ^C again or type .exit)
>>
1504345534218 Board Closing.`

Doneezy, 2017-10-21:
hi guys mine says access to COM3 is denied. i uploaded standardfirmata and repeatedly reinstalled serial port...got it working first time...but now.nothing..

informaticosgb, 2018-03-14:
@studentIvan thanks !

Ashwin-Shenoy-NITK, 2018-12-27:
Hi guys ! I tried the blinking problem but I am getting the following error  : 
```
1545920128406 Available COM3
1545920128425 Connected COM3
1545920138428 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:173
    throw err; // Unhandled 'error' event
    ^

Error [ERR_UNHANDLED_ERROR]: Unhandled error. ([object Object])
    at Board.emit (events.js:171:17)
    at Board.EventEmitter.emit (domain.js:442:20)
    at Board.log (E:\Nodetronics\node_modules\johnny-five\lib\board.js:648:8)
    at Board.(anonymous function) [as error] (E:\Nodetronics\node_modules\johnny-five\lib\board.js:659:14)
    at Board.<anonymous> (E:\Nodetronics\node_modules\johnny-five\lib\board.js:395:14)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```
I tried running the following program after uploading the standard firmata and standard firmata plus turn by turn but nothing worked :
```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink(500);
});
```
I am running the code in a windows 7 machine 64 bit OS. I also installed serialport.

dtex, 2018-12-27:
The Arduino IDE does not do a good job of alerting users when a sketch upload fails to build and upload. When you upload, is there anything weird being output in the status area at the bottom of the Arduino IDE?

muditgoel01, 2021-02-18:
Install firmata-party:
`npm i firmata-party `

Then run this command from terminal to reset the arduino:
`firmata-party uno --debug`

Thanks to @AndreyMartinez
https://www.youtube.com/watch?v=yxl3D5hjWAs