rwaldron/johnny-five #1035: Piezo - play sound

b00682613, 2016-02-13:
Is there a way to turn the buzzer on and off?

Like the five.led?
```javascript
piezo.toggle(); // buzzer turned on 
```

Here's my code. After the song is done playing the buzzer bleeps on and off continuously.
```javascript
var five = require("johnny-five"),
  board = new five.Board();

board.on("ready", function() {
  var piezo = new five.Piezo(3);

  piezo.play({
    song: "C D F D A - A A A A G G G G - - C D F D G - G G G G F F F F - -",
    beats: 1 / 4,
    tempo: 100
  });
});
```

rwaldron, 2016-02-15:
@b00682613 have you tried the `noTone` method? It's also aliased as `off`

b00682613, 2016-02-16:
@rwaldron I have got it working now. I'm using it like this:

```javascript
  var piezo = new five.Piezo(3)

  piezo.tone(2, 1)
  
  // Turn buzzer off after 3 seconds
  setTimeout(function () {
    piezo.noTone()
  }, 3000)
```

closed 2016-02-16 by b00682613