rwaldron/johnny-five #1036: HC-05 Bluetooth Port Number Issue

b00682613, 2016-02-13:
```javascript
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var led = new five.Led(13);

  this.repl.inject({
    led: led
  });
});
```
I have paired the HC-05 module with my PC on Windows. Under Bluetooth settings the COM Ports are:

Port | Direction | Name
------------ | ------------- | -------------
COM4 | Incoming | HC-05
COM5 | Outgoing      | HC-05 'Dev B'

```javascript
1455386996814 Device(s) COM4,COM5
1455386996829 Connected COM4
1455387006840 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you ve properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seco
nds for complete reset, then run your program again.
events.js:159
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
```

rwaldron, 2016-02-15:
Have you read through the HC-05 guide? https://github.com/rwaldron/johnny-five/wiki/Getting-Started-with-Johnny-Five-and-HC-05-Bluetooth-Serial-Port-Module

b00682613, 2016-02-15:
Yeah I followed the guide except I couldn't complete step 2, I kept getting this message:
```
Starting config    

Done!
```

Using the tutorial in the link I was able to set the : AT+UART=57600,0,0 and AT+PSWD=0000
After that I uploaded StandardFirmata onto the board.

arduinofy.blogspot.com.au/2013/10/tutorial-programming-hc-05-at-mode-with.html?_sm_au_=i3VLJKMHr4QFB5WP

rwaldron, 2016-02-15:
Let's ping a few others that have more experience with bluetooth than I do cc @ajfisher @BrianGenisio @darioodiaz

ajfisher, 2016-02-15:
I don't think you should have 2 COM ports registered for one BT module - I'm not sure why you'd have outgoing and incoming but that seems strange to me as COM ports (even under windows) can be bidirectional. I'd remove both and try and pair again just once.

Okay good way to test this is to do this:

Prerequisites:

* Install StandardFirmata on the arduino.
* Ensure your HC-05 is configured to 57600 baud
* Ensure your HC-05 and computer are paired and you are definitely connected (flashing waiting light goes solid)
* Make sure you have a terminal software (on windows I think it's hyperterminal or putty or something - basically something you can connect over serial with).

Then do this:

Wire up your HC-05 to your arduino now, make sure power is externally provided not via USB from your computer.

Open your terminal and connect to the BT Serial COM port at 57600 baud. At this point you'll get a blank screen. Assuming that screen stays on (and doesn't drop you out again) and the connection light is solid then you should be connected.

At this point hit RESET on the arduino and you should see the arduino do it's little Pin 13 LED dance. Once that's done, on your serial terminal you should see something like this:

```
����y��SimpleBotFirmata.ino
```

If that's working then you *should* be able to connect to it from your johnny five script now. If it's NOT doing that then one of your pre-requisites is not complete.

b00682613, 2016-02-16:
@waldron @ajfisher I got it working now.

Under bluetooth settings I found which port number to use:
Incoming (device initiates the connection)
Outgoing (your PC initiates the connection)

In my case the outgoing port is COM5.

```javascript
var five = require('johnny-five')
var board = new five.Board({
  port: 'COM5'
});

board.on("ready", function() {
  var led = new five.Led(13);

   this.repl.inject({
     led: led
   });
});
```

closed 2016-02-16 by b00682613

ajfisher, 2016-02-16:
Brilliant news!