rwaldron/johnny-five #1037: Detect if board gets disconnected

sofroniewn, 2016-02-16:
Hi
I was wondering if there is a way to detect if a board gets disconnected (i.e. the USB gets unplugged). I tried looking if there was a board fail event, but that does not appear to happen. I also thought about looking at whether there is a serial close event, but I am not sure if that is the right approach.
Any help or advice would be much appreciated
Thanks
Nick

rwaldron, 2016-02-17:
Hi, unfortunately, there is no way to disambiguate the following scenarios: 

- USB unplugged
- Board shorted
- Board brown out

The serialport sees these the same way: as a non-communicating device.

soundanalogous, 2016-02-17:
I have seen Firmata users (in perl-firmata, not johnny-five) poll a connection by sending a `REPORT_VERSION` message at a set interval. If a response is not received you could assume a disconnection. I don't think this is something that should be added to johnny-five as a convenience, but it is a technique a user could perform.

sofroniewn, 2016-02-17:
@rwaldron - thank you for your response. I do not need to disambiguate between those events, just knowing that any one of them has occurred and that the device is now non-communicating is sufficient for me. Does the serialport emit an event when a device becomes non-communicating?

@soundanalogous polling the connection could be an option. I will look into this more. Thanks

sofroniewn, 2016-02-17:
I just found this issue - https://github.com/voodootikigod/node-serialport/issues/265 - that looks like it might explain what I need. I have to run now, but will look into it tonight

rwaldron, 2016-02-17:
@soundanalogous 

> If a response is not received you could assume a disconnection.

How do they disambiguate the possibilities that I listed?

rwaldron, 2016-02-17:
I think we could just forward the serialport's close event

rwaldron, 2016-02-17:
@soundanalogous https://github.com/jgautier/firmata/pull/111

sofroniewn, 2016-02-17:
@rwaldron that sounds great!

rwaldron, 2016-02-17:
@sofroniewn cool, I will write up the patch for Johnny-Five as well

soundanalogous, 2016-02-17:
> How do they disambiguate the possibilities that I listed?

They use it for Ethernet but it could apply to Serial but yeah you would not know the reason for the disconnect based on that technique.

Forwarding the close event sounds like a good solution.

sofroniewn, 2016-02-17:
@rwaldron Perfect, I'll give it a test when it's ready. Thanks again :-)

rwaldron, 2016-02-17:
@sofroniewn "close" event forwarding is now in v0.9.21

rwaldron, 2016-02-17:
Thanks for the report!!

sofroniewn, 2016-02-17:
Unfortunately I'm not detecting the board close event with johnny-five when I unplug the USB. I've installed the new johnny-five, v0.9.21 using npm, and it contains the modifications you made to the code in the most recent commit. Below is the code I am trying to run. I see the "Board ready" message in the console, but not the "Board closed" message after I unplug the cable.

var five = require("johnny-five"),
board = new five.Board();

board.on("ready", function() {
  console.log('Board ready')
  var led = new five.Led(12);
  led.on()
});

board.on("close", function () {
	console.log('Board closed')
})

--------------------------------------------------------------------------------------------------------------------------

As a check I also installed the new version of firmata v.0.8.2 that you updated too. In this case I was able to pick up the close event after unplugging the USB using the following code

var Board = require("firmata");
var board = new Board("/dev/cu.usbmodem1411");

board.on("ready", function() {
	console.log('ready')
});

board.on("close", function() {
	console.log('closed')
});

--------------------------------------------------------------------------------------------------------------------------

Any suggestions as to why this might be the case would be much appreciated, I will keep on trying to get it working on my end in case I am making a silly mistake somewhere.
Thanks again
Nick

rwaldron, 2016-02-17:
> Unfortunately I'm not detecting the board close event with johnny-five

Can you do `npm install johnny-five@0.9.21` and then try again?

sofroniewn, 2016-02-17:
I just tried again, same behavior

rwaldron, 2016-02-17:
@sofroniewn yep, sorry about that. It was late and I rushed :( I'm looking at this now. Again, sorry about that

rwaldron, 2016-02-17:
@sofroniewn I've got this sorted out, but then discovered some nastiness that led me to this: https://github.com/voodootikigod/node-serialport/issues/638. I can land the fixes I have, but that needs to be addressed in serialport

sofroniewn, 2016-02-17:
@rwaldron ok thanks so much digging into to this - do it make sense to you why it worked when I was using firmata?

sofroniewn, 2016-02-17:
*does it

rwaldron, 2016-02-17:
> ok thanks so much digging into to this - do it make sense to you why it worked when I was using firmata?

Yes, there were 2 bugs I found: 

1. Bug in Johnny-Five: I was attaching the forwarding at the wrong point—this lead to a nice clean up 
2. The bug in serialport

sofroniewn, 2016-02-17:
ok nice, my current understanding of this issue is that if the board gets unplugged i will now receive the close event after you have fixed #1, but #2 means that if I try and write to that board after it is unplugged I will still get an exception, because the error handling has been removed (I think I was seeing this error when I unplugged the board while my led was strobing) and fixing that will require a change to serialport

rwaldron, 2016-02-17:
You are exactly correct.

sofroniewn, 2016-02-17:
I've just run npm install johnny-five@0.9.22 and tried the program I have up top again
but I'm still not seeing the event when I unplug the USB. Sorry about this.

rwaldron, 2016-02-17:
@sofroniewn what do you think about this: I could wrap Firmata's calls to transport.write in a "safe write" operation, catch the errors and "safely" forward them

rwaldron, 2016-02-17:
@sofroniewn can you post the output of running that program?

sofroniewn, 2016-02-17:
@rwaldron the "safe write" sounds very reasonable to me

sofroniewn, 2016-02-17:
Here is the output of running the following program https://gist.github.com/sofroniewn/511a40bc859c371e5c5b
I unplug the USB after the "Board ready" message and nothing happens. I then press control C twice to quit normally

sofroniewn-lm1:johnny-five-test sofroniewn$ node board-close.js
1455740159684 Device(s) /dev/cu.usbmodem1411
1455740159689 Connected /dev/cu.usbmodem1411
1455740162347 Repl Initialized
>> Board ready

(^C again to quit)
>>
1455740173804 Board Closing.

------------------------------------------------------------------------------------------------------------------------------

rwaldron, 2016-02-17:
@sofroniewn ok, thanks.

rwaldron, 2016-02-17:
Super strange... when I run that from the repo, it works as expected. However...

```
$ mkdir bc && cd bc && npm install johnny-five && wget https://gist.githubusercontent.com/sofroniewn/511a40bc859c371e5c5b/raw/5233237999b4f58d1aeef9afae1fdc4fe288aa68/board-close.js
\
> serialport@2.0.6 install /Users/rwaldron/bc/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/Users/rwaldron/bc/node_modules/johnny-five/node_modules/serialport/build/Release/node-v46-darwin-x64/serialport.node" is installed via remote
johnny-five@0.9.22 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── browser-serialport@2.0.3
├── color-convert@0.5.3
├── temporal@0.5.0
├── nanotimer@0.3.10
├── chalk@1.1.1 (escape-string-regexp@1.0.4, supports-color@2.0.0, ansi-styles@2.1.0, strip-ansi@3.0.0, has-ansi@2.0.0)
├── es6-shim@0.34.4
├── firmata@0.7.4 (es6-shim@0.33.13)
├── array-includes@3.0.2 (define-properties@1.1.2, es-abstract@1.5.0)
├── lodash@3.10.1
└── serialport@2.0.6 (bindings@1.2.1, sf@0.1.7, async@0.9.0, debug@2.2.0, nan@2.0.9, optimist@0.6.1)
--2016-02-17 15:23:02--  https://gist.githubusercontent.com/sofroniewn/511a40bc859c371e5c5b/raw/5233237999b4f58d1aeef9afae1fdc4fe288aa68/board-close.js
Resolving gist.githubusercontent.com... 199.27.76.133
Connecting to gist.githubusercontent.com|199.27.76.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 231 [text/plain]
Saving to: 'board-close.js'

board-close.js         100%[============================>]     231  --.-KB/s   in 0s

2016-02-17 15:23:03 (24.5 MB/s) - 'board-close.js' saved [231/231]

rwaldron at nova in ~/bc
$ node board-close.js
1455740623187 Device(s) /dev/cu.usbmodem1421
1455740623197 Connected /dev/cu.usbmodem1421
1455740627178 Repl Initialized
>> Board ready
```

:|

sofroniewn, 2016-02-17:
hmm puzzling

sofroniewn, 2016-02-17:
Does the board close event propogating depend on the firmata version? You have firmata@0.7.4 listed above (as I have installed on my computer), but you did not make the changes for board close firmata until v0.8.2

rwaldron, 2016-02-17:
> You have firmata@0.7.4 listed above (as I have installed on my computer), but you did not make the changes for board close firmata until v0.8.2

Hahaha, yep—that's it.

rwaldron, 2016-02-17:
Ok, try 0.9.23?

rwaldron, 2016-02-17:
```
rwaldron at nova in ~/bc
$ ..
rwaldron at nova in ~
$ rm -r bc
rwaldron at nova in ~
$ mkdir bc && cd bc && npm install johnny-five && wget https://gist.githubusercontent.com/sofroniewn/511a40bc859c371e5c5b/raw/5233237999b4f58d1aeef9afae1fdc4fe288aa68/board-close.js
\
> serialport@2.0.6 install /Users/rwaldron/bc/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/Users/rwaldron/bc/node_modules/johnny-five/node_modules/serialport/build/Release/node-v46-darwin-x64/serialport.node" is installed via remote
johnny-five@0.9.23 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── browser-serialport@2.0.3
├── color-convert@0.5.3
├── temporal@0.5.0
├── nanotimer@0.3.10
├── chalk@1.1.1 (escape-string-regexp@1.0.4, supports-color@2.0.0, ansi-styles@2.1.0, strip-ansi@3.0.0, has-ansi@2.0.0)
├── es6-shim@0.34.4
├── firmata@0.8.3 (es6-shim@0.33.13)
├── array-includes@3.0.2 (define-properties@1.1.2, es-abstract@1.5.0)
├── lodash@3.10.1
└── serialport@2.0.6 (bindings@1.2.1, async@0.9.0, sf@0.1.7, nan@2.0.9, debug@2.2.0, optimist@0.6.1)
--2016-02-17 15:52:54--  https://gist.githubusercontent.com/sofroniewn/511a40bc859c371e5c5b/raw/5233237999b4f58d1aeef9afae1fdc4fe288aa68/board-close.js
Resolving gist.githubusercontent.com... 23.235.46.133
Connecting to gist.githubusercontent.com|23.235.46.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 231 [text/plain]
Saving to: 'board-close.js'

board-close.js         100%[============================>]     231  --.-KB/s   in 0s

2016-02-17 15:52:55 (13.8 MB/s) - 'board-close.js' saved [231/231]

rwaldron at nova in ~/bc
$ node board-close.js
1455742386240 Device(s) /dev/cu.usbmodem1421
1455742386248 Connected /dev/cu.usbmodem1421
1455742390224 Repl Initialized
>> Board ready
Board closed
```

sofroniewn, 2016-02-17:
@rwaldron It works great now, thanks! I receive the board close event just fine.

I think adding the "safe write" would be really nice too - or if there was another way to prevent the board from trying to write to the serial port after a close event that could work instead

rwaldron, 2016-02-17:
> I think adding the "safe write" would be really nice too - or if there was another way to prevent the board from trying to write to the serial port after a close event that could work instead

I'll do the safewrite for now, but probably won't get to it until tomorrow

sofroniewn, 2016-02-17:
That's more than fine, I really appreciate how promptly you've dealt with this issue

rwaldron, 2016-02-17:
:) I appreciate your patience

arunyka, 2017-03-18:
I am getting following error when running arduino attached servos using johnny-five. please help me
Error: Writing to COM port (GetOverlappedResult): Unknown error code 31
    at Error (native)

dtex, 2017-03-18:
@arunyka looking for a little more info... how are you powering the servos?

arunyka, 2017-03-19:
i have attached 6 servos to the arduino uno board 
signal pins attached to 3,5,6,9,10,11 PWM pins
power is getting via power adapter connected to the uno board. 
i connected  5v and ground pins of the  uno to the breadboard + and -
the red and brown wires of each servo to the breadboard + and - slots. 
i have use jumper wires to establish connections with breadboard and uno board.

rwaldron, 2017-03-20:
> i connected 5v and ground pins of the uno to the breadboard + and -

Is there an external power source? For example: a battery pack, or wall power?

arunyka, 2017-03-20:
i am using 12v adaptor from wall power. but it is connected to the uno board power jack.

arunyka, 2017-03-20:
yes i have use power adaptor (12v) from wall power. It is connected to the arduino power jack

dtex, 2017-03-20:
You still have 1A limit on the 5V pin when using an AC adapter instead of USB power (it's 500mA on USB). That's not nearly enough for 6 servos. You will need to bypass the Arduino for your servo power.

arunyka, 2017-03-23:
Tnx, I gave separate power supply to control servos. It is  working now.

dtex, 2018-01-05:
It looks like both threads in this issue were resolved so I'm going to close it.

closed 2018-01-05 by dtex

alvaropfn, 2018-07-04:
hello friends, i'm trying to acquire the same behavior of @sofroniewn but with a different approach.

I has ESP8266 connected over firmata wifi instead of a arduino with a serial cable.

In tests with arduino with cable the close event works well but over ip don't.

[here's](https://gist.github.com/alvaropfn/7d62c623265d117ff179f0b30c7727df#file-disconnecttest-js) what i have