rwaldron/johnny-five #1040: Failed to install dependencies on ubiLinux on Edison

dougseven, 2016-02-22:
I am trying to use Node 4.3.1, 5.0.0 or 5.6.0 on the Intel Edison (the Azure IoT SDK requires it). The official Yocto image is built with Node.js 0.10.*. I am trying to see if ubiLinux is a viable solution.

1. I installed ubiLinux on the Edison using the instructions [here](http://www.emutexlabs.com/ubilinux/29-ubilinux/218-ubilinux-installation-instructions-for-intel-edison).
2. I configured WiFi, etc.
3. Ran apt-get update
4. Installed Node.js by installing curl, Node.js, and Node.js build tools as follows:

```
apt-get install curl
curl -sL https://deb.nodesource.com/setup_5.x | bash -
apt-get install --yes nodejs
apt-get install --yes build-essential
```

That installed Node.js 0.10.42

I installed NVM...

```
git clone https://github.com/creationix/nvm.git ~/.nvm && cd ~/.nvm && git checkout `git describe --abbrev=0 --tags`

. ~/.nvm/nvm.sh
```

Added these lines to ~/.profile

```
export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && . "$NVM_DIR/nvm.sh" # This loads nvm
```

Then I installed Node.js 5.0.0

```
nvm install 5.0.0
```

In my package.json for my test J5 app I have:

```javascript
{
  "name": "thinglabs-iot-sample",
  "repository": {
    "type": "git",
    "url": "https://github.com/ThingLabsIo/IoTLabs/tree/master/Edison/AzureIoT"
  },
  "version": "0.1.2",
  "private":true,
  "description": "Sample app that connects a device to Azure using Node.js",
  "main": "temp.js",
    "author": "Doug Seven",
  "license": "MIT",
  "dependencies": {
    "azure-iot-device": "1.0.1",
    "azure-iot-device-amqp": "1.0.1",
    "azure-iot-device-http": "1.0.1",
    "azure-iot-device-mqtt": "1.0.1",
    "johnny-five": "0.9.24",
    "edison-io": "0.9.1"
  }
}
```

When I run npm install this is the output

```
npm WARN engine edison-io@0.9.1: wanted: {"node":"latest"} (current: {"node":"5.0.0","npm":"3.3.6"})

> galileo-io@0.8.18 postinstall /root/iot-labs/azure-iot/node_modules/galileo-io
> node scripts/postinstall

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
  Do not quit the program until npm completes the installation process.  
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
{ [Error: Command failed: /bin/sh -c opkg info libmraa0
/bin/sh: 1: opkg: not found
]
  killed: false,
  code: 127,
  signal: null,
  cmd: '/bin/sh -c opkg info libmraa0' }
npm ERR! Linux 3.10.17-yocto-standard-r2
npm ERR! argv "/root/.nvm/versions/node/v5.0.0/bin/node" "/root/.nvm/versions/node/v5.0.0/bin/npm" "install"
npm ERR! node v5.0.0
npm ERR! npm  v3.3.6
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

npm ERR! galileo-io@0.8.18 postinstall: `node scripts/postinstall`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the galileo-io@0.8.18 postinstall script 'node scripts/postinstall'.
npm ERR! This is most likely a problem with the galileo-io package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node scripts/postinstall
npm ERR! You can get their info via:
npm ERR!     npm owner ls galileo-io
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/iot-labs/azure-iot/npm-debug.log
```
[npm-debug.txt](https://github.com/rwaldron/johnny-five/files/140151/npm-debug.txt)

rwaldron, 2016-02-22:
It looks like a pretty straight forward issue—the post install script that handles installing the "safe version" of libmraa is written with the assumption that this would be installed on Intel's IoT image, where it would use opkg to install and build libmraa. I'll get an Edison set up with ubilinux and sort this out.

rwaldron, 2016-02-22:
Note, I had to use `sudo bash ./flashall.sh` for it to work, otherwise I kept getting 

```
$ ./flashall.sh
-bash: ./flashall.sh: Permission denied
```

rwaldron, 2016-02-22:
That failed anyway...

```
$ sudo bash ./flashall.sh
Using U-Boot target: edison-blank
Now waiting for dfu device 8087:0a99
Timed out while waiting for dfu device 8087:0a99
DEBUG: lsusb
./flashall.sh: line 84: lsusb: command not found
DEBUG: dfu-util -l
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Deducing device DFU version from functional descriptor length
Found Runtime: [05ac:8286] ver=0151, devnum=7, cfg=1, intf=3, alt=0, name="UNKNOWN", serial="UNKNOWN"
Did you plug and reboot your board?
If yes, please try a recovery by calling this script with the --recovery option
```

rwaldron, 2016-02-22:
Swapping out the microusb fixed that last issue

rwaldron, 2016-02-22:
Well... until it didn't... https://gist.github.com/rwaldron/2daeb31017f63a7b693b

dougseven, 2016-02-22:
I have found that the Edison is very sensitive to microUSB cables.

rwaldron, 2016-02-22:
@dougseven can you try with edison-io@0.9.2?

Here's the result of my install: 

```
root@ubilinux:~/some-botty# npm install edison-io
npm WARN engine edison-io@0.9.2: wanted: {"node":"latest"} (current: {"node":"5.0.0","npm":"3.3.6"})
npm WARN engine edison-io@0.9.2: wanted: {"node":"latest"} (current: {"node":"5.0.0","npm":"3.3.6"})

> galileo-io@0.9.2 postinstall /root/some-botty/node_modules/galileo-io
> node scripts/postinstall

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
  Do not quit the program until npm completes the installation process.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

/root/some-botty
└─┬ edison-io@0.9.2
  └─┬ galileo-io@0.9.2
    ├── es6-promise@3.1.2
    ├── es6-shim@0.34.4
    └─┬ remapped@0.2.1
      ├── getobject@0.1.0
      └── traverse@0.6.6
```

Please be patient when the "Do not quit the program until npm completes the installation process." is displayed, it may take 2-3 minutes—it's not hanging, just working.

dougseven, 2016-02-22:
Hmmm - that is inconsistent with the message I got. FWIW - I let it run - but then I got the error

```
{ [Error: Command failed: /bin/sh -c opkg info libmraa0
/bin/sh: 1: opkg: not found
]
  killed: false,
  code: 127,
  signal: null,
  cmd: '/bin/sh -c opkg info libmraa0' }
```

I didn't try npm install edison-io, I just had it as a dependency and tried an npm install. To the best of my knowledge, that shouldn't make a difference, but I will try tonight.

dougseven, 2016-02-22:
I'll update the dependency to edison-io 0.9.2 and try it out.

rwaldron, 2016-02-22:
> Hmmm - that is inconsistent with the message I got. FWIW - I let it run - but then I got the error

With 0.9.2?

> I didn't try npm install edison-io, I just had it as a dependency and tried an npm install. To the best of my knowledge, that shouldn't make a difference, but I will try tonight.

That's correct, it doesn't make a difference, but I pushed out some changes that should prevent that failure

dougseven, 2016-02-22:
Upgraded to edison-io@0.9.2 did the trick. Johnny-Five running on the Edison using the Azure IoT SDK for Node.js

Thanks!

rwaldron, 2016-02-23:
Excellent!

closed 2016-02-23 by rwaldron