rwaldron/johnny-five #1043: Johnny Five and Arduino Shields

darnaud, 2016-02-25:
Hi!
My arduino project is an alarm system with with an Arduino Uno, an Arducam and Camera shield. Can anybody gives me tips on how to make these 3 work nicely together? Will this be possible with johnny-five?

henricavalcante, 2016-02-25:
I don't know if Johnny-five can handle Arducam but recently I saw this book on amazon which can help you:

http://www.amazon.com/Security-Systems-using-Android-Arduino/dp/1522760415/ref=sr_1_63?m=A2IAB2RW3LLT8D&s=merchant-items&ie=UTF8&qid=1452266704&sr=1-63

darnaud, 2016-02-26:
I will lokk into it! Thank you for your input!

lyzadanger, 2016-08-31:
Hi, @darnaud ! I hope you were able to make good progress on your project. Is there any other way we can help? I'd like to close this issue because it's been sitting for a while, but please reopen and comment if there is something specific we could help you with.

closed 2016-08-31 by lyzadanger

darnaud, 2016-08-31:
@henricavalcante Thanks to the book I found a solution! Thank you sir!
@lyzadanger my project went well thank you for asking. Problem solved ;)