rwaldron/johnny-five #1044: Layer on top of johnny-five - callbacks

b00682613, 2016-02-25:
@rwaldron @ajfisher  How could I pass callbacks in. Take this example:

```javascript
// my-board.js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var motion = new five.IR.Motion(7);

  motion.on("motionstart", function() {
    console.log("motionstart", Date.now());
  }); // pass callback from module.exports instead
});

module.exports = {
  motionDetected: function (cb) {
     // somehow pass cb into motion.on
     // motion.on('motionstart', cb)
  }
}

```

```javascript
// app.js
var board = require('./my-board')

board.motionDetected(function () {
  console.log('Motion detected.')
})
```

rwaldron, 2016-02-26:
I didn't test it, but this should work: 

```js
// my-board.js
var five = require("johnny-five");
var board = new five.Board();
var callbacks = new Map();

function register(type, callback) {
  if (!callbacks.has(type)) {
    callbacks.set(type, [callback]);
  } else {
    callbacks.get(type).push(callback);
  }
}

board.on("ready", function() {
  var motion = new five.IR.Motion(7);

  motion.on("motionstart", function() {
    console.log("motionstart", Date.now());
    if (callbacks.has("motionstart")) {
      callbacks.get("motionstart").forEach(function() {
        callback();
      });
    }

  }); // pass callback from module.exports instead
});

module.exports = {
  motionDetected: function (cb) {
    register("motionstart", cb);
  }
}
```

b00682613, 2016-02-27:
@rwaldron Thanks for this!

closed 2016-02-27 by b00682613

henricavalcante, 2016-02-27:
I think here isn't a right place to ask this but I'll be glad if @rwaldron or someone could answer for me what the difference between use JavaScript Map object and use a literal object with brackets like this callbacks['callback']? I made a simple performance tests to figure out the difference and realize that when I have more than ~10000 keys is better to use Map, less than this is better to use {}.

here is the code I used to test: https://github.com/henricavalcante/map-vs-object/

rwaldron, 2016-02-29:
Map can have an object reference as a key, but that's not relevant in this case, so you could certainly use a plain object instead.