rwaldron/johnny-five #1045: Third Party Arduino Shield

gcrev93, 2016-02-26:
How can I get Johnny Five to communicate with a third party mp3 player shield?

henricavalcante, 2016-02-26:
I've never used a mp3 shield before, but i think you should say which shield are you trying to use with johnny-five, [this](https://www.sparkfun.com/products/12660) one?

rwaldron, 2016-02-26:
Hi @gcrev93, this appears to require the SPI interface, which is not yet supported by the underlying protocol.

gcrev93, 2016-02-26:
DO you think it is possible using the Serial port?

soundanalogous, 2016-02-27:
Even if SPI was supported I don't think it would be possible. The required buffer is likely too large. Using an MP3 decoder like that requires a ring buffer in the firmware to stream the data. I'd be surprised if this was possible over a remote interface like Firmata (but I'd also be happy to be proven wrong someday).

soundanalogous, 2016-02-27:
What would work however is writing a Firmata plugin that exposes a player interface, then send controls via that player interface rather than trying to control streaming.

gcrev93, 2016-03-03:
that is the approach i am going for.. I want to use johnny five to send the controls. not really sure where to start but we will see

soundanalogous, 2016-03-03:
One technique is to expose the interface via I2C. That way you don't even need to create a custom plugin for Firmata. The board you attach the mp3 player shield will be an I2C slave device (but you will need to define the interface which exposes controls to the player - you will also handle the streaming from the SD card to the mp3 decoder in the same application). You can then communicate with the board from johnny-five / firmata.js using the i2c functions.

gcrev93, 2016-03-03:
I am still quite the beginner so this may be a challenge but i will have to do a lot of research to get into it. thank you!

rwaldron, 2016-03-04:
@soundanalogous a backpack—that's a brilliant idea! @gcrev93 we can help you make this happen :)

rwaldron, 2016-03-04:
@gcrev93 can you link me to the shield you're using? I'll get one ordered and we'll start planning a strategy

gcrev93, 2016-03-04:
@rwaldron the shield I got is this one : http://www.geeetech.com/wiki/index.php/Arduino_MP3_shield_board_with_TF_card .

It is compatible with the Sparkfun MP3 Shield code as well . https://github.com/sparkfun/MP3_Player_Shield/tree/V_1.5

gcrev93, 2016-03-14:
@rwaldron I also wanted to add that I will be presenting at OSCON and johnny five is a big feature of my talk. SO I will be definitely featuring this story into my talk. If the strategy gets done before that great! If not, I will still mention this moment and that it will be something to cme

rwaldron, 2016-03-14:
In Austin?

gcrev93, 2016-03-14:
Yes in Austin. My talk is May 19. Here is the link: http://conferences.oreilly.com/oscon/open-source-us/public/schedule/detail/49289

gcrev93, 2016-03-19:
Hey Rick

If you ever want to schedule a google hangout to hack on this idea together, let me know ! :)

dtex, 2018-01-05:
Just adding a note that the MP3 player functionality is probably better suited as a Johnny-Five plug-in vs. a Johnny-Five class.

closed 2019-01-07 by stale[bot]