rwaldron/johnny-five #1055: Generic pin state reading

alexcroox, 2016-03-08:
Sorry if this is in the documentation, I couldn't find it!

I want to do a basic digital read on a pin to know if it's high or low, even better if there's a `change` event I can listen to.

alexcroox, 2016-03-08:
I believe the solution is

```
var sensor = new five.Pin({
        "pin": 5
 });

sensor.on("high", function() {

        console.log('Scored!');
});
```

rwaldron, 2016-03-08:
You can you `Sensor.Digital`: 

```js
var vibration = new five.Sensor.Digital(5);
vibration.on("change", function() {
  // ...
});
```

alexcroox, 2016-03-09:
Nice thank you!

I assume I can expand that to specify the virtual board as well as the pin

```
var sensor = new five.Sensor.Digital({
        "pin": 5,
        "board": virtual
});
```

closed 2016-03-09 by alexcroox

rwaldron, 2016-03-09:
Yessir!