rwaldron/johnny-five #1056: Software Controlling MUX Shield 2 (Analogue) expander rows

alexcroox, 2016-03-08:
I have the MUX Shield 2 (I believe the analogue version as it has 3 rows instead of 2).

I would like to set the first row (I/O 1) to be digital inputs (to read a digital vibration sensor).

According to the [documentation](http://mayhewlabs.com/media/Mux_Shield_II_User_Guide.pdf) I need to set the mode through a couple of pins.

```
board.on("ready", function() {

        var virtual = new five.Board.Virtual(
            new five.Expander("MUXSHIELD2")
        );

        // Set I/O 1 row to digital in
        var modePin = new five.Pin(8);
        five.Pin.write(modePin, 0);

        var firstRowInPin = new five.Pin(10);
        five.Pin.write(firstRowInPin, 0);
});
```

I'm having no luck though, am I missing something? I have the option of cutting the connections on the back of the board to permanently set the mode but I'd rather do it through software if possible.

[image]

rwaldron, 2016-03-09:
The Expander instance takes care of all the pin mode management for you—interfering with its operations will put the expander and shield into an unknowable state. Just pass the `virtual` board to the constructor like:

```js   
var vibration = new five.Sensor.Digital({
  board: virtual,
  pin: (whatever pin the sensor is connected to on the shield)
});

```

alexcroox, 2016-03-10:
Ah very interesting! I'm getting a constant 5V signal to my sensor digital pin so I presumed the shield I/O row must be set to digital out instead of digital in. I must have wired something wrong :P

One advantage of cutting the hard wire on the back to force the mode is it frees up some of the PWM pins that I need for controlling some motors, which I don't think I can do with the expander pins. Is this a non option then as the Expander Instance has to auto control the mode (and therefore needs it to be a "factory state" at all time)?

rwaldron, 2016-03-10:
Yes, I think that correctly describes what I meant.

alexcroox, 2016-03-11:
Excuse my ignorance, but I could use the MUX shield 2 to turn one of the virtual IO rows to be analogue output and therefore send a PWM signal to control a motor right? I noticed the PIN wiki document shows how I can set a PIN's mode to be PWM (mode 3) so I can send a signal to a L298N Dual Motor Controller using the MUX Shield 2.

rwaldron, 2016-03-11:
No, the muxshield doesn't have PWM output (which is necessary for "analog" output) capability. The muxshield can be used for analog input, digital input or digital output, one mode at a time, per row.

rwaldron, 2016-03-28:
I'm going to close this now—reopen if necessary :)

closed 2016-03-28 by rwaldron