rwaldron/johnny-five #1057: MPU6050 sensor is not working with StandardFirmataEthernet

sarkarstanmoy, 2016-03-10:
Hi,

I am using Arduino UNO and upload StandardFirmataEthenet by putting valid ipaddress for both localIp and remoteIp. What I am trying to do is to send sensor data to firebase database. It is throwing error ECONNRESET.

var five = require("johnny-five");
var EtherPort = require("etherport");
var board = new five.Board({ 
  port: new EtherPort(3030) 
});

var Firebase = require("firebase");

board.on("ready", function() {
    
    var myFirebaseRef = new Firebase("URL..");
  
    
  var accelerometer = new five.Accelerometer({
    controller: "MPU6050"
  });

  accelerometer.on("change", function() {
    
  myFirebaseRef.set({
  inclination: this.inclination 
  
});
   
    });
});

 Can you please let me know what to do.

soundanalogous, 2016-03-10:
Do  you know if the sketch is running at all? Add a console log within the ready callback to see if that ever gets called. Whether or not you are using Serial or Ethernet should not matter as far as reading a sensor is concerned.

sarkarstanmoy, 2016-03-10:
Hi,

I have added console log within ready callback.

var five = require("johnny-five");
var EtherPort = require("etherport");
var board = new five.Board({ 
  port: new EtherPort(3030) 
});
var Firebase = require("firebase");

board.on("ready", function() {    
    console.log("ready");    
   var myFirebaseRef = new Firebase("URL");   
  var accelerometer = new five.Accelerometer({
    controller: "MPU6050"
  });
  accelerometer.on("change", function() {    
  myFirebaseRef.set({
  inclination: this.inclination   
});   
 });    
});

and I am getting below output

1457633816574 SerialPort Listening on port: 3030  
1457633816574 Connected Listening on port: 3030  
1457633818588 Repl Initialized  
>> ready

events.js:72
throw er; // Unhandled 'error' event
              ^
Error: read ECONNRESET

This issue occurs only when I uploaded StandardEthernetFirmata and try to use upload the sensor data to firebase. When I am not reading sensor data and try to send dummy data to firebase it is working fine. 

Can you please help me to figure out the problem? I have tried ConfigurableFirmata also with no success Also, I tried to generate Firmata using http://firmatabuilder.com/ by using Ethernet connection type having Analog Read/Output , Serial Firmata option enabled with no success.

sarkarstanmoy, 2016-03-10:
Hi

Actually I am using arduino Ethernet shield with arduino uno to send mpu6050 data to firebase.

1. If I upload standard firmata to arduino uno then I am able to see sensor data with firebase code comment out.

2.. Since I want to send data to firebase through HTTPS I have used aruino Ethernet shield. If I comment out sensor code I can easily send data to firebase. But as soon as I am enable sensor code it fails with ECONNRESET

Hope I am able to explain my problem. Please assist on this

sarkarstanmoy, 2016-03-10:
I am able to send dummy data using arduino Ethernet shield and Johnny five

soundanalogous, 2016-03-10:
I'm unable to determine based on the info you have provided whether the issue is with johnny-five, firebase or with StandardFirmataEthernet. If you can isolate the problem to use as few components as possible that will be helpful.

soundanalogous, 2016-03-10:
Please verify the following:

1. You can read sensor data using StandardFirmataEthernet and johnny-five (but not including any firebase code)

2. You can send data to Firebase from within a johnny-five application using StandardFirmata over a Serial connection

sarkarstanmoy, 2016-03-10:
But if I try to send sensor data using standard Ethernet firmata, Johnny five to firebase it is throwing error like ECONNRESET

soundanalogous, 2016-03-10:
You need to determine what is throwing the error in that case. The error message is not helpful. Do you have a way to trace your application (I'm not familiar with techniques for tracing node.js applications but I'm sure debuggers exist).

soundanalogous, 2016-03-11:
See this post on stackoverflow: http://stackoverflow.com/questions/17245881/node-js-econnreset

Is it possible you are attempting to send data more frequently than Firebase permits and thus Firebase is severing the TCP connection? What happens if you try to send data once every few seconds as opposed to each time the "change" event fires?

sarkarstanmoy, 2016-03-11:
[image]

I have attached the screenshot of debugging, I have remove fireabase and now only try to log data into console. But still getting error.

var five = require("johnny-five");
var EtherPort = require("etherport");
var board = new five.Board({ 
  port: new EtherPort(3030) 
});
board.on("ready", function() {    
    console.log("ready");
  var accelerometer = new five.Accelerometer({
    controller: "MPU6050"
  });
 this.wait(3000,function(){
      accelerometer.on("change", function() {
    console.log(this.inclination);     
    });     
 }); 
    accelerometer.on("error", function(er) {
     console.log(console.log(er, er.stack.split("\n")));
    });   
});  
  board.on("error", function(error)  {
    console.log(console.log(error, error.stack.split("\n")));
});

rwaldron, 2016-03-11:
Can you try something else? Just after `console.log("ready");`, add: 

```js
this.samplingInterval(100); // 100ms this will reduce the sampling frequency to ~10Hz
```

I'm wondering if there is too much data, too frequently?

sarkarstanmoy, 2016-03-11:
var five = require("johnny-five");
var EtherPort = require("etherport");
var board = new five.Board({ 
  port: new EtherPort(3030) 
});
board.on("ready", function() {    
    console.log("ready");
    this.samplingInterval(100); // 100ms this will reduce the sampling frequency to ~10Hz
    
    
 var accelerometer = new five.Accelerometer({
    controller: "MPU6050",
    sensitivity : 131
  });
  
  
   setInterval(function() {
       accelerometer.on("data", function() {
    
         console.log(this.inclination);    
     });
   },5000);
});  
  board.on("error", function(error)  {
    console.log(console.log(error));
});

Still getting same error

rwaldron, 2016-03-11:
Did you try adjusting the sampling interval? I don't have an Ethernet shield with me at the moment (I can try this when I get home later). Thanks for your patience

rwaldron, 2016-03-11:
Wait... I don't understand this:

```js   
setInterval(function() {
accelerometer.on("data", function() {

     console.log(this.inclination);    
 });
},5000);
});
``` 

That's a 0 interval operation that registers a new  "data" event every 0 milliseconds. Please retry the suggestion without the data event handler nested in the `setInterval`. Also, try using the "change" event instead of "data"

soundanalogous, 2016-03-11:
@sarkarstanmoy please add three backticks before and after your code blocks. It makes it easier for people on this thread to read. See here for more info: https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code-and-syntax-highlighting.

soundanalogous, 2016-03-11:
@sarkarstanmoy is the server you are running the johnny-five code on remote or is it on a LAN? If it is remote are you using a service like Heroku or are you using your own VPS or other environment that you provisioned from scratch?

sarkarstanmoy, 2016-03-12:
Hi,

Thanks for all of your suggestions. I am trying all but unfortunately it is not working. I wonder, what exactly I am doing wrong. Not able to figure it out.
 I am running johnny five on my laptop connected to LAN. Connected Ethernet Shield to my laptop by using LAN Ethernet cross cable. Enable share internet options ( http://windows.microsoft.com/en-in/windows/set-internet-connection-sharing#1TC=windows-7). I will try the suggestions and let you know.

soundanalogous, 2016-03-12:
Do you have a router you can use instead of the cross-cable. I have not had luck with cross cables and the Ethernet shield in the past.

rwaldron, 2016-03-12:
I will try to set up this hardware tomorrow

sarkarstanmoy, 2016-03-12:
Today I will rather try to connect to router directly and try. Will let you the results

sarkarstanmoy, 2016-03-12:
I am very happy to inform that it is working now.

1. I have update node to its latest version.
2. As soundanalogous suggested , I have connected Arduino Ethernet Shield to router and its start working. I have also checked with Firebase and it is working superb.

Probably, when I used cross cable to arduino to my laptop nodejs is trying to open the same socket which is already been used.

Johnny Five rocks and now I will further work on my project.

Thanks all of your wonderful suggestion and time.

rwaldron, 2016-03-12:
@sarkarstanmoy this is great news!! Thank you for posting back with your final outcome, it's very helpful for us and other users to be able to reference each other's solutions :)

I'm going to close this, if there are any further issues related to this, we can re-open or file a new issue where appropriate. Thanks again to everyone for their patience and respect to one-another.

closed 2016-03-12 by rwaldron