rwaldron/johnny-five #1058: Adding servo.to() with time inside of a temporal queue fails to complete the queue

makenai, 2016-03-14:
Here's a test case:

https://gist.github.com/makenai/1047a42855cde861532d

The expected output is

```
One
Two
Three
Four
```

with a servo move to accompany each. However, it only gets to:

```
One
```

and the job stops. If you remove the second argument of `servo.to()` and don't use an animation, it works as expected.

@dtex thinks this might be related to a shared queue object, and I think that makes sense.

makenai, 2016-03-14:
Related to this, here's another code snippet from @rwaldron - the LED stops pulsing when the servo is done:

```javascript
board.on("ready", function() {

  var led = new five.Led(11);
  var servo = new five.Servo({
    pin: 9,
    startAt: 5
  });

  var moves = [
    {degrees: 20, time: 200},
    {degrees: 5, time: 200},
    {degrees: 20, time: 200},
    {degrees: 5, time: 200},
  ];

  function wiggleIt() {
    var move = moves.shift();
    if (move) {
      console.log(move);
      servo.to(move.degrees, move.time).once("move:complete", wiggleIt);
    }
  }

  wiggleIt();

  led.pulse();
});
```

dtex, 2016-03-15:
Cross Referencing #1012 and https://github.com/rwaldron/temporal/issues/13

Note that #1013 did not/could not fix the problem

dtex, 2016-03-17:
Ugh, t all works without the clear() on temporal. I mean, that's a good thing and all but I could have sworn we had a problem and for the life of me I can't recreate it. @makenai is this all working for you now? If so, I guess we can just close this :-/

makenai, 2016-03-17:
@dtex - same results here. I can no longer reproduce the problem. I have mixed feelings about this, but I feel like we can close this issue. Thanks!

rwaldron, 2016-03-17:
Well, we still need to remove `temporal.clear()` ;)

closed 2016-03-17 by rwaldron

dtex, 2016-03-17:
Huh? That's weird. It was already commented out in the version I installed last night.

rwaldron, 2016-03-17:
Sure, but I figured we should probably just delete the line?