rwaldron/johnny-five #1060: Intel edison problem using sensor TSL2561

hannes93, 2016-03-15:
Hey there,
Is there somebody out who now how to get lux value of tsl2561 sensror?

I´m gong to use the intel edison with node.ja and the TSL2561 sensor from Adafruit....
Thanke you for your time.
Greetz \h

lyzadanger, 2016-08-31:
Hi, @hannes93 ! Were you able to get the sensor to work? The TSL2561 is an I2C sensor that is supported by Johnny-Five's `Light` component class. You can find [documentation here](http://johnny-five.io/api/light/). Let me know if I can be of any further help!

closed 2016-08-31 by lyzadanger

rwaldron, 2016-09-02:
FWIW, this sensor's controller is presently broken, as reported in https://github.com/rwaldron/johnny-five/issues/1196 I have work in progress towards fixing.