rwaldron/johnny-five #1061: Stepper motor with L298N motor driver

alexcroox, 2016-03-15:
Following the wiki guide to steppers with a driver, I am using the L298N with [this stepper motor](http://www.ebay.co.uk/itm/261985525062?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT)

[image]

```
new five.Stepper({
  type: five.Stepper.TYPE.DRIVER,
  stepsPerRev: 200,
  pins: [ 11, 12 ]
});
```

**Which is the `step` and `direction` pin on that driver?** 

I'm excited by the prospect of having to use 2 digital pins for a single stepper as I have to drive 2x stepper motors from a single Uno so the more pins I can spare the better (all other examples online use all 4 IN pins on the driver to the Ardiuno.

dtex, 2016-03-15:
Stepper is not my area of expertise, but I don't think DRIVER is the right type for this motor controller. I'm pretty sure that's intended for things like the [Easy Driver](https://www.sparkfun.com/products/12779) and [Big Easy Driver](https://www.sparkfun.com/products/12859)

alexcroox, 2016-03-18:
ah OK thanks for the info

closed 2016-03-18 by alexcroox

calugo, 2017-05-01:
Hello! 
This is closed, however I would like to ask if using  the type FOUR_WIRE, makes this work? I have been tinkering with the example codes, but the motor behaves erratically, here is my code:

```
var board = new five.Board();

board.on("ready", function() {

  /**
   * In order to use the Stepper class, your board must be flashed with
   * either of the following:
   *
   * - AdvancedFirmata https://github.com/soundanalogous/AdvancedFirmata
   * - ConfigurableFirmata https://github.com/firmata/arduino/releases/tag/v2.6.2
   *
   */

  var stepper = new five.Stepper({
    type: five.Stepper.TYPE.FOUR_WIRE,
    stepsPerRev: 200,
    pins: {
      motor1: 8,
      motor2: 9,
      motor3: 10,
      motor4: 11
    }
  });

  stepper.rpm(200).cw().step(4000, function() {
    console.log("Done moving CCW");

  });
});
```
This is for a bipolar stepper like [this](https://www.sparkfun.com/products/9238) and the L298N driver.

I guess what I am asking is if anybody has this driver working with J5 and a 4 wire bipolar stepper

Thanks in advance and sorry for the hassle.

Carlos.

dtex, 2017-05-01:
Not a hassle at all. I'm using that exact motor controller to control a similar stepper and yes I use TYPE.FOUR_WIRE. In what way is it behaving erratically? What's it do when you call step()?

calugo, 2017-05-02:
Hello!

Well, it is a bit odd, the motor "hums" but does not move at all, however
when i flash a standard *ino code which is equivalent (perform a number of
steps cw and ccw) it works just fine.

I wonder if the pin order in the node code is somehow different.... I
really cannot tell.

Any suggestions?

Regards,

Carlos

calugo, 2017-05-02:
Oh, terrible mistake! I had a loose wire,

Thanks a lot anyway!