rwaldron/johnny-five #1063: Deprecate use of ".Array" for "Collection" classes

rwaldron, 2016-03-17:
This was a bad naming choice, because it's not an `Array`. I'm sorry.

reconbot, 2016-03-17:
I like `collection`, it fits in more places then batch and is a pretty common term for this sort of thing.

dtex, 2016-03-17:
Collection is good, and they all depend on the collections mixin now right?

What about pluralized class names (Servos, Leds, ESCs, etc)? Those exists and don't we have:

````js
Thing();
Things(); // uses /mixins/collections.js

Thing.Array = Things;
````

In most "collectable" classes?

rwaldron, 2016-03-17:
> What about pluralized class names (Servos, Leds, ESCs, etc)? Those exists and don't we have:

Yes, those will remain. I just want to address this crappy naming

frxnz, 2016-03-17:
I'll back `Collection` as well. I think it's probably the most familiar from a programming perspective.

dtex, 2016-03-17:
If we're deprecating ```foo.Array``` why do we need to add ```foo.Collection``` given that we already have pluralized class names? If there is one "right" way it's easier to maintain, document and learn.

rwaldron, 2016-03-17:
> If we're deprecating foo.Array why do we need to add foo.Collection given that we already have pluralized class names? If there is one "right" way it's easier to maintain, document and learn.

Well, TBH... Because I'm hyper-nitpicky. I received an email about `Sensor` collections with some _very_ compelling example code (made me see the need from an important perspective, more on that later) and | started thinking about implementation, which led to naming, which lead to icky sounding things like `Motions` and `Proximities` :| Maybe my approach needs rethinking...

dtex, 2016-03-17:
> Maybe my approach needs rethinking

You mean the part where you ask everyone else :-P

So the pickle here is that instead of being named for a thing (lidar, sonar), these classes are named for types of things (proximity) and pluralizing a "types of thing" name doesn't feel right.

In that case, I say "Collection" is perfect. It's also the perfect name for a bunch of dissimilar devices that are used as a group (i.e. In an animation() controlling a robot with LEDs, steppers, and servos all working in concert), but when we have a group of clear and explicit things of the same type (Servos, ESCs, LEDs) we should just pluralize it.

scottgonzalez, 2016-03-17:
+1 for Collection

rwaldron, 2016-03-17:
@dtex I like the way you've rationalized this

Thanks everyone for weighing in... I don't think the other options have much of a leg to stand on.

closed 2016-04-02 by rwaldron