rwaldron/johnny-five #1065: Stepper class can't use digital pin 0

alexcroox, 2016-03-22:
When my pins are [0,1,2,3] the stepper class fails with `Error: Stepper.TYPE.FOUR_WIRE expects:pins.motor1,pins.motor2,pins.motor3,pins.motor4`

when they are [4,5,6,7] it runs fine

https://github.com/rwaldron/johnny-five/blob/5d3901940e20c33ab36dc710ee9cbb5ecb19de50/lib/stepper.js#L211

it's doing a ! false check instead of a typeof undefined. Pin 0 would trigger that false flag.

Is there another way to define pin 0 without it triggering a FALSE condition? I've tried 'D0' with no luck.

rwaldron, 2016-03-22:
Even if this were changed to a more robust pin check (which it should be) nothing can be used with digital pin 0 or 1, ever. This isn't something that can be fixed or changed, as there is nothing to fix or change because 0 is RX and 1 is TX of the serial line that carries the host/client communication.

rwaldron, 2016-03-22:
>  to a more robust pin check (which it should be) 

Strike that, it would likely interfere with boards that _do_ allow 0 and 1 access (which is why Johnny-Five has historically been "hands-off" about such things).

alexcroox, 2016-03-22:
Ah I didn't know that apologies! This morning I did a bit of Googling and looking through the pin docs here to work out if I could use Pin 1/0 on the Uno as I could see the TX/RX on the pin labels. I couldn't find a definitive answer so wrongly assumed I could. Sorry about that!

dtex, 2016-03-22:
> it would likely interfere with boards that do allow 0 and 1 access

Confused by this. Don't we need to change the check in order for boards that allow 0 and 1 access to ever work?

soundanalogous, 2016-03-22:
At some point I may open up pins 0 and 1 when using Ethernet, WiFi or BLE as the transport.

rwaldron, 2016-03-22:
@dtex we can't universally block pins 0 and 1 because other boards, like Tessel 2, don't have that restriction

rwaldron, 2016-03-22:
@alexcrook no apologies necessary! We can update the Board docs to include some information about it :)

rwaldron, 2016-03-22:
@dtex I don't know of any other boards that have stepper support

rwaldron, 2016-03-22:
I guess we can fix this on the possibility that others exist ¯\_(ツ)_/¯

soundanalogous, 2016-03-22:
Also many of the newer Arduino boards (Zero, 101, etc) use CDC serial over USB and pins 0 and 1 are a true UART and not affected by Serial over USB.

dtex, 2016-03-22:
> we can't universally block pins 0 and 1

I think there is a disconnect. I'm not looking to block those, just change that conditional to see if valid values of the correct type were actually passed instead of checking for truthiness.

Resseguie, 2016-03-23:
I agree with @dtex that we should fix the truthiness test so we don't forget about it when/if future boards are supported without that limitation.

closed 2016-03-23 by rwaldron