rwaldron/johnny-five #1067: board doesn't get ready if arduino is connected after johnny-five started

stijnbuurman, 2016-03-23:
I am using johnny-five in combination with NW.js. If i start the program when the arduino is already connected everything works fine. If I connect the arduino after the NW started it does trigger the "connect" event (with the right port) but never the "ready" event. Also manually checking "isReady" does not work.

Is there a way to fix this?

rwaldron, 2016-03-24:
Johnny-Five will actively wait for a board to be connected if none is found when the `Board` constructor is called to create a new board instance—so, I think there is a bug somewhere ;)

It will be easier for both of us if we're following the same baseline. @cotejp recently wrote this updated guide: http://tangiblejs.com/posts/nw-js-johnny-five-arduino-wicked-trio so I think we should both follow this, then we'll be on the same page, sound good?

cotejp, 2016-03-24:
Unfortunately, my tutorial does not explicitely address this situation which I was not aware of. I'll try to find some time to look at it.

rwaldron, 2016-03-24:
@cotejp sure, I just wanted to use it as the baseline for reproducing an NW.js + Johnny-Five environment :)

stijnbuurman, 2016-03-24:
@rwaldron I actually did follow @cotejp 's guide (thanks for that). Could browser-serialPort be the problem?
I'm actually making a drag-and-drop visual programming environment for a study project. It's going to be used by children so it would be nice they don't have to restart the program if they connect the Arduino too late. Thanks for looking at it!

rwaldron, 2016-03-24:
@stijnbuurman I didn't get a chance to dig into this today, but I've got it lined up for first thing tomorrow morning. I really appreciate your patience :)

> Could browser-serialPort be the problem?

Hm. Interesting possibility—this should only be used when Johnny-Five is run from a chrome extension, but it's worth investigating.

rwaldron, 2016-03-24:
Somehow I completely overlooked that part in @cotejp's guide 0_o... and this is very likely the issue. It's possible that browser-serialport isn't giving me the information I need that allows for detecting the presence (or not) of plugged in boards. That's normally done with node-serialport's `list()` method, which browser-serialport also supports: https://github.com/garrows/browser-serialport#list-ports, but there must be something going on?

rwaldron, 2016-03-24:
Can you add this to your program and post back what it outputs: 

```
console.log(process.versions.nw);
```

cotejp, 2016-03-25:
I tried and I have no problem making it work on my end. Basically, I start the application (which immediately tries to create a `Board` object), wait a few seconds and then plug in the Arduino. The `ready` event is triggered as expected.

Sample code:

```
nw.require("nwjs-j5-fix").fix();
var five = nw.require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  console.log("I'm ready!");
});
```

If you try the code above, do you still get the problem ?

stijnbuurman, 2016-03-25:
Hmm this is weird. I tried your sample code which was only slightly different as mine, but it's still not working. 
I am also on Mac with NW.js v0.13. (this is also what process.versions.nw returns)

my node version was 5.8.0 and now 5.9.1 but that doesn't make a difference either..

stijnbuurman, 2016-03-25:
Well apparently the problem was my (genuine) Arduino. I tried a Chinese Arduino and it works as expected! I have no clue why my Arduino did work as expected when it was already connected on the start of the program...

Can i conclude that my Arduino is broken in some way? Did you use a genuine Arduino?

cotejp, 2016-03-25:
That's odd. My tests were done with a genuine Arduino. I think it's hard to conclude anything at this stage. What version of Firmata do you have installed ? I have StandardFirmataPlus.

stijnbuurman, 2016-03-25:
I used the StandardFirmata for both Arduino's. When i try StandardFirmataPlus it still only gets to connected. Also when I restart the NW.js app (and keep it connected), it does get ready so the Arduino is not crashing or something. I think we should just forget about this and I should just use another Arduino!

Thanks a lot!

rwaldron, 2016-04-04:
Sounds like this is resolved, so I'm going to close now. If further issues arise, we can reopen accordingly :)

closed 2016-04-04 by rwaldron