rwaldron/johnny-five #1068: Support For DHT11

denzelwamburu, 2016-03-24:
I know this question has been asked some times back, i would just want to confirm if the library provides support for DHT11 now? or is it HTU21D alone?

rwaldron, 2016-03-24:
Sorry, there is currently no support for the DHT11. There are many other temperature sensors that are supported: 

- LM35
  - [Mouser](http://www.mouser.com/search/ProductDetail.aspx?R=0virtualkey0virtualkeyLM35DZ-NOPB)
  - [TI](http://www.ti.com/product/lm35)
- TMP36
  - [Adafruit](https://www.adafruit.com/products/165?utm_source=j5)
  - [Analog Devices](http://www.analog.com/en/mems-sensors/digital-temperature-sensors/tmp36/products/product.html)
  - [SparkFun](https://www.sparkfun.com/products/10988?utm_source=j5)
- TMP102
  - [SparkFun](https://www.sparkfun.com/products/11931?utm_source=j5)
- DS18B20 (requires [ConfigurableFirmata](https://github.com/firmata/arduino/tree/configurable))
  - [Adafruit](https://www.adafruit.com/products/374?utm_source=j5)
  - [Maxim Integrated](http://www.maximintegrated.com/en/products/analog/sensors-and-sensor-interface/DS18B20.html)
  - [SparkFun](https://www.sparkfun.com/products/245?utm_source=j5)
- MPU6050
  - [Invensense](http://www.invensense.com/products/motion-tracking/6-axis/mpu-6050/)
  - [SparkFun](https://www.sparkfun.com/products/11028?utm_source=j5)
- Grove Thermometer
  - [Seeed Studio](http://www.seeedstudio.com/depot/Grove-Thermometer-Sensor-p-774.html)
- BMP180
  - [Adafruit](https://www.adafruit.com/products/1603?utm_source=j5)
  - [SparkFun](https://www.sparkfun.com/products/11824?utm_source=j5)
  - [Grove](http://www.seeedstudio.com/depot/Grove-Barometer-Sensor-BMP180-p-1840.html?utm_source=j5)
- MPL115A2
  - [Adafruit](https://www.adafruit.com/products/992?utm_source=j5)
  - [SparkFun](https://www.sparkfun.com/products/9721?utm_source=j5)
- MPL3115A2
  - [Adafruit](https://www.adafruit.com/products/1893?utm_source=j5)
  - [Sparkfun](https://www.sparkfun.com/products/11084?utm_source=j5)
    - [SparkFun Weather Shield](https://www.sparkfun.com/products/12081?utm_source=j5)
    - [SparkFun Photon Weather Shield](https://www.sparkfun.com/products/13630?utm_source=j5)
- HTU21D
  - [Adafruit](https://www.adafruit.com/product/1899?utm_source=j5)
  - [Sparkfun](https://www.sparkfun.com/products/12064?utm_source=j5)
- MCP9808
  - [Adafruit](https://www.adafruit.com/product/1782?utm_source=j5)
- SI7020
  - [Tessel Climate](https://shop.tessel.io/Modules/Climate%20Module?utm_source=j5)
- MS5611
  - [Amazon](http://www.amazon.com/MS5611-High-resolution-Atmospheric-Pressure-Module/dp/B00F4P6LKE?utm_source=j5)
- TH02
  - [Grove](http://www.seeedstudio.com/depot/Grove-TemperatureHumidity-Sensor-HighAccuracy-Mini-p-1921.html?utm_source=j5)

denzelwamburu, 2016-03-24:
@rwaldron i needed a hygrometer sensor apart form HTU21D

rwaldron, 2016-03-24:
All of the supported components are listed on each API page, so for Hygrometer, there are currently two http://johnny-five.io/api/hygrometer/ 

I will look around for others that can be added for more device diversity

rwaldron, 2016-03-24:
The best I can do for now is to provide support via an I2C backpack, which means you'll need another microcontroller, any arduino will do the trick.

rwaldron, 2016-03-24:
In v0.9.35...

- https://github.com/rwaldron/johnny-five/blob/master/docs/multi-DHT11_I2C_NANO_BACKPACK.md
- https://github.com/rwaldron/johnny-five/blob/master/docs/hygrometer-DHT11_I2C_NANO_BACKPACK.md
- https://github.com/rwaldron/johnny-five/blob/master/docs/temperature-DHT11_I2C_NANO_BACKPACK.md

[image]

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var hygrometer = new five.Hygrometer({
    controller: "DHT11_I2C_NANO_BACKPACK"
  });

  hygrometer.on("change", function() {
    console.log("Hygrometer");
    console.log("  relative humidity : ", this.relativeHumidity);
    console.log("--------------------------------------");
  });
});
```

soundanalogous, 2016-03-24:
nice! especially considering the odd timing delays the DHT11 requires. Offloading it all to I2C keeps things unblocked in the main processor.

rwaldron, 2016-03-24:
Yeah, I figured it was about time I addressed this sensor ;) 

My next iteration will be to allow specifying either the DHT11 or DHT22

denzelwamburu, 2016-03-25:
Great, just what I needed 🙌.

denzelwamburu, 2016-03-25:
For the meantime I will close the issue, let me try to implement it, will reopen the issue incase of anything.

closed 2016-03-25 by denzelwamburu

rwaldron, 2016-03-25:
@denzelwamburu sounds good!

eugenebessan225, 2020-08-28:
hello! Please, i'm working on an automatic systèm for greenhouse. it's my project in order to obtain my bachelor and i face to certain difficulty with johnny five. My DHT22 sensor returns 0 as result and i d'int know what's happened. help me pls

dtex, 2020-08-28:
It's hard to help without more information. Can you share code and a picture of your wiring?

eugenebessan225, 2020-08-28:
`const { Board, Thermometer } = require("johnny-five");
const board = new Board();

board.on("ready", () => {
  const thermometer = new Thermometer({
    controller: "DHT22_I2C_NANO_BACKPACK"
  });

  thermometer.on("change", () => {
    const {celsius, fahrenheit, kelvin} = thermometer;
    console.log("Thermometer");
    console.log("  celsius      : ", celsius);
    console.log("  fahrenheit   : ", fahrenheit);
    console.log("  kelvin       : ", kelvin);
    console.log("--------------------------------------");
  });
});`
Here is my code

eugenebessan225, 2020-08-28:
i've just copy this here. And in arduino, i'he televerse the standard firmata code.

> It's hard to help without more information. Can you share code and a picture of your wiring?

eugenebessan225, 2020-08-28:
i'm trying to upload the wiring but i can't.

dtex, 2020-08-28:
>  returns 0 as result 

For Celsius only right? You're getting 32 and 273 for the others?