rwaldron/johnny-five #1069: Not able to send / Receive data using HC-05 Bluetooth module

sarkarstanmoy, 2016-03-24:
Hi,

I have followed [https://github.com/rwaldron/johnny-five/wiki/Getting-Started-with-Johnny-Five-and-HC-05-Bluetooth-Serial-Port-Module](url) and able to configure bluetooth module. I have also followed node-serial [https://github.com/voodootikigod/node-serialport](url) and use below code

```javascript

var SerialPort = require("serialport").SerialPort
var serialPort = new SerialPort("COM3", {
  baudRate: 57600 ,
        dataBits: 8,
        parity: 'none',
        stopBits: 1,
        //parser: SerialPort.parsers.readline("\n"),
        flowControl: false
}, false); // this is the openImmediately flag [default is true]

serialPort.open(function (error) {
  if ( error ) {
    console.log('failed to open: '+error);
  } else {
    console.log('open');
    serialPort.on('data', function(data) {
      console.log('data received: ' + data);
    });
    serialPort.write("ls\n", function(err, results) {
      console.log('err ' + err);
      console.log('results ' + results);
    });
  }
});
```
and got below output

```
open
err undefined
results 3
data received: ��yStandardFirmata.ino�
```

But when I connected to Bluetooth using BlueTerm app in my android phone  and try to send data nothing is displayed in console. Can you please help me on this. I have also followed #165 issue but still not able to figure out the solution. Request you to kindly help me on this.

sarkarstanmoy, 2016-03-31:
Any update on this issue

soundanalogous, 2016-03-31:
First of all, the issue you are experiencing appears to be related to [node-serialport](https://github.com/voodootikigod/node-serialport), not johnny-five. It may be helpful to open this issue for that project.

You are receiving the `REPORT_FIRMWARE` query results upon establishing the serial connection as expected: `����y��StandardFirmata.ino�`, the console just isn't printing the non ascii characters. However, I'm not sure why you're getting the error on serial write.

sarkarstanmoy, 2016-04-02:
Thanks for the reply. As you have suggested I will raise this issue to node-serialport team.

dtex, 2018-01-05:
Hi @sarkarstanmoy Is it okay to close this issue?

dtex, 2018-01-16:
Hi @sarkarstanmoy 

I'm going to close this since we haven't heard back. Feel free to re-open if necessary.

closed 2018-01-16 by dtex