rwaldron/johnny-five #1072: LSM9DS0

teuteuguy, 2016-03-28:
Any chance that Johnny Five can support this sensor ?

9 degrees of freedom

rwaldron, 2016-03-28:
I started working on this last week, I will update when I have something worth testing.

teuteuguy, 2016-04-30:
hey rwaldron, any news on this by any chance ?

gpkc, 2016-05-18:
I'd like to be able to use this one too. Any news?

dtex, 2018-01-11:
Hi @teuteuguy and @gpkc ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the LSM9DSO there.

closed 2018-01-11 by dtex