rwaldron/johnny-five #1075: GPS class conversion problem at gps.js #317,318

pacozaa, 2016-03-31:
- I'm trying to receive the NMEA massage from my custom GPS board and it works fine with Arduino IDE.

[image]

- When come to johnny-five event 

> change or sentences not trigger. 

So I go to the gps.js class and debug it if the data come. Yes it is, but not properly parse.
[image]
[image]

and I try to change something on the gps.js but still not working for me
[image]
[image]

- Maybe because I use level shifter to convert the signal from GPS board to Arduino MEGA, but Why Arduino IDE can print it out?

dtex, 2016-03-31:
@pacozaa Can you try this code? It's just grabbing data via firmata and bypassing Johnny-five. 
````
var Board = require("firmata");
var board = new Board("/dev/cu.usbmodem1421", function() {
  console.log("ready");
  // Arduino is ready to communicate

  board.serialConfig({
    portId: 0x03,
    baud: 4800
  });

  var inString = [];

  board.serialRead(0x03, function(data) {
    if (data === 0x13) {
      console.log(inString);
    } else {
      inString.push(data);
    }

    if (inString.length > 100) {
      console.log("delimiter not found");
      console.log(JSON.stringify(inString));
      inString = [];
    }
  });

});
````
When you post the results, please copy and paste a couple of arrays instead of sending a screenshot. I'd like to experiment with some bit shifting.

pacozaa, 2016-03-31:
@dtex I've try your code here is the output.

> ready
[[25],[89],[26],[25],[230],[6],[198],[214],[99],[198],[6],[6],[6],[6],[166],[70],[134],[214],[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42],[53],[51],[13],[10],[36],[71],[80],[71],[71],[65],[44],[44],[44],[44],[44],[44],[48],[44],[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44]]
[[44],[48],[48],[48],[48],[42],[52],[56],[13],[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42],[53],[51],[13],[10],[36],[71],[80],[71],[71],[65],[44],[44],[44],[44],[44],[44],[48],[44],[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44],[44],[48],[48],[48],[48],[42],[52],[56],[13]]
[[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42],[53],[51],[13],[10],[36],[71],[80],[71],[71],[65],[44],[44],[44],[44],[44],[44],[48],[44],[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44],[44],[48],[48],[48],[48],[42],[52],[56],[13],[10],[36],[71],[80],[71],[83],[65],[44],[65]]
[[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42],[53],[51],[13],[10],[36],[71],[80],[71],[71],[65],[44],[44],[44],[44],[44],[44],[48],[44],[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44],[44],[48],[48],[48],[48],[42],[52],[56],[13],[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44]]
[[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[71],[83],[86],[44],[51],[44],[49],[44],[49],[50],[44],[48],[49],[44],[48],[48],[44],[48],[48],[48],[44],[44],[48],[50],[44],[48],[48],[44],[48],[48],[48],[44],[44],[48],[51],[44],[48],[48],[44],[48],[48],[48],[44],[44],[48],[52],[44],[48],[48],[44],[48],[48],[48],[44],[42],[55],[67],[13],[10],[36],[71],[80],[71],[83],[86],[44],[51],[44],[50],[44],[49],[50],[44],[48],[53],[44],[48],[48],[44],[48],[48],[48],[44],[44],[48]]
[[54],[44],[48],[48],[44],[48],[48],[48],[44],[44],[48],[55],[44],[48],[48],[44],[48],[48],[48],[44],[44],[48],[56],[44],[48],[48],[44],[48],[48],[48],[44],[42],[55],[55],[13],[10],[36],[71],[80],[71],[83],[86],[44],[51],[44],[51],[44],[49],[50],[44],[48],[57],[44],[48],[48],[44],[48],[48],[48],[44],[44],[49],[48],[44],[48],[48],[44],[48],[48],[48],[44],[44],[49],[49],[44],[48],[48],[44],[48],[48],[48],[44],[44],[49],[50],[44],[48],[48],[44],[48],[48],[48],[44],[42],[55],[49],[13],[10],[36],[71],[80]]
[[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42],[53],[51],[13],[10],[36],[71],[80],[71],[71],[65],[44],[44],[44],[44],[44],[44],[48],[44],[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44],[44],[48],[48],[48],[48],[42],[52],[56],[13],[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44]]
[[44],[44],[44],[44],[44],[44],[44],[78],[42],[53],[51],[13],[10],[36],[71],[80],[71],[71],[65],[44],[44],[44],[44],[44],[44],[48],[44],[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44],[44],[48],[48],[48],[48],[42],[52],[56],[13],[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42]]
[[53],[51],[13],[10],[36],[71],[80],[71],[71],[65],[44],[44],[44],[44],[44],[44],[48],[44],[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44],[44],[48],[48],[48],[48],[42],[52],[56],[13],[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42],[53],[51],[13],[10],[36],[71],[80],[71],[71]]
[[65],[44],[44],[44],[44],[44],[44],[48],[44],[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44],[44],[48],[48],[48],[48],[42],[52],[56],[13],[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42],[53],[51],[13],[10],[36],[71],[80],[71],[71],[65],[44],[44],[44],[44],[44],[44],[48],[44]]
[[48],[48],[44],[44],[44],[77],[44],[48],[46],[48],[44],[77],[44],[44],[48],[48],[48],[48],[42],[52],[56],[13],[10],[36],[71],[80],[71],[83],[65],[44],[65],[44],[49],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[42],[49],[69],[13],[10],[36],[71],[80],[71],[83],[86],[44],[51],[44],[49],[44],[49],[50],[44],[50],[57],[44],[48],[48],[44],[48],[48 [... cut ...]
[[44],[44],[48],[56],[44],[48],[48],[44],[48],[48],[48],[44],[42],[55],[55],[13],[10],[36],[71],[80],[71],[83],[86],[44],[51],[44],[51],[44],[49],[50],[44],[48],[57],[44],[48],[48],[44],[48],[48],[48],[44],[44],[49],[48],[44],[48],[48],[44],[48],[48],[48],[44],[44],[49],[49],[44],[48],[48],[44],[48],[48],[48],[44],[44],[49],[50],[44],[48],[48],[44],[48],[48],[48],[44],[42],[55],[49],[13],[10],[36],[71],[80],[82],[77],[67],[44],[44],[86],[44],[44],[44],[44],[44],[44],[44],[44],[44],[44],[78],[42],[53]]

dtex, 2016-03-31:
Cool, I'll see what I can make of these

pacozaa, 2016-03-31:
I found the pattern a little bit. These value is decimal.

Just search on 36(ctrl+f type 36)

> [36],[71],[80],[71],[83],[65],[44]

is

> $GPGSA,

dtex, 2016-03-31:
Yep, those are totally NMEA sentences. I am completely stumped by why it would come out one way in J5 and another way in Firmata. The serialRead callbacks are essentially the same.

Would you mind making this change to gps.js down at line 318:

````js
this.io.serialRead(state.portId, function(data) {

    inString.push(data);
    if (inString.length > 100) {
      console.log(JSON.stringify(inString));
      inString = [];
    }

    input += new Buffer(data).toString("ascii");
    var sentences = input.split("\r\n");
    ...
````
I feel bad that you are having to keep checking things and I can't just fix it for you.

pacozaa, 2016-03-31:
@dtex Nope. I'm working on it. Hey feel free to ask me to checking things. We are making j5 better.

pacozaa, 2016-03-31:
The output of above code is in [this link](https://cloudup.com/cpXhWVZqyNc).

I save it in the Cloudup this time because the output is too big. 

Thank you.

rwaldron, 2016-04-02:
@dtex 

```js
  board.serialRead(0x03, function(data) {
    inString.push(data);

    if (inString.length > 100) {
      console.log(JSON.stringify(inString));
      inString = [];
    }
  });
```

Isn't that pushing an array into an array? Did you mean to do: 

```js
inString = inString.concat(data);
```

@pacozaa what is the exact device being used? Can someone link me to it?

pacozaa, 2016-04-03:
@rwaldron The device information is [here](https://cloudup.com/cI4LJwfWvTv).

pacozaa, 2016-04-04:
Hi all, I replace `inString.push(data);` by  `inString = inString.concat(data);` and This is the output.

> ready
[36,71,80,71,71,65,44,48,49,48,55,52,49,46,55,51,50,44,44,44,44,44,48,44,48,48,44,44,44,77,44,48,46,48,44,77,44,44,48,48,48,48,42,53,51,13,10,36,71,80,71,83,65,44,65,44,49,44,44,44,44,44,44,44,44,44,44,44,44,44,44,44,42,49,69,13,10,36,71,80,82,77,67,44,48,49,48,55,52,49,46,55,51,50,44,86,44,44,44,44,44]
[44,44,49,52,49,49,49,49,44,44,44,78,42,52,68,13,10,36,71,80,71,71,65,44,48,49,48,55,52,50,46,55,51,48,44,44,44,44,44,48,44,48,48,44,44,44,77,44,48,46,48,44,77,44,44,48,48,48,48,42,53,50,13,10,36,71,80,71,83,65,44,65,44,49,44,44,44,44,44,44,44,44,44,44,44,44,44,44,44,42,49,69,13,10,36,71,80,82,77,67,44]
[48,49,48,55,52,50,46,55,51,48,44,86,44,44,44,44,44,44,44,49,52,49,49,49,49,44,44,44,78,42,52,67,13,10,36,71,80,71,71,65,44,48,49,48,55,52,51,46,55,51,48,44,44,44,44,44,48,44,48,48,44,44,44,77,44,48,46,48,44,77,44,44,48,48,48,48,42,53,51,13,10,36,71,80,71,83,65,44,65,44,49,44,44,44,44,44,44,44,44,44,44]
[44,44,44,44,44,42,49,69,13,10,36,71,80,71,83,86,44,51,44,49,44,49,50,44,49,50,44,48,48,44,48,48,48,44,50,48,44,50,48,44,48,48,44,48,48,48,44,50,54,44,50,49,44,48,48,44,48,48,48,44,50,48,44,48,49,44,48,48,44,48,48,48,44,42,55,70,13,10,36,71,80,71,83,86,44,51,44,50,44,49,50,44,48,50,44,48,48,44,48,48,48]
[44,44,48,51,44,48,48,44,48,48,48,44,44,48,52,44,48,48,44,48,48,48,44,44,48,53,44,48,48,44,48,48,48,44,42,55,66,13,10,36,71,80,71,83,86,44,51,44,51,44,49,50,44,48,54,44,48,48,44,48,48,48,44,44,48,55,44,48,48,44,48,48,48,44,44,48,56,44,48,48,44,48,48,48,44,44,48,57,44,48,48,44,48,48,48,44,42,55,65,13,10]
[36,71,80,82,77,67,44,48,49,48,55,52,51,46,55,51,48,44,86,44,44,44,44,44,44,44,49,52,49,49,49,49,44,44,44,78,42,52,68,13,10,36,71,80,71,71,65,44,48,49,48,55,52,52,46,55,51,48,44,44,44,44,44,48,44,48,48,44,44,44,77,44,48,46,48,44,77,44,44,48,48,48,48,42,53,52,13,10,36,71,80,71,83,65,44,65,44,49,44,44,44]

pacozaa, 2016-04-04:
I edit the code as above comment but this time I edited in GPS class
at line 311 in `lib/gps.js`
```javascript
GPS.prototype.listen = function() {

  var state = priv.get(this);
  var input = "";

  // Start the read loop
  var inString = [];
  this.io.serialRead(3, function(data) {

    inString = inString.concat(data);
    if (inString.length > 100) {
      console.log(JSON.stringify(inString));
      inString = [];
    }
    input += new Buffer(data).toString("ascii");
    var sentences = input.split("\r\n");
    if (sentences.length > 1) {
      for (var i = 0; i < sentences.length - 1; i++) {
        this.parseNmeaSentence(sentences[i]);
      }
      input = sentences[sentences.length - 1];
    }
  }.bind(this));
};

```

and this is the output.

> 1459732958491 Device(s) COM5  
1459732958507 Connected COM5  
1459732961715 Repl Initialized  
[134,120,134,24,120,120,134,30,120,30,224,134,6,120,30,128,120,134,152,134,96,120,248,120,30,152,128,96,120,96,0,230,24,230,230,152,30,30,134,0,120,224,120,120,24,120,30,6,120,134,126,120,120,0,120,134,120,230,224,134,224,134,224,134,224,134,224,134,224,134,224,134,0,120,30,0,120,30,6,120,30,224,134,224,134,224,134,248,152,152,134,96,120,128,158,30,152,128,96,120,96]
[0,230,126,158,96,6,30,134,0,120,224,120,120,24,120,30,24,120,134,126,120,120,102,120,134,224,134,224,134,224,134,224,134,0,120,134,0,120,120,134,224,134,224,134,230,152,224,134,0,120,134,0,120,134,230,152,224,134,224,134,0,120,120,120,120,102,102,120,224,120,30,152,128,96,120,96,0,230,126,158,152,158,0,30,134,6,30,134,6,120,134,224,134,224,134,224,134,224,134,224,134]
[224,134,224,134,224,134,224,134,224,134,224,134,224,134,224,134,224,134,152,134,6,120,6,158,30,152,128,96,120,96,0,230,24,230,230,152,30,30,134,0,120,224,120,120,24,120,30,24,120,134,126,120,120,102,120,134,120,230,224,134,224,134,224,134,224,134,224,134,224,134,224,134,0,120,30,0,120,30,6,120,30,224,134,224,134,224,134,248,152,152,134,96,120,96,230,128,152,128,96,120,96]

dtex, 2016-04-04:
Thanks @pacozaa Rick's right, I was pushing an array into an array but I was still able to use the earlier data by running a search and replace in my text editor. Still the crazy discrepancy between the firmata only implementation and johnny-five.

rwaldron, 2016-04-04:
I'm not sure I understand what the discrepancy is? When I run the following program:

```js
var Board = require("../");

Board.requestPort(function(error, port) {
  if (error) {
    console.log(error);
    return;
  }
  var board = new Board(port.comName);

  board.on("ready", function() {
    console.log("READY");

    var SW_SERIAL0 = board.SERIAL_PORT_IDs.SW_SERIAL0;
    // var maxBytesToRead = 4;

    board.serialConfig({
      portId: SW_SERIAL0,
      baud: 9600,
      rxPin: 2,
      txPin: 3
    });

    board.serialRead(SW_SERIAL0, function(data) {
      console.log(data);
      // console.log(new Buffer(data).toString("ascii"));
    });
  });
});
```

... with any GPS shield that I have available to me, I get output that looks like this: 

(ITEAD Shield 1.1)
```
$ node examples/sw-serial-read-gps.js
READY
[ 70 ]
[ 22 ]
[ 230 ]
[ 70 ]
[ 70 ]
[ 102 ]
[ 86 ]
[ 198 ]
[ 230 ]
[ 99 ]
[ 6 ]
[ 118 ]
[ 54 ]
[ 86 ]
[ 118 ]
[ 230 ]
[ 38 ]
[ 22 ]
[ 38 ]
[ 22, 198, 118 ]
[ 141, 89, 198 ]
[ 6, 150, 198, 22 ]
[ 230, 6, 198, 22, 150 ]
[ 230, 54, 198, 214, 99 ]
[ 214, 54, 70, 230, 54, 198 ]
[ 214, 99, 198, 6, 6, 6 ]
[ 6, 166, 86, 118, 214 ]
[ 10, 36, 71, 80, 71, 83 ]
[ 65, 44, 77, 44, 51, 44 ]
[ 49, 48, 44, 49, 52, 44, 51 ]
[ 50, 44, 50, 53, 44, 49, 56 ]
[... 150 lines cut ...]
(from https://github.com/rwaldron/johnny-five/issues/1075#issuecomment-203753261)
```js
var inString = [];
this.io.serialRead(state.portId, function(data) {

    inString.push(data);
    if (inString.length > 100) {
      console.log(JSON.stringify(inString));
      inString = [];
    }

    input += new Buffer(data).toString("ascii");
    var sentences = input.split("\r\n");
```

... But I don't think that's what you're referring to? Maybe? I don't know. 

....................

Looking at the manual, I have a bunch of maybe dumb questions to ask: 

1. Are you using `baud: 4800` in your Johnny-Five program?
2. Are you using windows?

I've just ordered one of these chips, not sure when I'll get it, but I'll keep you all posted

dtex, 2016-04-04:
@rwaldron Also note that there is a level shifter in line between the GPS and the Arduino, but we get valid data when using just firmata so that's probably not the culprit.

pacozaa, 2016-04-05:
@rwaldron 

1. Yes, I'm using board rate 4800 in Johnny-Five
2.  Yes, I'm using windows.

@dtex  Thank you  :+1:

dtex, 2016-04-05:
Maybe the baud is not getting from the opts object to firmata? That would totally explain everything we are seeing. The chips that we have tested lately all run at the default 9600 baud rate. I'll take a look when I get to the office, in about an hour.

dtex, 2016-04-05:
Yeppers, wrote a test and it is failing. Feeling like a total dumbass right now.  I'll get that fixed and a PR submitted this morning.

pacozaa, 2016-04-05:
@dtex  Thank you very much. I will test this and get back here as soon as possible.

pacozaa, 2016-04-06:
Everything wokrs after I copy and paste both file from @dtex 's [PR](https://github.com/dtex/johnny-five/commit/2c78265dd313e79ddb2f86e639b8d5508d87728d). I will close issue this as soon as his PR accept.

Thank you and very appreciate you guys :+1:

PR #1082 (2016-04-08): Make baud property settable.
Fixes #1075
commit a4108a54 (2016-04-08): Make baud property settable.

Fixes #1075
closed 2016-04-08 by dtex