rwaldron/johnny-five #1077: Where should custom properties go?

bartmichu, 2016-04-03:
I guess it's not unusual to give custom properties to components in all sorts of projects, and it begs to refactor the following code
```
"use strict";

var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function () {
  var pot = new five.Sensor({
    pin: "A0"
  });

  var lowerLimit = 0;
  var upperLimit = 1023;

  pot.within([pot.range[0], lowerLimit], function () {
    console.log("lower limit reached");
  });

  pot.within([upperLimit, pot.range[1]], function () {
    console.log("upper limit reached");
  });
});
```
into something like
```
var pot = new five.Sensor({
  pin: "A0",
  lowerLimit: 0,
  upperLimit: 1023
});
```
or maybe
```
var pot = new five.Sensor({
  pin: "A0",
  customOptions: {
    lowerLimit: 0,
    upperLimit: 1023
  }
});
```
It looks like it doesn't work at the moment, and even if it does then I don't really know if I'm going to introduce some conflicts in the future.

Is there a special property where my stuff should be placed? If not, would you consider specifying on one just for this purpose?

rwaldron, 2016-04-03:
> It looks like it doesn't work at the moment, and even if it does then I don't really know if I'm going to introduce some conflicts in the future.

I can understand that concern. I will reserve the property "custom" for this purpose. It will be available via all component class constructors.

bartmichu, 2016-04-03:
Sounds great, thanks!

rwaldron, 2016-04-03:
@bartmichu I just pushed this out in v0.9.36

closed 2016-04-03 by rwaldron