rwaldron/johnny-five #1080: Barometer: read pressure only once

microspace, 2016-04-04:
I launcher barometer example from here http://johnny-five.io/examples/grove-barometer-edison/
How to read pressure data only ONCE, that exit program?

rwaldron, 2016-04-04:
Just so I understand correctly, you want to completely exit the program after reading the barometer? What happens the read value?

microspace, 2016-04-04:
Yes, I want to completely exit the program. Program should return pressure value.  Is it possible?

rwaldron, 2016-04-04:
Here's a complete example, including a test program to try it out: 

(test.js)
```js
var cp = require("child_process");
var barometer = cp.spawn("node", [ "read-once-barometer" ]);

barometer.stdout.on("data", function (data) {
  console.log("stdout: " + data);
});

barometer.stderr.on("data", function (data) {
  console.log("stderr: " + data);
});

barometer.on("exit", function (code) {
  console.log("child process exited with code " + code);
});
```

(read-once-barometer.js)
```js
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison(),
  // this ensures that no other information is output
  repl: false,
  debug: false,
});

board.on("ready", function() {
  var barometer = new five.Barometer({
    controller: "BMP180"
  });

  barometer.on("data", function() {
    process.stdout.write(this.pressure);
    process.exit();
  });
});
```

closed 2016-04-04 by rwaldron

microspace, 2016-04-05:
Thank you so much!