rwaldron/johnny-five #1081: Line Tracker Example

denzelwamburu, 2016-04-05:
Can you kindly update the line tracker example?
The current one requires a ReflectedArray file which can't seem to be found.
`  ReflectArray = require("./reflect.array"), `
http://johnny-five.io/examples/line-follower/
Or is the file in the library?

denzelwamburu, 2016-04-05:
Am using [SparkFun Line Follower Array](https://www.sparkfun.com/products/13582) and it doesn't seem to have the `emitter` object, anyway to go about it?

henricavalcante, 2016-04-05:
Maybe this one? https://github.com/rwaldron/johnny-five/blob/master/eg/ir-reflect-array.js

rwaldron, 2016-04-06:
I don't even know why that code is there, it's not doing anything.

rwaldron, 2016-04-06:
Unfortunately, the `QRE1113` is a completely different protocol that's not supported by the `IR.Reflect.Array` class. That's not to say that it _can't_ be supported, just that the work hasn't been done yet because no one has ever asked about it (that I can recall). 

@BrianGenisio any thoughts?

rwaldron, 2016-04-06:
I just ordered one of these components, I'll let you know when it arrives and keep you posted on support progress.

denzelwamburu, 2016-04-07:
@rwaldron Okay, thanks.

BrianGenisio, 2016-04-07:
The emitter pin is something that the Pol0lu sensor has.  It looks like this SparkFun emitter is always on.  Looks like a pretty straight-forward I2C interface to write against, but @rwaldron is right... it isn't currently supported.  Only the Analog Pololu sensor has support currently, but a clean I2c interface would be really nice for that sensor (less wires FTW!)

I think we are talking about pulling the Reflection Array into a separate J5 plugin.    I'm ordering one also so I can hack on it.

BrianGenisio, 2016-05-16:
@denzelwamburu I have a POC of the SparkFun reflectance array working with Johnny-Five.  Can you pull my fork here, and see if it works for you? https://github.com/BrianGenisio/johnny-five/tree/sen-13582

You don't need `pins` or `enable` with this component, but you need to tell it which controller to use.  An example:

```js
five.Board().on("ready", function() {
  var eyes = new five.IR.Reflect.Collection({
    controller: 'SEN_13582',
    autoCalibrate: true
  });

  eyes.on("line", function(line) {
    console.log("line: ", line);
  });
});
```

denzelwamburu, 2016-05-18:
@BrianGenisio  Great thank you. Let me check it out then i will get back to you :)

BrianGenisio, 2017-01-10:
@denzelwamburu Did you ever try this branch to see if the code here works for you?

@rwaldron I just realized that I have a mostly working driver sitting in that branch for that sensor, which also enables that component to add other sensor array variants.  I forgot I wrote it.  Hoping to get this resurrected some time soon.