rwaldron/johnny-five #1086: SerialPort on Galileo-io

ortizvinicius, 2016-04-07:
Hi, im using johnny-five with galileo-io in a galileo gen 2 running yocto, i am trying to set a serial communication with SerialPort module, but i am getting the error: "**Illegal instruction**"

The code:

`var Serialport = require("serialport").SerialPort;
var five = require("johnny-five");
var Galileo = require("galileo-io");
var board = new Galileo();

board.on("ready", function() {
  var sp = new Serialport("/dev/ttyS0", {
    baudRate: 57600
  });

  sp.on("open", function() {
    console.log("Port is open!");

    // Once the port is open, you may read or write to it.
    sp.on("data", function(data) {
      console.log("Received: ", data);
    });

    setInterval(function(){ sp.write(new Buffer(["0120000003"])) }, 5000);
  });
});
`

ortizvinicius, 2016-04-08:
Same result:
```
1460118111287 Device(s) Intel Galileo Gen 2 
1460118111426 Connected Intel Galileo Gen 2 
1460118111444 Repl Initialized 
\>> Illegal instruction
```

About new Buffer(["0120000003"]) Im trying to write the string "0120000003", the correct way is new Buffer("0120000003") ?

rwaldron, 2016-04-08:
@ortizvinicius that depends, are you trying to send the string of characters "0120000003"? Or series of bytes that is comprised of those values?

```
> new Buffer("0120000003")
<Buffer 30 31 32 30 30 30 30 30 30 33>
> new Buffer([0,1,2,0,0,0,0,0,0,3])
<Buffer 00 01 02 00 00 00 00 00 00 03>
```

ortizvinicius, 2016-04-08:
string of characters

rwaldron, 2016-04-08:
Got it—then give that a try :)

ortizvinicius, 2016-04-08:
both give the same result :/

rwaldron, 2016-04-08:
Can you paste the entire output? 

Then, try just this: 

```js
var Serialport = require("serialport").SerialPort;
var sp = new Serialport("/dev/ttyS0", {
  baudRate: 57600
});

sp.on("open", function() {
  console.log("Port is open!");

  // Once the port is open, you may read or write to it.
  sp.on("data", function(data) {
    console.log("Received: ", data);
  });
  
  sp.write(new Buffer("0120000003"));
});
```

And paste the result as well

ortizvinicius, 2016-04-08:
For the first the output is:
```
1460118111287 Device(s) Intel Galileo Gen 2 
1460118111426 Connected Intel Galileo Gen 2 
1460118111444 Repl Initialized 
\>> Illegal instruction
```

For the last is just illegal instruction

rwaldron, 2016-04-08:
Ok, so that means this bug isn't in Johnny-Five or Galileo-IO. Let's keep narrowing it down... try this: 

```js
var Serialport = require("serialport").SerialPort;
var sp = new Serialport("/dev/ttyS0", {
  baudRate: 57600
});
```

(and nothing else)

ortizvinicius, 2016-04-08:
illegal instruction

ortizvinicius, 2016-04-08:
with this last code i got illegal instruction too

fivdi, 2016-04-08:
It looks like there is something wrong with binaries somewhere. Perhaps Node.js binaries or serialport binaries.

rwaldron, 2016-04-08:
@ortizvinicius thanks for your patience, one last thing to try before we talk about refiling this over at [node-serialport](https://github.com/voodootikigod/node-serialport/issues). 

Try: 

```js
var Serialport = require("serialport").SerialPort;
var sp = new Serialport("/dev/ttyS0");
```

Then: 

```js
var Serialport = require("serialport").SerialPort;
```

Then, at the terminal, type: 

```
node -e "console.log(require('fs').existsSync('/dev/ttyS0'))";
```

ortizvinicius, 2016-04-08:
Thank you.

Well, the first shows illegal instruction, the second shows nothing, and the last says only "true"

rwaldron, 2016-04-08:
This is very helpful, let's first ping @reconbot and have him read up on the thread and then help us figure out the move forward

henricavalcante, 2016-04-08:
I dont know if it can help but have you tried to run as root access with `sudo`?

ortizvinicius, 2016-04-08:
@henricavalcante sudo dont work in yocto

reconbot, 2016-04-08:
Can you try something for me?
```
var Serialport = require("serialport").SerialPort;
var sp = new Serialport("/dev/ttyS0", console.log);
```
And give the output?

Also what version of yocto are you running on the Galileo?

fivdi, 2016-04-08:
@reconbot is it possible that it's happening because pre-compiled serialport binaries are being installed but that the binaries don't or can't work on a Galileo?

Is it possible to somehow force serialport to be compiled rather than have pre-compiled binaries installed when  `npm install serialport` is called?

ortizvinicius, 2016-04-08:
Hi @reconbot the code outputs only "Illegal Instruction"

I'm not sure of how to see the version haha, but i think is 201409031130

rwaldron, 2016-04-08:
> is it possible that it's happening because pre-compiled serialport binaries are being installed but that the binaries don't or can't work on a Galileo?

As far as I can tell this is exactly the problem. When I was telling @reconbot about this (he's about 3 feet away from me), I remembered when I wrote [this](https://bocoup.com/weblog/intel-galileo-autonomous-navigation-rover), I was unable to use Serialport and ended up using `fs.createReadStream`.

reconbot, 2016-04-08:
@ortizvinicius Now, please try the latest beta version of serialport, we've recently made some changes and it would be helpful to know if they change anything for you. See https://github.com/voodootikigod/node-serialport/issues/733 for more details.

`serialport@2.0.7-beta5`

@fivdi we don't yet have precompiled arm binaries so I don't think that would be the case, but the binary does seem to be compiled with unsupported cpu instructions. `node-pre-gyp rebuild` would rebuild the binary. (you might have to fish the command out of your `node_modules/.bin/` directory)

ortizvinicius, 2016-04-08:
Instaled the beta and the outputs is illegal instruction again

reconbot, 2016-04-08:
Alright, lets move this to the right project. Please open a [new issue](https://github.com/voodootikigod/node-serialport/issues/new) on node-seriaport and we'll debug there. I have a galileo at home that might still work. I know for sure we work on an edison, but you might have to do some tricks to get it to use the older system.

fivdi, 2016-04-08:
The Galileo 2 is from Intel so it's not going to have an ARM processor!!! It has an Intel Quark SoC X1000 processor which supports the Pentium instruction set. I'm unsure how compatible this processor is with the processors in PCs but would imagine that the serialport install process detects it as an Intel processor and installs pre-compiled Intel binaries.

reconbot, 2016-04-08:
Ahh good point! That's a mistake on my part. We should still move the issue over to the right project, but Did the rebuild work?

fivdi, 2016-04-08:
I haven't tried yet. I haven't used a Galileo 2 since sometime early last year but will take a look and see.

ortizvinicius, 2016-04-09:
Moved to https://github.com/voodootikigod/node-serialport/issues/747 
Thank you guys!

closed 2016-04-09 by ortizvinicius