rwaldron/johnny-five #1087: Docs trouble with scale

reconbot, 2016-04-08:
I'm running the following with the latest johnny-five with an arduino.

```js

'use strict';
const five = require('johnny-five');
let board = new five.Board();
board.on('ready', function() {
  let led = new five.Led(11);
  let pot = new five.Sensor('A0');

  pot.scale(0, 255).on('change',(brightness) =>{
    console.log(brightness, this.value);
    led.brightness(brightness);
  });
});
```

`brightness` isn't being scaled, `this.value` is. The [docs](http://johnny-five.io/api/sensor/#analog-sensor) suggest that `brightness` should be scaled. I think the docs are just wrong.

bartmichu, 2016-04-08:
If it's from the wiki then I believe it's my fault - I apologize if that's the case, I will try and fix it right away.

bartmichu, 2016-04-08:
Wiki is fixed now, I guess it will propagate to the website soon.

@reconbot please use ```this.value``` as per documentation before I changed it.

rwaldron, 2016-04-08:
@bartmichu thanks for the super fast fix and follow up! :+1:

rwaldron, 2016-04-08:
Site is redeployed

closed 2016-04-08 by rwaldron