rwaldron/johnny-five #1088: I2C issue with Explorer Hat / Raspberry Pi3

punitganshani, 2016-04-09:
I am using `Explorer Hat` with RPI3 and it has GPIO as below
https://github.com/pimoroni/explorer-hat/blob/master/documentation/GPIO-pins.md

I am trying to interface with `HT16K33` which I believe supports I2C. As per GPIO pins the I2C address should be 0x48.

`i2cdetect -y 1` shows 2 addresses `0x28` and `0x48`

But when I access the`HT16K33` LED matrix, I get this error

```
/home/pi/code/matrix/node_modules/johnny-five/lib/led/ledcontrol.js:500
   throw new Error("Invalid HT16K33 controller address: " + address);
```

Code:

```
var five = require("johnny-five");
var Raspi = require("raspi-io");

var board = new five.Board({
    io: new Raspi()
});

board.on("ready", function() { 

 // Led.Matrix HT16K33
    var matrix = new five.Led.Matrix({
        addresses: [0x48],
        controller: "HT16K33",
        rotation: 3,
    });

});
```

Looks like `0x48` is not in the list of allowed I2C addresses

rwaldron, 2016-04-09:
Thanks for the report. I looked at the link you provided and I don't see anywhere that it says HT16K33 is using address 0x48? More confusingly, I don't see that chip on this Explorer Hat thing: https://shop.pimoroni.com/products/explorer-hat... Maybe I'm misunderstanding you?

Let's start by simplifying all this a little bit. Take the Explorer Hat off the Pi, then connect _only_ your HT16K33 and do the i2cdetect scan. What does that output?

Take a look here: https://learn.adafruit.com/adafruit-led-backpack/changing-i2c-address it lists all the possible valid addresses for this chip.

punitganshani, 2016-04-10:
Without Explorer Hat, I see address `0x70` and it works!

I'll explore how to use it with Explorer Hat as well

closed 2016-04-10 by punitganshani

rwaldron, 2016-04-10:
Woo! Glad we could resolve that quickly :D